arendst/Tasmota #11521: ZooZee SE131 devices get warm in idle state

andiweli, 2021-03-30:
### PROBLEM DESCRIPTION
My [ZooZee SE131](https://templates.blakadder.com/zoozee_SE131.html) outlets get warm (29-32°C) even if not switched on. They're just connected to WiFi, have for example a lamp connected (10 watts LED) but not switched on.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): ZOOZEE_SF131
- [X] Tasmota binary firmware version number used: 9.3.1
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: OTA
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
15:27:39.629 MQT: stat/tasmota_42F67D/RESULT = {"NAME":"ZOOZEE_SE131","GPIO":[0,0,320,0,0,0,0,0,0,32,0,224,0,0],"FLAG":0,"BASE":18}
15:27:39.839 MQT: stat/tasmota_42F67D/RESULT = {"Module":{"0":"ZOOZEE_SE131"}}
15:27:40.092 MQT: stat/tasmota_42F67D/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"320":"Led_i1"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"32":"Button1"},"GPIO14":{"0":"None"},"GPIO15":{"224":"Relay1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
15:28:17.246 MQT: stat/tasmota_42F67D/STATUS = {"Status":{"Module":0,"DeviceName":"Stehlampe im Schlafzimmer","FriendlyName":["Stehlampe im Schlafzimmer"],"Topic":"tasmota_42F67D","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
15:28:17.252 MQT: stat/tasmota_42F67D/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-DE.bin","RestartReason":"Power On","Uptime":"0T22:25:22","StartupUTC":"2021-03-29T15:02:55","Sleep":50,"CfgHolder":4617,"BootCount":25,"BCResetTime":"2020-08-26T22:17:27","SaveCount":429,"SaveAddress":"F8000"}}
15:28:17.257 MQT: stat/tasmota_42F67D/STATUS2 = {"StatusFWR":{"Version":"9.3.1(tasmota)","BuildDateTime":"2021.02.23 12:08:50","Boot":4,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"489/699"}}
15:28:17.261 MQT: stat/tasmota_42F67D/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WirelessLAN",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
15:28:17.270 MQT: stat/tasmota_42F67D/STATUS4 = {"StatusMEM":{"ProgramSize":594,"Free":408,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
15:28:17.276 MQT: stat/tasmota_42F67D/STATUS5 = {"StatusNET":{"Hostname":"Stehlampe im Schlafzimmer","IPAddress":"10.0.0.104","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"CC:50:E3:42:F6:7D","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:28:17.283 MQT: stat/tasmota_42F67D/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.100","MqttPort":1884,"MqttClientMask":"Stehlampe im Schlafzimmer","MqttClient":"Stehlampe im Schlafzimmer","MqttUser":"astuermer","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
15:28:17.291 MQT: stat/tasmota_42F67D/STATUS7 = {"StatusTIM":{"UTC":"2021-03-30T13:28:17","Local":"2021-03-30T15:28:17","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"07:31","Sunset":"20:17"}}
15:28:17.299 MQT: stat/tasmota_42F67D/STATUS10 = {"StatusSNS":{"Time":"2021-03-30T15:28:17"}}
15:28:17.307 MQT: stat/tasmota_42F67D/STATUS11 = {"StatusSTS":{"Time":"2021-03-30T15:28:17","Uptime":"0T22:25:22","UptimeSec":80722,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER":"OFF","Wifi":{"AP":1,"SSId":"WirelessLAN","BSSId":"D8:32:14:B3:36:D1","Channel":6,"RSSI":84,"Signal":-58,"LinkCount":4,"Downtime":"0T00:00:49"}}}

```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
15:29:56.216 CFG: CR 489/699, Busy 0
15:29:56.219 LOG: SerialLog 0, WebLog 4, MqttLog 0, SysLog 0, LogHost , LogPort 514, TelePeriod 300
15:29:56.222 HTP: Einstellungen
15:29:56.001 CFG: in Flash gespeichert am F7, Anzahl 430, Bytes 4096
15:29:57.810 HTP: Hauptmenü
15:30:00.763 HTP: Konsole
15:30:07.441 WIF: Prüfe Verbindung...
15:30:27.425 WIF: Prüfe Verbindung...
15:30:47.442 WIF: Prüfe Verbindung...
15:31:07.418 WIF: Prüfe Verbindung...
15:31:27.454 WIF: Prüfe Verbindung...
15:31:47.447 WIF: Prüfe Verbindung...
15:32:07.417 WIF: Prüfe Verbindung...
15:32:27.440 WIF: Prüfe Verbindung...
15:32:47.422 WIF: Prüfe Verbindung...
15:33:04.938 MQT: tele/tasmota_42F67D/STATE = {"Time":"2021-03-30T15:33:04","Uptime":"0T22:30:09","UptimeSec":81009,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER":"OFF","Wifi":{"AP":1,"SSId":"WirelessLAN","BSSId":"D8:32:14:B3:36:D1","Channel":6,"RSSI":86,"Signal":-57,"LinkCount":4,"Downtime":"0T00:00:49"}}
15:33:07.445 WIF: Prüfe Verbindung...
15:33:27.438 WIF: Prüfe Verbindung...
15:33:47.687 WIF: Prüfe Verbindung...
15:34:07.699 WIF: Prüfe Verbindung...

```

### TO REPRODUCE
No special steps required.

### EXPECTED BEHAVIOUR
The device should stay "cool" as long as there is no current. From what I expect it should only use a very low "Standby" from about 0,1-0,3 watts.

### SCREENSHOTS
none

### ADDITIONAL CONTEXT
I thought lowering the log-level could help, so the processor of the outlet has no/lesser work to do.
So I set the serial log level to "0" and the web log level to "2".
But it still is around 30°C.

andiweli, 2021-03-30:
I have 12 of this devices, 2 are still with original "Tuya" firmware. They stay cool.
All others have Tasmota and get warm, even if the connected device isn't switched on or even if nothing is connected.

Jason2866, 2021-03-30:
Tasmota has a lot of more functions as Tuya. So the ESP8266 is not so often in idle mode as with Tuya firmware. You can rise the `sleep` value or use Tasmota-light since it is only a power plug
without any other function.

andiweli, 2021-03-30:
> Tasmota has a lot of more functions as Tuya. So the ESP8266 is not so often in idle mode as with Tuya firmware. You can rise the `sleep` value or use Tasmota-light since it is only a power plug
> 
> without any other function.

Thanks. 
I will try setting sleep to 250 at first and report back here.

Jason2866, 2021-03-31:
The temperatures you have are normal.

closed 2021-04-01 by ascillato2