arendst/Tasmota #11522: Dimming level is not sent from one dimmer to the other using a Device Group

JeroenEeuwes, 2021-03-30:
### PROBLEM DESCRIPTION
Two Shelly dimmers don't update the dimming level to each other when using a Device Group.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Shelly Dimmer and Shelly Dimmer 2
- [X] Tasmota binary firmware version number used: 9.3.1
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: firefox using the web interface of the modules
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
Shelly Dimmer:
15:46:28.517 MQT: tsm/slaapkamer-Jeroenen/stat/RESULT = {"NAME":"Shelly Dimmer 1","GPIO":[0,3200,0,3232,5568,5600,0,0,192,0,193,288,0,4736],"FLAG":0,"BASE":18}
15:46:28.544 MQT: tsm/slaapkamer-Jeroenen/stat/RESULT = {"Module":{"0":"Shelly Dimmer 1"}}
15:46:28.603 MQT: tsm/slaapkamer-Jeroenen/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"3200":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3232":"Serial Rx"},"GPIO4":{"5568":"SHD Boot 0"},"GPIO5":{"5600":"SHD Reset"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"192":"Switch_n1"},"GPIO13":{"0":"None"},"GPIO14":{"193":"Switch_n2"},"GPIO15":{"288":"Led1"},"GPIO16":{"0":"None"},"GPIO17":{"4736":"ADC Temp"}}

Shelly Dimmer 2:
15:48:02.163 MQT: tsm/slaapkamer-Jeroenen2/stat/RESULT = {"NAME":"Shelly Dimmer 2","GPIO":[0,3200,0,3232,5568,5600,0,0,193,0,192,0,320,4736],"FLAG":0,"BASE":18}
15:48:02.375 MQT: tsm/slaapkamer-Jeroenen2/stat/RESULT = {"Module":{"0":"Shelly Dimmer 2"}}
15:48:02.632 MQT: tsm/slaapkamer-Jeroenen2/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"3200":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3232":"Serial Rx"},"GPIO4":{"5568":"SHD Boot 0"},"GPIO5":{"5600":"SHD Reset"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"193":"Switch_n2"},"GPIO13":{"0":"None"},"GPIO14":{"192":"Switch_n1"},"GPIO15":{"0":"None"},"GPIO16":{"320":"Led_i1"},"GPIO17":{"4736":"ADC Temp"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
Shelly Dimmer:
15:48:46.739 MQT: tsm/slaapkamer-Jeroenen/stat/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":234,"Free":277,"Rules":"on switch1#state=2 do Power Toggle endon on switch2#state=2 do Power Toggle endon on switch1#state=4 do Dimmer + endon on switch2#state=4 do Dimmer - endon on switch1#state=8 do Dimmer 100 endon on switch2#state=8 do Dimmer 100 endon "}
15:48:46.789 MQT: tsm/slaapkamer-Jeroenen/stat/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":108,"Free":403,"Rules":"ON switch2#state=2 DO tsm/%topic%/cmnd/POWER OFF ENDON ON switch2#state=4 DO tsm/%topic%/cmnd/DIMMER - ENDON"}
15:48:46.840 MQT: tsm/slaapkamer-Jeroenen/stat/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

Shelly Dimmer 2:
15:49:43.257 MQT: tsm/slaapkamer-Jeroenen2/stat/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
15:49:43.468 MQT: tsm/slaapkamer-Jeroenen2/stat/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
15:49:43.720 MQT: tsm/slaapkamer-Jeroenen2/stat/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
Shelly Dimmer:
15:50:22.043 MQT: tsm/slaapkamer-Jeroenen/stat/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota-JJ","FriendlyName":["slaapkamer-Jeroenen"],"Topic":"slaapkamer-Jeroenen","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[11,11,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
15:50:22.051 MQT: tsm/slaapkamer-Jeroenen/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:16:58","StartupUTC":"2021-03-30T14:33:24","Sleep":50,"CfgHolder":4617,"BootCount":29,"BCResetTime":"2021-02-26T21:10:38","SaveCount":745,"SaveAddress":"F7000"}}
15:50:22.055 MQT: tsm/slaapkamer-Jeroenen/stat/STATUS2 = {"StatusFWR":{"Version":"9.3.1(tasmota)","BuildDateTime":"2021-03-09T16:12:28","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"509/699"}}
15:50:22.060 MQT: tsm/slaapkamer-Jeroenen/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Jeroenen-GeslotenStad32","Tasmota-fallback"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000800B","080514000100060000005A0A000000000000","00000080","00006048","00000000"]}}
15:50:22.068 MQT: tsm/slaapkamer-Jeroenen/stat/STATUS4 = {"StatusMEM":{"ProgramSize":593,"Free":408,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
15:50:22.076 MQT: tsm/slaapkamer-Jeroenen/stat/STATUS5 = {"StatusNET":{"Hostname":"slaapkamer-Jeroenen-5357","IPAddress":"192.168.2.33","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.254","Mac":"98:F4:AB:D1:74:ED","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:50:22.082 MQT: tsm/slaapkamer-Jeroenen/stat/STATUS6 = {"StatusMQT":{"MqttHost":"pimeterkast","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D174ED","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
15:50:22.091 MQT: tsm/slaapkamer-Jeroenen/stat/STATUS7 = {"StatusTIM":{"UTC":"2021-03-30T14:50:22","Local":"2021-03-30T15:50:22","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:31","Sunset":"19:17"}}
15:50:22.095 MQT: tsm/slaapkamer-Jeroenen/stat/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
15:50:22.103 MQT: tsm/slaapkamer-Jeroenen/stat/STATUS10 = {"StatusSNS":{"Time":"2021-03-30T15:50:22","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":52.1},"ENERGY":{"TotalStartTime":"2021-02-28T00:00:00","Total":0.004,"Yesterday":0.000,"Today":0.000,"Power":0},"TempUnit":"C"}}
15:50:22.109 MQT: tsm/slaapkamer-Jeroenen/stat/STATUS11 = {"StatusSTS":{"Time":"2021-03-30T15:50:22","Uptime":"0T00:16:58","UptimeSec":1018,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER":"OFF","Dimmer":57,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Jeroenen-GeslotenStad32","BSSId":"24:A4:3C:2D:59:40","Channel":11,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:05"}}}

Shelly Dimmer 2:
15:50:55.691 MQT: tsm/slaapkamer-Jeroenen2/stat/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota-JJ2","FriendlyName":["slaapkamer-Jeroenen2"],"Topic":"slaapkamer-Jeroenen2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
15:50:55.698 MQT: tsm/slaapkamer-Jeroenen2/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:17:25","StartupUTC":"2021-03-30T14:33:30","Sleep":50,"CfgHolder":4617,"BootCount":19,"BCResetTime":"2021-02-26T12:30:45","SaveCount":659,"SaveAddress":"F7000"}}
[... cut ...]
16:07:41.599 MQT: tsm/slaapkamer-Jeroenen2/stat/RESULT = {"POWER":"ON"}
16:07:41.604 MQT: tsm/slaapkamer-Jeroenen2/stat/POWER = ON
16:07:53.741 MQT: tsm/slaapkamer-Jeroenen2/stat/RESULT = {"POWER":"OFF"}
16:07:53.746 MQT: tsm/slaapkamer-Jeroenen2/stat/POWER = OFF

```

Note that using DevGroupSend 4=xxx doesn't actaully change the dimming level either:

```lua
16:10:20.539 CMD: DevGroupSend 4=128
16:10:20.549 MQT: tsm/slaapkamer-Jeroenen/stat/RESULT = {"DevGroupSend":"4=128"}
```

**(Please, remember to close the issue when the problem has been addressed)**

pcdiem, 2021-03-30:
There is a know bug in 9.3.1. It fails to initialize the device group ties when 9.3.1 is the initial version. Please enter the command DevGroupTie1 1 and try again.

JeroenEeuwes, 2021-03-30:
> 
> 
> There is a known bug in 9.3.1. It fails to initialize the device group ties when 9.3.1 is the initial version. Please enter the command DevGroupTie1 1 and try again.

Thanks for your message. I entered the command on both dimmers, but the result remains unchanged. I can turn the dimmer on/off, but the dimming level on the second module doesn't change when I change it on the first module.

JeroenEeuwes, 2021-03-30:
```lua
Shelly Dimmer:
18:45:11.793 CMD: DevGroupTie1 1
18:45:11.801 MQT: tsm/slaapkamer-Jeroenen/stat/RESULT = {"DevGroupTie1":1}
18:45:34.439 MQT: tsm/slaapkamer-Jeroenen/stat/RESULT = {"POWER":"ON","Dimmer":33}
18:45:53.575 MQT: tsm/slaapkamer-Jeroenen/stat/RESULT = {"POWER":"ON","Dimmer":87}
18:46:59.227 MQT: tsm/slaapkamer-Jeroenen/stat/RESULT = {"POWER":"ON","Dimmer":2}
18:48:10.513 MQT: tsm/slaapkamer-Jeroenen/stat/RESULT = {"POWER":"OFF"}
18:48:10.517 MQT: tsm/slaapkamer-Jeroenen/stat/POWER = OFF

Shelly Dimmer2:
18:45:25.300 CMD: DevGroupTie1 1
18:45:25.307 MQT: tsm/slaapkamer-Jeroenen2/stat/RESULT = {"DevGroupTie1":1}
18:45:34.236 MQT: tsm/slaapkamer-Jeroenen2/stat/RESULT = {"POWER":"ON"}
18:45:34.242 MQT: tsm/slaapkamer-Jeroenen2/stat/POWER = ON
18:45:42.284 MQT: tsm/slaapkamer-Jeroenen2/stat/RESULT = {"POWER":"ON","Dimmer":16}
18:48:08.952 MQT: tsm/slaapkamer-Jeroenen2/stat/RESULT = {"POWER":"ON","Dimmer":71}
18:48:10.248 MQT: tsm/slaapkamer-Jeroenen2/stat/RESULT = {"POWER":"OFF"}
18:48:10.253 MQT: tsm/slaapkamer-Jeroenen2/stat/POWER = OFF
```

As you can see, I can change the dimming level in the web interface on both modules, but the new dimming level does not get send to the other module. The only command that gets send is on/off.

pcdiem, 2021-03-30:
The docs were changed as some point to incorrect values for DevGroupSend. The item code for the light brightness is 5. Please post the console output from the following commands on Shelly Dimmer2:
Backlog Weblog 4; DevGroupSend 5=128; DevGroupSend 5=64

JeroenEeuwes, 2021-03-31:
```lua
06:52:02.598 CMD: Backlog Weblog 4; DevGroupSend 5=128; DevGroupSend 5=64
06:52:02.676 MQT: tsm/slaapkamer-Jeroenen2/stat/RESULT = {"WebLog":4}
06:52:02.880 SRC: Backlog
06:52:02.883 CMD: Group 0, Index 1, Command "DEVGROUPSEND", Data "5=128"
06:52:02.887 DGR: Sending SK-JJ message to network: seq=8, flags=0, 5=128
06:52:02.891 DGR: Received SK-JJ message from local: seq=8, flags=0, 5=128*
06:52:02.896 MQT: tsm/slaapkamer-Jeroenen2/stat/RESULT = {"DevGroupSend":"5=128"}
06:52:03.130 DGR: Sending SK-JJ message to 192.168.2.33: seq=8, flags=0, last ack=7
06:52:03.133 SRC: Backlog
06:52:03.136 CMD: Group 0, Index 1, Command "DEVGROUPSEND", Data "5=64"
06:52:03.140 DGR: Sending SK-JJ message to network: seq=9, flags=0, 5=64
06:52:03.144 DGR: Received SK-JJ message from local: seq=9, flags=0, 5=64*
06:52:03.149 MQT: tsm/slaapkamer-Jeroenen2/stat/RESULT = {"DevGroupSend":"5=64"}
06:52:03.280 DGR: Received SK-JJ message from 192.168.2.33: seq=8, flags=8
06:52:03.380 DGR: Sending SK-JJ message to 192.168.2.33: seq=9, flags=0, last ack=8
06:52:03.431 DGR: Received SK-JJ message from 192.168.2.33: seq=9, flags=8
06:52:03.833 CFG: Saved to flash at F6, Count 676, Bytes 4096
06:52:20.093 WIF: Checking connection...
06:52:21.051 DGR: Received SK-JJ message from 192.168.2.33: seq=17, flags=64
```

pcdiem, 2021-03-31:
Did the dimmer level change on Shelly Dimmer 2 when  you did those DevGroupSend commands?

JeroenEeuwes, 2021-03-31:
If you only enter the commands nothing happens. If you turn the lamp on first and then enter the commands in the console the dimming level of the lamp actually changes (I was bright enough to set it high before issuing the commands). The dimmer level on the web interface does not change and remains on "high" even though the lamp is not.

JeroenEeuwes, 2021-03-31:
If I issue the commands on the Shelly Dimmer the light connected to the Shelly Dimmer 2 changes brightness too (I put the lamp on high again first).

So the communication between the 2 modules seems to work fine, I'd say.

pcdiem, 2021-03-31:
The web interface dimmer level does not update in real time. If you change the dimmer level from a command/MQTT/device groups, the web interface will not change unless you refresh the page.

Are you're saying the dimmer level updates on Shelly Dimmer 2 when you issue the DevGroupSend command on Shelly Dimmer 1 but it does not update when you issue the Dimmer command?

JeroenEeuwes, 2021-03-31:
Ok, I didn't know the dimmer level in the web interface is not updated. Things like temperature and On/Off are updates, so I thought that everything on the page would be. 

I'm not really sure what changed between yesterday and today, but it seems to work now. Did it help that I have sent a DevGroupSend 5=128 ?

The only other thing I changed was setting the timezone correctly with 
```lua
TimeZone 99
TimeDST 0,0,3,1,2,120
TimeSTD 0,0,10,1,3,60
```
but I guess it's unlikely that would have any effect.

closed 2021-04-01 by ascillato2