arendst/Tasmota #11526: Changed IP address of Zbbridge and Alexa can no longer see devices

bwims, 2021-03-31:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I have a Zbbridge with 3 lights and 2 sensors.   With Hue emulation, Alexa detected the lights and all was well.  I later needed to reorganize my IP address pool and assigned a different address to the zbbridge.  I removed the lights and performed a discovery.  Alexa can no longer see the lights.  I have tried turning emulation on/off, rebooting but to no avail.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): __Zbbridge___
- [ ] Tasmota binary firmware version number used: __9.3.1(zbbridge)___
  - [ x] Pre-compiled
  - [ ] Self-compiled
- [x ] Flashing tools used: __Tasmotizer_1.2__
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
10:41:24.348 MQT: stat/zbbridge1/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
10:41:24.555 MQT: stat/zbbridge1/RESULT = {"Module":{"75":"Sonoff ZbBridge"}}
10:41:24.807 MQT: stat/zbbridge1/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
10:42:25.729 MQT: stat/zbbridge1/STATUS = {"Status":{"Module":75,"DeviceName":"Zigbee Bridge","FriendlyName":["Zigbee Bridge"],"Topic":"zbbridge1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":8,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
10:42:25.733 MQT: stat/zbbridge1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T13:31:22","StartupUTC":"2021-03-30T21:11:03","Sleep":50,"CfgHolder":4617,"BootCount":23,"BCResetTime":"2021-03-06T14:53:44","SaveCount":84,"SaveAddress":"1FB000"}}
10:42:25.736 MQT: stat/zbbridge1/STATUS2 = {"StatusFWR":{"Version":"9.3.1(zbbridge)","BuildDateTime":"2021-02-23T12:05:44","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"418/699"}}
10:42:25.738 MQT: stat/zbbridge1/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["wimsIOT","wims"],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","000000A0","40006080","00000010"]}}
10:42:25.744 MQT: stat/zbbridge1/STATUS4 = {"StatusMEM":{"ProgramSize":573,"Free":1216,"Heap":25,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1007C4","04400001","00000002","00000000","00000000","00020000","00000000","00000000"],"Drivers":"1,2,7,9,10,20,23,41","Sensors":""}}
10:42:25.748 MQT: stat/zbbridge1/STATUS5 = {"StatusNET":{"Hostname":"zbbridge1-2539","IPAddress":"192.168.1.211","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"C8:2B:96:04:C9:EB","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
10:42:25.753 MQT: stat/zbbridge1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.91","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_04C9EB","MqttUser":"DVES_USER","MqttCount":3,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
10:42:25.758 MQT: stat/zbbridge1/STATUS7 = {"StatusTIM":{"UTC":"2021-03-31T10:42:25","Local":"2021-03-31T10:42:25","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+00:00","Sunrise":"05:29","Sunset":"18:19"}}
10:42:25.761 MQT: stat/zbbridge1/STATUS10 = {"StatusSNS":{"Time":"2021-03-31T10:42:25"}}
10:42:25.766 MQT: stat/zbbridge1/STATUS11 = {"StatusSTS":{"Time":"2021-03-31T10:42:25","Uptime":"0T13:31:22","UptimeSec":48682,"Vcc":3.480,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":3,"Wifi":{"AP":1,"SSId":"wimsIOT","BSSId":"50:D4:F7:45:1E:F6","Channel":11,"RSSI":100,"Signal":-41,"LinkCount":2,"Downtime":"0T00:00:16"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Presumably, as stated in the problem description.   It's the kind of problem you can only experience once.
 I have not yet deleted and remapped anything, and hope to avoid that.  If I did get them working again, I would not want to reproduce the problem.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

As stated in the problem description.  Alexa should see my lights.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Barracuda09, 2021-04-01:
Hello,

I have the same problem with current Tasmota version, after the Echo firmware update.

s-hadinger, 2021-04-01:
If it worked already and the problem happened after network changes, it is most likely due to network blocking multicast traffic required by UPNP. Please set `weblog 4` and paste the logs from the period when you do the discovery.

Barracuda09, 2021-04-01:
No it has nothing to do with network change, it was about deleting and starting over again, and that fails.

s-hadinger, 2021-04-01:
Logs ?

Barracuda09, 2021-04-01:
@s-hadinger  the problem is here:
https://github.com/arendst/Tasmota/blob/0934fad06dcbf68f7f339878c7cd58c5d4bc7367/tasmota/xdrv_20_hue.ino#L410-L418

For some reason id is sometimes a 8 bit value:
https://github.com/arendst/Tasmota/blob/0934fad06dcbf68f7f339878c7cd58c5d4bc7367/tasmota/xdrv_20_hue.ino#L596

And in case of Zbbridge it is an 16 bit value so after the fix it becomes: **5c:cf:7f:13:9f:3d:00:11-01xx**
 which is now rejected by the new firmware of the Echo

s-hadinger, 2021-04-01:
Oh right.

Barracuda09, 2021-04-01:
Yes, so a solution could be to construct _HueDeviceId_ using this 16 bit value as two 8 bit values

bwims, 2021-04-01:
Many thanks for your help.  I have added my weblog(4) above as an edit.

Thanks!

s-hadinger, 2021-04-01:
@Barracuda09 Maybe we can use the following format: `5c:cf:7f:13:9f:3d:yy:11-xx` with xx being the low 8 bits, and yy the high 8 bits. This would make no change for regular lights, and make unique ids for Zigbee

Barracuda09, 2021-04-01:
@s-hadinger Yes, that is what I have done indeed

Barracuda09, 2021-04-01:
@s-hadinger 

It works for me but the Hue Light-API is saying:
**Unique id of the device. The MAC address of the device with a unique endpoint id in the form: AA:BB:CC:DD:EE:FF:00:11-XX**

So i'm not fully confided that this will keep working in the future. (when they update the Echo firmware again)

bwims, 2021-04-01:
Sounds like you've recompiled everything... is there any chance this will find its way into development?

s-hadinger, 2021-04-01:
@Barracuda09 Great. Are you preparing a PR?

bwims, 2021-04-02:
Just to clarify, am I to understand that Amazon changed its discovery protocol for Hue devices between the time when I first installed my Zb bridge and changed its IP address, and the emulation is therefore currently broken?

PR #11576 (2021-04-04): Fix Alexa discovery for ZBBridge
## Description:

**Related issue (if applicable):** fixes #11526

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with core ESP32 V.1.0.6
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).

_NOTE [... cut]
closed 2021-04-04 by s-hadinger