arendst/Tasmota #11535: [REQUEST] Zigbee Router firmware for EFR32MG21 in Sonoff ZBBridge by ITead?

Hedda, 2021-04-01:
**Have you looked for this feature in other issues and in the docs?**  

Yes.

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

No.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

Please consider providing a "Zigbee Router" firmware for EFR32MG21 in Sonoff ZBBridge by ITead as an alternative firmware for it.

Sonoff ZBBridge Zigbee to WiFi bridge can already work as a "Zigbee Coordinator" and gateway/hub with the existing NCP firmware that the Tasmota project provides for the Zigbee radio MCU SoC inside it, but if the project could also provide a "Zigbee Router" firmware for it then users could buy additional Sonoff ZBBridge devices and flash with that alternative firmware in order to use them as Zigbee router devices (sometimes also known Zigbee signal repeaters or Zigbee range extenders).

Sonoff ZBBridge could make a perfect Zigbee Router device due to its low-cost, powerful Zigbee radio, and stand-alone design.

The idea would to be to run Tasmota on its ESP8266 MCU and Silabs EmberZNet Zigbee Router firmware on its EFR32MG21 MCU.

https://www.itead.cc/sonoff-zbbridge.html

https://www.cnx-software.com/2020/07/27/sonoff-zigbee-bridge-tasmota-firmware-home-assistant-zigbee2mqtt/

https://www.cnx-software.com/2020/04/16/sonoff-zbbridge-wifi-to-zigbee-gateway/

**Describe alternatives you've considered**  

Some of the currently popular Zigbee Router alternatives include these:

CC2652P router by tube0013
https://community.home-assistant.io/t/zigbee-router-on-steroids/280896
https://github.com/tube0013/tube_gateways/tree/main/tube_cc_router
https://www.tubeszb.com/product/cc_router/4?cp=true&sa=false&sbp=false&q=false&category_id=3

CC2530 (with CC2591 or CC2592 Power Amplifier) DIY based router as recommended the Zigbee2MQTT community: 
https://www.zigbee2mqtt.io/how_tos/how_to_create_a_cc2530_router.html

IKEA Trådfri Signal Repeater
https://www.ikea.com/us/en/p/tradfri-signal-repeater-30400407/
https://www.ikea.com/ca/en/p/tradfri-signal-repeater-30400407/
https://www.ikea.com/gb/en/p/tradfri-signal-repeater-80424255/

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

Hedda, 2021-04-01:
Btw, I wrote these sections for Home Assistant's ZHA integration documentation to emphasize the importance of Zigbee routers:

### Using router devices

Zigbee uses mesh networking at the core of its architecture design and adding devices capable of acting as Zigbee router (sometimes also known as Zigbee signal repeaters or extenders) is very important because they increasing the total number of Zigbee devices that your Zigbee coordinator can control in its network as well as having an integral role in extending the overall range and coverage of your Zigbee network mesh. The fact is that without having at least a couple of good Zigbee router devices available you will normally not have a stable Zigbee network.

The total number of Zigbee devices that you have on a Zigbee network depends on a few things, but you should know that Zigbee coordinator hardware and firmware usually only start to play a larger role once you have around 40 devices or more in your Zigbee mesh network. More important is how many directly connected devices ("direct children") versus how many routers are and can be connected to your Zigbee coordinator. Each type and model of Zigbee coordinator also gave a different limit for how many child devices can connect directly to it. Even if that limit is low on your Zigbee coordinator if your Zigbee coordinator hardware is powerful and its firmware capable of it then you can still have a total of hundreds or even thousands of Zigbee devices as long as they are indirectly connected through a mesh network with Zigbee router devices instead.

The least powerful Zigbee coordinator hardware supported by Zigpy is CC2530/2531 and its default firmware only supports 15 devices connected directly to the coordinator. However, by having a backbone of good always-on Zigbee routers in your Zigbee network, the mesh network size can be extended exponentially. You can assume that most, if not all mains-powered/AC-powered devices, e.g., wall-plugs and always powered-on lightbulbs in your Zigbee network can serve as a Zigbee router and can typically each act as a router for around 5 additional end devices, while some dedicated Zigbee router devices can each act as a router for as many as 50 devices. You can for example use a DIY CC2530/CC2531 with router firmware which have a limit of 21 devices, while a DIY CC2652P/CC2652R with router firmware which have a limit of 50 devices, or the IKEA TRÅDFRI Signal Repeater which can route about 10-20 devices.

An example using a CC2531 with default coordinator firmware + two CC2531 routers; Your Zigbee network device limit will be:

- Coordinator: 15 devices - 2 routers = 13
- Router one: + 21 devices
- Router two: + 21 devices
- Total device limit = **55 devices**

As an comparison example, using a CC2652 with default coordinator firmware + two CC652 routers; Your Zigbee network device limit will be:

- Coordinator: 50 devices - 2 routers = 48
- Router one: + 50 devices
- Router two: + 50 devices
- Total device limit = **148 devices**

### Improving Zigbee network range

Low signal quality can lead to transmission errors and related issues. This section has a list of fundamental tips on how to improve signal quality. Improving signal quality also maximizes range and resolves most problems related to transmission errors. Please try to follow at least some of these recommendations before posting on the community forums or reporting issues to developers and submitting debug logs.

1. **Add more Zigbee routers between devices far away and the next closest router. Distribute more Zigbee routers in areas with poor reception. Zigbee network topology uses a "mesh network" design which means that each device that acts as a Zigbee router will extend the range of your Zigbee network as a whole. While there are exceptions, almost all permanently powered devices serve as a Zigbee router. Adding more permanently powered Zigbee devices allows getting better coverage. There are also dedicated Zigbee routers which you can [find by doing a community search for "Zigbee signal repeater" or "Zigbee range extender"](https://community.home-assistant.io/t/advice-on-zigbee-range-extending/175882/)). Devices that can not act as routers are typically battery-operated and known as "end devices". Some end devices have issues connecting through routers (e.g. Xiaomi/Aqara devices).**
2. Use a USB extension cable. This allows positioning the Zigbee coordinator adapter for better signal quality. Position the Zigbee coordinator away from Wi-Fi access-points/routers or other sources of 2.4GHz signals to avoid signal interference. The best location depends on your building's floorplan. Ideally, you want to place the Zigbee coordinator somewhere in the middle. Building materials do influence signal quality too, for example, dense/thick concrete, bricks, masonry, etc. dampen all wireless signals. Place the Zigbee coordinator with some distance from walls, ceilings and floors. Also, try different orientations of your Zigbee antenna (or your whole Zigbee coordinator adapter if it has an internal antenna). Some Antennas have a stronger signal in a certain direction. Simply changing orientation can improve signal quality already. Note: A bad USB extension cable may lead to connection issues between the system and the Zigbee coordinator adapter, symptoms of this are disconnection messages.
[... cut ...] , ceilings and floors.
6. Buy more powerful Zigbee radio hardware with better radio range, preferably with an external antenna. If you are not only experimenting but want a permanently stable and healthy Zigbee network with potentially many devices then you should consider upgrading to a more powerful Zigbee coordinator USB adapter. Generally, those with an external antenna will have better range, therefore you will also want to avoid buying an internal adapter unless it has an external antenna.

ascillato2, 2021-04-01:
Moving to discussions as it is a feature request and not an issue.

closed 2021-04-01 by ascillato2