arendst/Tasmota #11542: Relais schaltet kurz bei Spannung anlegen - sollte aber nicht!

He2205, 2021-04-01:
### PROBLEM DESCRIPTION
Hallo Zusammen,

ich hab jetzt einige Beiträge dazu gehört und es auch so probiert aber das Relais schaltet immer noch kurz ein beim Bewegen diraltenalten.

hab das template geladen {"NAME": "ESP-01S-Relais", "GPIO": [29,52,255,255,0,0,0,0,0,0,0,0], "FLAG": 0, "BASE": 18}
und das funktioniert auch mit dem Schild, wenn ich auf der Weboberfläche EIN / AUS drücke.

ich hab savedata 1 gemacht und poweronstate 0

in der Konsole gehört und bleibt auch so nach dem ausgeschalten.
Version ist die 9.3.1 Tasmota
MQTT Server bewusst noch nicht erledigtaltet.
mit tasmotizer 1.2 und Erase Flash geflasht.

Leider geht das Relais beim Einschalten immer noch kurz und dann geht das Garagen Tor auf.

Hab ich war gehört ???

Danke im Voraus ....

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [x ] Tasmota binary firmware version number used: tasmota_-DE.bin____
  - [ ] Pre-compiled
  - [ x] Self-compiled
- [ x] Flashing tools used: tasmotiter 1.2
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-04-01:
It is not a Tasmota issue. Since ESP-01 only has restricted GPIOs available 
and you have to use this GPIOs for peripherals to connect. Relais blinking is not avoidable.
See https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/

He2205, 2021-04-01:
ah ok,
thank you for the link.
that waht i read is, the esp01 is not good to use with relais.
all the gpio are on high at the boot, right?

i must use an esp12f or an wemos?

ascillato2, 2021-04-01:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-04-01 by ascillato2

He2205, 2021-04-01:
with the wemos is his the same.
i use the gpio 4 (D1)  and 5 (D2).
The relais is also short switch on.