arendst/Tasmota #11549: Unit of meassure for BH1750 missing

azakow, 2021-04-02:
### PROBLEM DESCRIPTION
for each sensor attached Tasmota provides the unit of measure, i.e.
```
{
  "Time" : "2021-04-02T08:01:16",
  "BME280" : {
    "Temperature" : 8.1,
    "Humidity" : 57.4,
    "DewPoint" : 0.2,
    "Pressure" : 1008.3
  },
  "BH1750" : {
    "Illuminance" : 5869
  },
  "AS3935" : {
    "Event" : 4,
    "Distance" : 1,
    "Energy" : 46810,
    "Stage" : 11
  },
  "WindMeter" : {
    "Speed" : {
      "Act" : 0.0,
      "Avg" : 0.48,
      "Min" : 0.0,
      "Max" : 4.86
    }
  },
  "PressureUnit" : "hPa",
  "TempUnit" : "C",
  "SpeedUnit" : "km/h"
}

```
According to the data sheet of sensor BH1750 the unit of measure is "lx", hence Tasmota should provide this sensor information i.e. "IlluminanceUnit" : "lx"

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): D1 Mini
- [X] Tasmota binary firmware version number used: 9.3.1
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: Tasmotiser 1.2
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
08:17:32.966 MQT: stat/tasmota_5AB05A/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
08:17:33.177 MQT: stat/tasmota_5AB05A/RESULT = {"Module":{"18":"Generic"}}
08:17:33.431 MQT: stat/tasmota_5AB05A/RESULT = {"GPIO0":{"4992":"WindMeter Spd"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"640":"I2C SDA"},"GPIO5":{"608":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"4672":"AS3935"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
08:18:16.867 MQT: stat/tasmota_5AB05A/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_5AB05A","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
08:18:16.873 MQT: stat/tasmota_5AB05A/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T00:54:16","StartupUTC":"2021-04-02T06:24:00","Sleep":50,"CfgHolder":4617,"BootCount":21,"BCResetTime":"2021-03-31T19:59:15","SaveCount":3688,"SaveAddress":"F4000"}}
08:18:16.877 MQT: stat/tasmota_5AB05A/STATUS2 = {"StatusFWR":{"Version":"9.3.1(tasmota)","BuildDateTime":"2021-03-31T13:18:42","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"362/699"}}
08:18:16.881 MQT: stat/tasmota_5AB05A/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BINAPPL2",""],"TelePeriod":300,"Resolution":"558180C4","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
08:18:16.890 MQT: stat/tasmota_5AB05A/STATUS4 = {"StatusMEM":{"ProgramSize":606,"Free":396,"Heap":23,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000014CF","010013C0","C000F981","00004144","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6,9,10,67,68"}}
08:18:16.897 MQT: stat/tasmota_5AB05A/STATUS5 = {"StatusNET":{"Hostname":"tasmota_5AB05A-4186","IPAddress":"192.168.5.36","Gateway":"192.168.5.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.5.254","Mac":"48:3F:DA:5A:B0:5A","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
08:18:16.903 MQT: stat/tasmota_5AB05A/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.5.238","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_5AB05A","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
08:18:16.910 MQT: stat/tasmota_5AB05A/STATUS7 = {"StatusTIM":{"UTC":"2021-04-02T07:18:16","Local":"2021-04-02T08:18:16","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:25","Sunset":"19:21"}}
08:18:16.920 MQT: stat/tasmota_5AB05A/STATUS10 = {"StatusSNS":{"Time":"2021-04-02T08:18:16","BME280":{"Temperature":8.1,"Humidity":56.2,"DewPoint":-0.1,"Pressure":1008.4},"BH1750":{"Illuminance":6573},"AS3935":{"Event":0,"Distance":0,"Energy":0,"Stage":11},"WindMeter":{"Speed":{"Act":2.43,"Avg":0.67,"Min":0.00,"Max":4.86}},"PressureUnit":"hPa","TempUnit":"C","SpeedUnit":"km/h"}}
08:18:16.927 MQT: stat/tasmota_5AB05A/STATUS11 = {"StatusSTS":{"Time":"2021-04-02T08:18:16","Uptime":"0T00:54:16","UptimeSec":3256,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"xxxxxxx-2","BSSId":"xxxxxxxxxxx","Channel":1,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Attach sensor BH1750 and see output  JSON

### EXPECTED BEHAVIOUR
According to the data sheet of sensor BH1750 the unit of meassure is "lx", hence Tasmota should provide this information i.e. "IlluminanceUnit" : "lx"

### SCREENSHOTS
-

### ADDITIONAL CONTEXT
data sheet for sensor BH1750
https://components101.com/sensors/bh1750-ambient-light-sensor

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-04-02:
AFAICT, those units are not meant to accompany every piece of sensor data, only those where there are features in Tasmota to select which unit is used, SetOption24 for mmHg/hPa and SetOption8 for F/C. Then parsing the data without knowing the current selection would not work very well.

azakow, 2021-04-02:
I had a very focused view on this sensor only. It makes sense, to have units of measssure for Tasmota features only.

closed 2021-04-02 by azakow