arendst/Tasmota #11553: PowerOnState 0 not honoured, OBI plug (51) acts as if PowerOnState was 1

RREE, 2021-04-02:
### PROBLEM DESCRIPTION
A recently upgraded OBI plug always switches on the relais after start up despite PowerOnState being 0

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): 51 OBI plug
- [x] Tasmota binary firmware version number used: 9.3.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tasmotizer 1.2
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
11:55:07.293 CMD: Backlog Template; Module; GPIO 255
11:55:07.363 MQT: heim/s131/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
11:55:07.572 MQT: heim/s131/RESULT = {"Module":{"51":"OBI Socket"}}
11:55:07.824 MQT: heim/s131/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"288":"Led1"},"GPIO5":{"224":"Relay1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"290":"Led3"},"GPIO13":{"0":"None"},"GPIO14":{"32":"Button1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
11:55:24.411 APP: Serial logging disabled
11:55:33.452 MQT: heim/s131/STATE = {"Time":"2021-04-02T11:55:33","Uptime":"0T00:10:09","UptimeSec":609,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"ebert-langer","BSSId":"38:10:D5:BC:3F:98","Channel":11,"RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:03"}}
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
11:57:12.088 CMD: Status 0
11:57:12.096 MQT: heim/s131/STATUS = {"Status":{"Module":51,"DeviceName":"s131","FriendlyName":["Steckdose 131"],"Topic":"s131","ButtonTopic":"0","Power":1,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
11:57:12.101 MQT: heim/s131/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T00:11:48","StartupUTC":"2021-04-02T10:45:24","Sleep":50,"CfgHolder":4617,"BootCount":18,"BCResetTime":"2021-04-02T11:33:24","SaveCount":36,"SaveAddress":"F8000"}}
11:57:12.106 MQT: heim/s131/STATUS2 = {"StatusFWR":{"Version":"9.3.1(tasmota)","BuildDateTime":"2021-03-09T16:12:28","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"389/699"}}
11:57:12.111 MQT: heim/s131/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["ebert-langer",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
11:57:12.121 MQT: heim/s131/STATUS4 = {"StatusMEM":{"ProgramSize":593,"Free":408,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
11:57:12.129 MQT: heim/s131/STATUS5 = {"StatusNET":{"Hostname":"s131","IPAddress":"192.168.178.131","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"2C:3A:E8:2F:31:22","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
11:57:12.136 MQT: heim/s131/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.7","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2F3122","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:57:12.144 MQT: heim/s131/STATUS7 = {"StatusTIM":{"UTC":"2021-04-02T10:57:12","Local":"2021-04-02T11:57:12","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:25","Sunset":"19:21"}}
11:57:12.151 MQT: heim/s131/STATUS10 = {"StatusSNS":{"Time":"2021-04-02T11:57:12"}}
11:57:12.158 MQT: heim/s131/STATUS11 = {"StatusSTS":{"Time":"2021-04-02T11:57:12","Uptime":"0T00:11:48","UptimeSec":708,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"ebert-langer","BSSId":"38:10:D5:BC:3F:98","Channel":11,"RSSI":100,"Signal":-49,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:00:00.050 CFG: Loaded from flash at F7, Count 37
00:00:00.055 QPC: Count 1
00:00:00.060 SRC: Restart
00:00:00.061 Project tasmota s131 Version 9.3.1(tasmota)-2_7_4_9(2021-03-09T16:12:28)
00:00:00.213 WIF: Checking connection...
00:00:00.214 WIF: Attempting connection...
00:00:00.551 WIF: Connecting to AP1 ebert-langer Channel 11 BSSId 38:10:D5:BC:3F:98 in mode 11n as s131...
00:00:01.755 WIF: Checking connection...
00:00:01.756 WIF: Connected
00:00:02.008 HTP: Web server active on s131 with IP address 192.168.178.131
00:00:02.460 RTC: UTC 2021-04-02T10:58:29, DST 2021-03-28T02:00:00, STD 2021-10-31T03:00:00
11:58:30.014 MQT: Attempting connection...
11:58:30.028 MQT: Connected
11:58:30.030 MQT: heim/s131/LWT = Online (retained)
11:58:30.032 MQT: heim/s131/cmnd/POWER = 
11:58:30.034 MQT: Subscribe to heim/s131/cmnd/#
11:58:30.035 MQT: Subscribe to heim/tasmotas/cmnd/#
11:58:30.037 MQT: Subscribe to cmnd/DVES_2F3122_fb/#
11:58:30.038 MQT: Unsubscribe from homeassistant/status
11:58:30.041 MQT: heim/s131/INFO1 = {"Module":"OBI Socket","Version":"9.3.1(tasmota)","FallbackTopic":"cmnd/DVES_2F3122_fb/","GroupTopic":"heim/tasmotas/cmnd/"}
11:58:30.047 MQT: heim/s131/INFO2 = {"WebServerMode":"Admin","Hostname":"s131","IPAddress":"192.168.178.131"}
11:58:30.056 MQT: heim/s131/INFO3 = {"RestartReason":"Software/System restart"}
11:58:30.063 MQT: heim/s131/RESULT = {"POWER":"ON"}
11:58:30.067 MQT: heim/s131/POWER = ON
11:58:30.323 SRC: MQTT
11:58:30.325 CMD: Group 0, Index 1, Command "POWER", Data "ON"
11:58:30.329 MQT: heim/s131/RESULT = {"POWER":"ON"}
11:58:30.332 MQT: heim/s131/POWER = ON
11:58:33.445 QPC: Reset
11:58:34.416 MQT: heim/s131/STATE = {"Time":"2021-04-02T11:58:34","Uptime":"0T00:00:09","UptimeSec":9,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"ebert-langer","BSSId":"38:10:D5:BC:3F:98","Channel":11,"RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:03"}}
11:58:35.412 APP: Boot Count 19
11:58:35.564 CFG: Saved to flash at F6, Count 38, Bytes 4096
11:58:39.404 MQT: tasmota/discovery/2C3AE82F3122/config = {"ip":"192.168.178.131","dn":"s131","fn":["Steckdose 131",null,null,null,null,null,null,null],"hn":"s131","mac":"2C3AE82F3122","md":"OBI Socket","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"9.3.1","t":"s131","ft":"heim/%topic%/","tp":["cmnd","stat","tele"],"rl":[1,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":1,"lt_st":0,"sho":[0,0,0,0],"ver":1} (retained)
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Set "PowerOnState 0" then restart "restart 1"

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The Relais should be off after restart (Power 0)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-04-02:
Seems that you are having mqtt retained messages in your broker. Your broker is telling Tasmota to power on the relay.

ascillato2, 2021-04-02:
Please, tell us if deleting retained messages solves your issue. Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-04-02 by ascillato2