arendst/Tasmota #11561: setting tmtoken returns "Text overflow by 41 char(s)"

FerFemNemBem, 2021-04-02:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I want to set my tmtoken as done in 8.4 by typing:
tmtoken 1299993482:AARR44jB-d8UUOP_4ZphUzzziLqfP4JfARE
(this is not my real token but I have not shortened or expanded it - onlich changed chars and numbers)
The console returns:
CFG: Text overflow by 41 char(s)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
22:05:03.587 MQT: steckdose/oben/kueche/geschirrspueler/RESULT = {"NAME":"ZX-2820-675 V2 Module","GPIO":[0,0,0,32,2688,2656,0,0,2624,321,224,0,0,1],"FLAG":0,"BASE":18}
22:05:03.989 MQT: steckdose/oben/kueche/geschirrspueler/RESULT = {"Module":{"0":"ZX-2820-675 V2 Module"}}
22:05:04.243 MQT: steckdose/oben/kueche/geschirrspueler/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"32":"Button1"},"GPIO4":{"2688":"HLW8012 CF"},"GPIO5":{"2656":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"321":"Led_i2"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
22:03:50.887 MQT: steckdose/oben/kueche/geschirrspueler/STATUS = {"Status":{"Module":0,"DeviceName":"Steckdose Geschirrspüler oben","FriendlyName":["Steckdose Geschirrspüler oben"],"Topic":"steckdose/oben/kueche/geschirrspueler/","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
22:03:50.894 MQT: steckdose/oben/kueche/geschirrspueler/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:10:13","StartupUTC":"2021-04-02T19:53:37","Sleep":250,"CfgHolder":5266,"BootCount":4,"BCResetTime":"2021-04-02T18:07:20","SaveCount":17,"SaveAddress":"FB000"}}
22:03:50.900 MQT: steckdose/oben/kueche/geschirrspueler/STATUS2 = {"StatusFWR":{"Version":"9.3.1(tasmota)","BuildDateTime":"2021-03-31T16:01:02","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"694/699"}}
22:03:50.908 MQT: steckdose/oben/kueche/geschirrspueler/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":2,"LogHost":"192.168.178.72","LogPort":514,"SSId":["FFNB-WLAN",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00208009","2805C80001000600001E5A00000000000000","000000C0","00006000","00000000"]}}
22:03:50.916 MQT: steckdose/oben/kueche/geschirrspueler/STATUS4 = {"StatusMEM":{"ProgramSize":517,"Free":484,"Heap":29,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F000483","04000001","00000007","000000C0","40000800","00008000","00000000","00000000"],"Drivers":"1,2,3,4,9,10,29,40","Sensors":"1,2,3"}}
22:03:50.924 MQT: steckdose/oben/kueche/geschirrspueler/STATUS5 = {"StatusNET":{"Hostname":"steckdose/oben/kueche/geschirrs","IPAddress":"192.168.178.23","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"D8:F1:5B:D1:D7:DF","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
22:03:50.929 MQT: steckdose/oben/kueche/geschirrspueler/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.72","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_D1D7DF","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
22:03:50.935 MQT: steckdose/oben/kueche/geschirrspueler/STATUS7 = {"StatusTIM":{"UTC":"2021-04-02T20:03:50","Local":"2021-04-02T22:03:50","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"06:42","Sunset":"19:44"}}
22:03:50.941 MQT: steckdose/oben/kueche/geschirrspueler/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
22:03:50.950 MQT: steckdose/oben/kueche/geschirrspueler/STATUS10 = {"StatusSNS":{"Time":"2021-04-02T22:03:50","ENERGY":{"TotalStartTime":"2021-04-02T21:01:04","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":254,"Current":0.000}}}
22:03:50.957 MQT: steckdose/oben/kueche/geschirrspueler/STATUS11 = {"StatusSTS":{"Time":"2021-04-02T22:03:50","Uptime":"0T00:10:13","UptimeSec":613,"Heap":29,"SleepMode":"Dynamic","Sleep":250,"LoadAvg":4,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"FFNB-WLAN","BSSId":"2C:91:AB:6D:1E:6C","Channel":6,"RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:07"}}}
22:03:51.381 MQT: steckdose/oben/kueche/geschirrspueler/STATE = {"Time":"2021-04-02T22:03:51","Uptime":"0T00:10:14","UptimeSec":614,"Heap":30,"SleepMode":"Dynamic","Sleep":250,"LoadAvg":15,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"FFNB-WLAN","BSSId":"2C:91:AB:6D:1E:6C","Channel":6,"RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:07"}}
22:03:51.391 MQT: steckdose/oben/kueche/geschirrspueler/SENSOR = {"Time":"2021-04-02T22:03:51","ENERGY":{"TotalStartTime":"2021-04-02T21:01:04","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":253,"Current":0.000}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
type "tmtoken 1299993482:AARR44jB-d8UUOP_4ZphUzzziLqfP4JfARE" to console and "Enter"

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
tmtoken should be set

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-04-03:
Congratiolations!!! You're the first to report "out of character pool" error message as can be seen at the end of report using command status 2.

The character pool holds all user configurable character strings and until now has always been big enough. To solve your issue just empty the pool a bit by shortening some strings like topic, friendlynames, devicenames or buttonnames until the reported pool size can store your tmtoken.

FerFemNemBem, 2021-04-03:
Hi @arendst

thanks for pointing to this.
I assume the my long WIFI-PW (64 chars) is part of the problem. Beside this I don't have that much variables/display names set.
This is my "user_config_override.h":
```
// -- Master parameter control --------------------
#define CFG_HOLDER             5266

// -- Project -------------------------------------
#define PROJECT                "Steckdose Geschirrspüler oben"
#define FRIENDLY_NAME          "Steckdose Geschirrspüler oben"

// the small one
#define USER_TEMPLATE          "{\"NAME\":\"ZX-2820-675 V2 Module\",\"GPIO\":[0,0,0,17,133,132,0,0,131,57,21,0,0],\"FLAG\":15,\"BASE\":18}"

#define STA_SSID1              "ABCD-WLAN"
#define STA_PASS1              "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX"
#define WIFI_ARP_INTERVAL      30
#define WIFI_SCAN_AT_RESTART   true
#define WIFI_SCAN_REGULARLY    true

#define MQTT_HOST              "192.168.178.72"
#define MQTT_PORT              1883

#define SYS_LOG_HOST           "192.168.178.72"
#define SYS_LOG_LEVEL          LOG_LEVEL_INFO

#define MQTT_FULLTOPIC         "steckdose/oben/kueche/geschirrspueler/"
#define MQTT_TOPIC             "steckdose/oben/kueche/geschirrspueler/"

#define APP_TIMEZONE 99
#define APP_SLEEP 250
#define TELE_PERIOD 60
#define USE_TELEGRAM
#define USE_RULES
#define USE_EXPRESSION
#define SUPPORT_IF_STATEMENT
#define ENERGY_VOLTAGE_ALWAYS  true
```
Beside this I have commented out all unused stuff from the "my_user_config.h" file on order to shrink the image.
Is there any other way to free some space or to expand the character pool?

Jason2866, 2021-04-03:
Your mqtt topic is long too

sfromis, 2021-04-03:
You're using approx 300 bytes more than I see with my configs (no long passwords or names). Assuming that this is the base level, you may have to budget your use of XL strings.

FerFemNemBem, 2021-04-03:
Changing the topic would cause changing >100 queries in grafana. So this is no easy task for me - ok, not your problem. ;)
I'm wondering why the character pool was sufficient in 8.4 (I use compile-templates and the configuration for 9.3.1 is exactly the same as for 8.4)
Because I compile by myself - ist there any place inside the code where I can expand the character pool - ideally without any side effects?
Thank you!

arendst, 2021-04-03:
No.

Revisit your full topic and topic name. The current definitions of both makes no sense.

FerFemNemBem, 2021-04-03:
I was lying!
I has changed the configuration (a little): 
```
#define PROJECT                "Steckdose Geschirrspueler oben"
#define FRIENDLY_NAME          "Steckdose Geschirrspueler oben"
```
"CR":"669/699"
```
#define PROJECT                "Steckdose Geschirrspüler oben"
#define FRIENDLY_NAME          "Steckdose Geschirrspüler oben"
"CR":"696/699"
```
So using umlauts seems to change the internal string representation and takes much more space.
Thank you all for your support.

sfromis, 2021-04-03:
The recommended FullTopic is either `"%prefix%/%topic%/` or `"%topic%/%prefix%/`. Using this means that the topic will be included where relevant, instead of repeating the info, and omitting the prefix.

FerFemNemBem, 2021-04-03:
I have shortened some strings and now all is working as expected.
Thank you for pointing to character pool and all other suggestions!

closed 2021-04-03 by FerFemNemBem