arendst/Tasmota #11577: Sonoff Pow R2 Modul & KNX no yesterday result will be send

wb86, 2021-04-04:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Tasmota allways send 0.000 to the "Energy Yesterday" KNX address.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Pow R2
- [x] Tasmota binary firmware version number used: 9.3.1.2
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

10:15:59.076 CMD: Backlog Template; Module; GPIO 255
10:15:59.151 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
10:15:59.387 RSL: RESULT = {"Module":{"43":"Sonoff Pow R2"}}
10:15:59.601 RSL: RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"3072":"CSE7766 Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3104":"CSE7766 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
10:16:10.419 RSL: STATE = {"Time":"2021-04-04T10:16:10","Uptime":"0T15:00:11","UptimeSec":54011,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxx","BSSId":"xxx","Channel":8,"RSSI":100,"Signal":-45,"LinkCount":1,"Downtime":"0T00:00:05"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

10:17:20.225 CMD: Status 0
10:17:20.232 RSL: STATUS = {"Status":{"Module":43,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_E31016","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
10:17:20.238 RSL: STATUS1 = {"StatusPRM":{"Baudrate":4800,"SerialConfig":"8E1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T15:01:21","StartupUTC":"2021-04-03T18:15:59","Sleep":50,"CfgHolder":4617,"BootCount":23,"BCResetTime":"2021-03-14T13:11:17","SaveCount":174,"SaveAddress":"F6000"}}
10:17:20.244 RSL: STATUS2 = {"StatusFWR":{"Version":"9.3.1.2(tasmota)","BuildDateTime":"2021.03.26 15:53:25","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"381/699"}}
10:17:20.249 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxx",""],"TelePeriod":600,"Resolution":"558180C0","SetOption":["02008001","2805C8000100068000005A0A000000000000","00000080","00006000","00000000"]}}
10:17:20.257 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":611,"Free":392,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000407","9FD2C787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,11,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
10:17:20.263 RSL: STATUS5 = {"StatusNET":{"Hostname":"Solar","IPAddress":"10.10.10.217","Gateway":"10.10.10.10","Subnetmask":"255.255.255.0","DNSServer":"10.10.10.10","Mac":"98:F4:AB:E3:10:16","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
10:17:20.270 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-04-04T09:17:20","Local":"2021-04-04T10:17:20","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:20","Sunset":"19:24"}}
10:17:20.274 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
10:17:20.282 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-04-04T10:17:20","ENERGY":{"TotalStartTime":"1970-01-01T00:00:04","Total":1.067,"Yesterday":0.887,"Today":0.161,"Power":71,"ApparentPower":77,"ReactivePower":29,"Factor":0.93,"Voltage":236,"Current":0.325}}}
10:17:20.288 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-04-04T10:17:20","Uptime":"0T15:01:21","UptimeSec":54081,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxx","BSSId":"xxx","Channel":8,"RSSI":100,"Signal":-41,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

10:26:09.094 DMP: 55 5A 02 DF 28 00 03 22 00 3D C7 00 B5 4B 4D 9F 18 01 08 A2 F1 7B 6C B9
10:26:09.199 DMP: 55 5A 02 DF 28 00 03 22 00 3D C7 00 B5 4B 4D 9F 18 01 09 5A F1 7B 6E 74
10:26:09.304 DMP: 55 5A 02 DF 28 00 03 22 00 3D C7 00 B5 4B 4D 9F 18 01 09 5A F1 7B 6F 75
10:26:09.409 DMP: 55 5A 02 DF 28 00 03 22 00 3D C7 00 B5 4B 4D 9F 18 01 09 5A F1 7B 71 77
10:26:09.520 DMP: 55 5A 02 DF 28 00 03 22 00 3D C7 00 B5 4B 4D 9F 18 01 09 5A F1 7B 72 78
10:26:09.619 DMP: 55 5A 02 DF 28 00 03 22 00 3D C7 00 B5 4B 4D 9F 18 01 08 D0 F1 7B 74 EF
10:26:09.723 DMP: 55 5A 02 DF 28 00 03 22 00 3D C7 00 B5 4B 4D 9F 18 01 08 D0 F1 7B 75 F0
10:26:09.774 WIF: Prüfe Verbindung...
10:26:09.829 DMP: 55 5A 02 DF 28 00 03 1F 00 3D C7 00 B5 4B 4D 9F 18 01 08 D0 F1 7B 77 EF
10:26:09.934 DMP: 55 5A 02 DF 28 00 03 1F 00 3D C7 00 B5 4B 4D 9F 18 01 08 D0 F1 7B 79 F1
10:26:10.038 DMP: 55 5A 02 DF 28 00 03 1F 00 3D C7 00 B5 4B 4D 9F 18 01 09 2A F1 7B 7A 4D
10:26:10.143 DMP: 55 5A 02 DF 28 00 03 1F 00 3D C7 00 B5 4B 4D 9F 18 01 09 2A F1 7B 7C 4F
10:26:10.248 DMP: 55 5A 02 DF 28 00 03 1F 00 3D C7 00 B5 4B 4D 9F 18 01 09 2A F1 7B 7D 50
10:26:10.354 DMP: 55 5A 02 DF 28 00 03 1F 00 3D C7 00 B5 4B 4D 9F 18 01 09 10 F1 7B 7F 38
10:26:10.418 RSL: STATE = {"Time":"2021-04-04T10:26:10","Uptime":"0T15:10:11","UptimeSec":54611,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxx","BSSId":"xxx","Channel":8,"RSSI":100,"Signal":-36,"LinkCount":1,"Downtime":"0T00:00:05"}}
10:26:10.426 KNX: Spannung gesendet an 0.5.2 
[... 56 lines cut ...]
10:26:15.603 DMP: 55 5A 02 DF 28 00 03 1F 00 3D C7 00 B5 5D 4D 9F 18 01 08 D4 F1 7B CC 5A
10:26:15.708 DMP: 55 5A 02 DF 28 00 03 1F 00 3D C7 00 B5 5D 4D 9F 18 01 09 8B F1 7B CE 14
10:26:15.813 DMP: 55 5A 02 DF 28 00 03 1F 00 3D C7 00 B5 5D 4D 9F 18 01 09 8B F1 7B CF 15

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Everytime when Tasmota send it to KNX.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-04-07:
Thanks for reporting. I will look into it.

ascillato, 2021-04-13:
Bug fixed with PR #11718

PR #11718 (2021-04-13): KNX: Fix Energy Yesterday Value
## Description:

KNX: Fix Energy Yesterday Value

**Related issue (if applicable):** fixes #11577

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with core ESP32 V.1.0.6
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributo [... cut]
closed 2021-04-13 by ascillato2