arendst/Tasmota #11594: Change Documentation to recommend Belkin WeMo for Amazon Alexa

RolfWojtech, 2021-04-05:
### PROBLEM DESCRIPTION
Amazon apparently keeps changing how the Hue Bridge discovery works. This results in plenty of errors for users with older Alexa devices or older Tasmota firmwares. 
Examples:
https://github.com/arendst/Tasmota/pull/11415
https://github.com/arendst/Tasmota/issues/9155
https://github.com/arendst/Tasmota/pull/9176
https://github.com/arendst/Tasmota/issues/5849
https://forum.creationx.de/forum/index.php?thread/550-sonoff-tasmota-mit-alexa-und-hue-emulation/

However the Alexa setup instructions at https://tasmota.github.io/docs/Alexa/ recommend to use Hue Bridge mode even if it is not necessary for multi-relay devices:
"Use Belkin WeMo for devices with a single relay or Hue Bridge for devices with one or more relays or for lights."
This is very misleading because it makes it sound like multi-relay devices will not expose all relays in Belkin WeMo Emulation, but actually both relays will show up as expected. I wasted hours because of this misleading documentation.

I propose changing that to: 
"Belkin WeMo has better compatibility with Alexa and is therefore recommended for devices with one or more relays. Hue Bridge can be used for devices with one or more relays and must be used for lights."#

Note: not sure about the "must be used for lights". Maybe "must be used for dimmable lights" is more accurate?

Also the SetOption109 1 should probably be documented on that page if it is still current.

### REQUESTED INFORMATION
Software Information Irrelevant since this a documenation change request.

### TO REPRODUCE
Software Information Irrelevant since this a documenation change request.

### EXPECTED BEHAVIOUR
Software Information Irrelevant since this a documenation change request.

### SCREENSHOTS
Software Information Irrelevant since this a documenation change request.

Jason2866, 2021-04-05:
Belkin emulation was enhanced not so long ago to support more than one relais.
So Hue Emulation was the only solution for more relais.
HueEmulation is the only solution for lights control (dimming and color control)

RolfWojtech, 2021-04-05:
#11415 also fixed my issue and now both Belkon and Hue Bridge mode work.
But with the information from @Jason2866 that the multi-relais switch was recently added (great job btw!) it probably makes sense to update the documentation nontheless to make it clear to people that they now have options.

Jason2866, 2021-04-05:
The docu is open source too. You are invited to do.

ascillato2, 2021-04-07:
> it probably makes sense to update the documentation nontheless to make it clear to people that they now have options.

You are right. Please, help us and edit the docs. Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-04-07 by ascillato2