arendst/Tasmota #11596: Tasmota32: ds18b20 not recognized

danilobar1, 2021-04-05:
### PROBLEM DESCRIPTION
I flashed a tasmota32.bin firmware to a ESP32D0WDQ6 chip (AzDelivery ESP32 NodeMCU Board).
Than I connected ds18b20 as described in https://tasmota.github.io/docs/DS18x20/.
I configured web interface to read temp data from GPIO34. After restart no data is presented.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): AzDelivery ESP32 NodeMCU Board (ESP32D0WDQ6 chip) + DS18B20 sensor (+ other sensors correctly recognized)
- [x] Tasmota binary firmware version number used: 9.3.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: esptool.py
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  15:32:31.939 MQT: stat/turtle_acquarium_monitor/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
15:32:32.153 MQT: stat/turtle_acquarium_monitor/RESULT = {"Module":{"1":"ESP32-DevKit"}}
15:32:32.410 MQT: stat/turtle_acquarium_monitor/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"258":"Relay_i3"},"GPIO26":{"256":"Relay_i1"},"GPIO27":{"257":"Relay_i2"},"GPIO28":{"0":"None"}}
15:32:32.423 MQT: stat/turtle_acquarium_monitor/RESULT = {"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"1184":"DHT11"},"GPIO33":{"259":"Relay_i4"},"GPIO34":{"1312":"DS18x20"},"GPIO35":{"4864":"ADC Range1"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
15:36:10.928 CMD: Status 0
15:36:10.940 MQT: stat/turtle_acquarium_monitor/STATUS = {"Status":{"Module":1,"DeviceName":"Turtle Aquarium Monitor","FriendlyName":["Pump","Light1","Light2","Moon Light"],"Topic":"turtle_acquarium_monitor","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
15:36:10.950 MQT: stat/turtle_acquarium_monitor/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:31:54","StartupUTC":"2021-04-05T14:04:16","Sleep":50,"CfgHolder":4617,"BootCount":11,"BCResetTime":"2021-04-05T12:01:06","SaveCount":35}}
15:36:10.958 MQT: stat/turtle_acquarium_monitor/STATUS2 = {"StatusFWR":{"Version":"9.3.1(tasmota)","BuildDateTime":"2021-02-23T12:19:31","Core":"1_0_5_RC6","SDK":"v3.3.4-402-gcd59d107b","CpuFrequency":80,"Hardware":"ESP32","CR":"468/699"}}
15:36:10.967 MQT: stat/turtle_acquarium_monitor/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["B1WiFi",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
15:36:10.980 MQT: stat/turtle_acquarium_monitor/STATUS4 = {"StatusMEM":{"ProgramSize":1148,"Free":1984,"Heap":231,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","00140001","000000CF","010013C0","40000981","00004000","00200000","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,12,20,21,24,26,27,29,50","Sensors":"1,2,3,5,6"}}
15:36:10.990 MQT: stat/turtle_acquarium_monitor/STATUS5 = {"StatusNET":{"Hostname":"turtle_acquarium_monitor-4884","IPAddress":"192.168.1.35","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.252","Mac":"30:AE:A4:F5:93:14","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
15:36:10.995 MQT: stat/turtle_acquarium_monitor/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.252","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F59314","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
15:36:11.004 MQT: stat/turtle_acquarium_monitor/STATUS7 = {"StatusTIM":{"UTC":"2021-04-05T14:36:10","Local":"2021-04-05T15:36:10","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:18","Sunset":"19:26"}}
15:36:11.021 MQT: stat/turtle_acquarium_monitor/STATUS10 = {"StatusSNS":{"Time":"2021-04-05T15:36:11","ANALOG":{"Range1":9},"DHT11":{"Temperature":19.0,"Humidity":23.0,"DewPoint":-2.7},"TempUnit":"C"}}
15:36:11.030 MQT: stat/turtle_acquarium_monitor/STATUS11 = {"StatusSTS":{"Time":"2021-04-05T15:36:11","Uptime":"0T00:31:55","UptimeSec":1915,"Heap":229,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"B1WiFi","BSSId":"18:0F:76:7A:9C:65","Channel":11,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  15:37:17.387 HTP: Main Menu
15:37:18.998 HTP: Console
15:37:19.884 DHT: Timeout waiting for start signal low pulse
15:37:29.860 MQT: tele/turtle_acquarium_monitor/STATE = {"Time":"2021-04-05T15:37:29","Uptime":"0T00:33:13","UptimeSec":1993,"Heap":235,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"B1WiFi","BSSId":"18:0F:76:7A:9C:65","Channel":11,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:05"}}
15:37:29.900 MQT: tele/turtle_acquarium_monitor/SENSOR = {"Time":"2021-04-05T15:37:29","ANALOG":{"Range1":9},"DHT11":{"Temperature":19.0,"Humidity":23.0,"DewPoint":-2.7},"TempUnit":"C"}
15:37:32.082 WIF: Checking connection...
15:37:52.075 WIF: Checking connection...
15:37:59.846 MQT: tele/turtle_acquarium_monitor/STATE = {"Time":"2021-04-05T15:37:59","Uptime":"0T00:33:43","UptimeSec":2023,"Heap":235,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"B1WiFi","BSSId":"18:0F:76:7A:9C:65","Channel":11,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:05"}}
15:37:59.886 MQT: tele/turtle_acquarium_monitor/SENSOR = {"Time":"2021-04-05T15:37:59","ANALOG":{"Range1":9},"DHT11":{"Temperature":19.0,"Humidity":23.0,"DewPoint":-2.7},"TempUnit":"C"}
15:38:01.869 DHT: Timeout waiting for start signal high pulse
15:38:12.059 WIF: Checking connection...
15:38:29.862 MQT: tele/turtle_acquarium_monitor/STATE = {"Time":"2021-04-05T15:38:29","Uptime":"0T00:34:13","UptimeSec":2053,"Heap":235,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"B1WiFi","BSSId":"18:0F:76:7A:9C:65","Channel":11,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:05"}}
15:38:29.902 MQT: tele/turtle_acquarium_monitor/SENSOR = {"Time":"2021-04-05T15:38:29","ANALOG":{"Range1":9},"DHT11":{"Temperature":20.0,"Humidity":23.0,"DewPoint":-1.8},"TempUnit":"C"}
15:38:31.880 DHT: Timeout waiting for start signal high pulse
15:38:32.057 WIF: Checking connection...

```

### TO REPRODUCE
Connect a ds18b20 sensor and configure a GPIO to use it: nothing seems to be happening. 
It also doesn't show null values either but it seems that the sensor isn't really activated.
The same thing happens with both tasmota32.bin and tasmota32-sensor.bin 

### EXPECTED BEHAVIOUR
To read sensor temperature value from web console.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]
[image]
This is the sensor used:
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-04-05:
I see no mention of you satisfying the requirement of having a pullup resistor. My experience is that SetOption74 did not work, at not on the pins I tried using.

danilobar1, 2021-04-05:
> I see no mention of you satisfying the requirement of having a pullup resistor. My experience is that SetOption74 did not work, at not on the pins I tried using.

I followed everything described at the link https://tasmota.github.io/docs/DS18x20/
The sensor has the black cable connected to the ground, the red connected to 3.3v, the yellow connected to the gpio and I inserted a 4.7kO resistor between the gpio and vcc.
[image]

danilobar1, 2021-04-05:
After found this comment on an old similar issue: https://github.com/arendst/Tasmota/issues/7984#issuecomment-602458166, I tried to change the gpio pin from 34 to 4, and IT WORKS!!!

In the documentation there is no reference to which pin to use, I am not clear why specifying the GPIO4 instead of 34 it works.
the SetOption74 is OFF.
This issue can also be closed, but I would be grateful to any of you if you could document the reason for this behavior.
Thank you all.

arendst, 2021-04-05:
As can be seen on the gui module configuration page there is a prefix for every gpio. Your gpio34 has a prefix of IA. This means the esp32 hardware does support Analog in and Digital Out. Your ds18b20 sensor needs a digital IO gpio so that is why it doesn't work on gpio34. 

esp32 gpios starting at 32 are special gpios. You'll need some background information to connect anything.

closed 2021-04-05 by arendst

danilobar1, 2021-04-05:
@arendst thank you so much for explaining why.

corbolais, 2023-01-28:
@arendst 

> As can be seen on the gui module configuration page

Could you please provide a hint where this page might be found? That would be very helpful.

Thank you.

sfromis, 2023-01-28:
Click "Configuration" and "Configure Module"

corbolais, 2023-01-29:
> Click "Configuration" and "Configure Module"

Thanks for replying.

But there is only modules listed, no GPIOs. And no "AI" either.

sfromis, 2023-01-29:
Then you are likely using a template
"Configure Template"

corbolais, 2023-01-29:
> Then you are likely using a template "Configure Template"

Correct, I'm using a local template. The GPIOs listed show no "AI". What am I missing here?

sfromis, 2023-01-29:
Not sure what you are looking for. `AI` is not a thing, and while the pins from 32 and up have `IA`, these are the pins you *already* were told not to use. That indicator is only shown on "Configure Module", where the pins show up if configured as "User" on "Configure Template" (where the indicator is not shown).

corbolais, 2023-01-29:
> Not sure what you are looking for. `AI` is not a thing, 

Ok.

> and while the pins from 32 and up have `IA`, these are the pins you _already_ were told not to use. 

Sure I did use GPIO33 for an AM2301. It's a workin'. I must have missed why those GPIOs are special and to be skipped. Could you reiterate it once more, please?

> That indicator is only shown on "Configure Module", where the pins show up if configured as "User" on "Configure Template" (where the indicator is not shown).

This was the bit I was missing. Thank you.

cheers

sfromis, 2023-01-29:
The "why" is due to how the hardware is made, but I mistyped the specific limitation, which is for gpio34 and up, not from 32/33 which have `AO` instead of `IA`.

kub3let, 2024-07-28:
Since this thread pops up as first google result  for `Tasmota DS18B20 not working`

If Tasmota does not detect your DS18B20, make sure you are using a `4.7kΩ` pullup resistor (measure it with a multimeter)

My resistor box was wrongly sorted so while I thought it was 4.7k ohm it wasn't...

ffrediani, 2025-04-04:
I have the very same bloody problem and have no hint what why it never shows it on the main Dashboard when you enabled the pin for it. DS18B20 is wired correctly, with 4.7k ohm resistor betwen Vcc and Data and it never shows on the main dashboard with no explanation.

Even if the sensor is not connected yet it should still show something like "null" or zero until it gets connected, but nothing. And this seems to be only with DS18B20

ffrediani, 2025-04-04:
> Since this thread pops up as first google result for `Tasmota DS18B20 not working`
> 
> If Tasmota does not detect your DS18B20, make sure you are using a `4.7kΩ` pullup resistor (measure it with a multimeter)
> 
> My resistor box was wrongly sorted so while I thought it was 4.7k ohm it wasn't...

Is there a exact PIN to use DS18B20 ? I tried on several and it never works.
I am using a ESP32-C3 Zero

lalo-uy, 2025-04-05:
4k7 is the value showed at the datasheet for 5v example. To have the same
current on a 3.3V you may use 3k3. Can go as low as 1K for better noise
immunity if the wires are long.