arendst/Tasmota #11599: Another issue with periodically wrong SML values in counter reading

matzefisi, 2021-04-05:
### PROBLEM DESCRIPTION
After fixing one of my issues with your solution in issue #10007, I still have some times wrong values in the continues increasing counter readings (in German: Zählerstände). A few times a day the readings show a 0 or numbers crazy high (see screenshot from Grafana attached) I assume here also a problem with the differen value types in the SML message.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodeMCU 0.8
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: Platform IO
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
19:27:01.129 MQT: tasmota/smartmeter/stat/RESULT = {"NAME":"","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
19:27:01.346 MQT: tasmota/smartmeter/stat/RESULT = {"Module":{"0":""}}
19:27:01.599 MQT: tasmota/smartmeter/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
19:27:06.612 MQT: tasmota/smartmeter/tele/STATE = {"Time":"2021-04-05T19:27:06","Uptime":"4T22:28:49","UptimeSec":426529,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":29,"MqttCount":3,"Wifi":{"AP":1,"SSId":"xxxxxxx","BSSId":"DC:39:6F:E7:CF:3E","Channel":6,"RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:03"}}
19:27:06.622 MQT: tasmota/smartmeter/tele/SENSOR = {"Time":"2021-04-05T19:27:06","Haus":{"Power_curr":602.66,"Verbrauch":339.38,"Einspeisung":613.79,"Meter_number":"090145425a01000b9048"},"Heizung":{"Power_curr":862.90,"Gesamt":3221.60,"Nachttarif":950.86,"Tagtarif":2270.74,"Meter_number":"0901454d480000865e5a"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
19:29:11.359 CMD: Status 0
19:29:11.367 MQT: tasmota/smartmeter/stat/STATUS = {"Status":{"Module":0,"DeviceName":"Smartmeter","FriendlyName":["Smartmeter"],"Topic":"smartmeter","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
19:29:11.376 MQT: tasmota/smartmeter/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"4T22:30:54","StartupUTC":"2021-03-31T19:58:17","Sleep":50,"CfgHolder":4617,"BootCount":134,"BCResetTime":"2020-03-27T11:14:33","SaveCount":373,"SaveAddress":"F9000"}}
19:29:11.383 MQT: tasmota/smartmeter/stat/STATUS2 = {"StatusFWR":{"Version":"9.3.1.2(tasmota)","BuildDateTime":"2021-03-31T21:45:32","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"455/699"}}
19:29:11.389 MQT: tasmota/smartmeter/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["AuMMS14o","AuMMS14o_Gast"],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
19:29:11.401 MQT: tasmota/smartmeter/stat/STATUS4 = {"StatusMEM":{"ProgramSize":622,"Free":380,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","87DAC787","043E8001","000000CF","010013C0","C000F989","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6,53"}}
19:29:11.408 MQT: tasmota/smartmeter/stat/STATUS5 = {"StatusNET":{"Hostname":"smartmeter-4378","IPAddress":"192.168.2.201","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"60:01:94:3B:51:1A","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
19:29:11.415 MQT: tasmota/smartmeter/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.5","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3B511A","MqttUser":"DVES_USER","MqttCount":3,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
19:29:11.424 MQT: tasmota/smartmeter/stat/STATUS7 = {"StatusTIM":{"UTC":"2021-04-05T18:29:11","Local":"2021-04-05T19:29:11","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:18","Sunset":"19:26"}}
19:29:11.433 MQT: tasmota/smartmeter/stat/STATUS10 = {"StatusSNS":{"Time":"2021-04-05T19:29:11","Haus":{"Power_curr":602.90,"Verbrauch":339.40,"Einspeisung":613.79,"Meter_number":"090145425a01000b9048"},"Heizung":{"Power_curr":864.80,"Gesamt":3221.63,"Nachttarif":950.86,"Tagtarif":2270.77,"Meter_number":"0901454d480000865e5a"}}}
19:29:11.441 MQT: tasmota/smartmeter/stat/STATUS11 = {"StatusSTS":{"Time":"2021-04-05T19:29:11","Uptime":"4T22:30:54","UptimeSec":426654,"Heap":17,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":3,"Wifi":{"AP":1,"SSId":"xxxxxxxxxx","BSSId":"DC:39:6F:E7:CF:3E","Channel":6,"RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:03"}}}
19:29:16.584 MQT: tasmota/smartmeter/tele/STATE = {"Time":"2021-04-05T19:29:16","Uptime":"4T22:30:59","UptimeSec":426659,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":3,"Wifi":{"AP":1,"SSId":"xxxxxxxxxx","BSSId":"DC:39:6F:E7:CF:3E","Channel":6,"RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:03"}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
This is quite hard since it occurs only a few times a day, but I will continue to troubleshoot on my installation.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Counter readings go continiuosly higher and not jump to 0 or event to 100.000.000

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
My SML script:
```
>D

>B
=>sensor53 r
>M 2
+1,12,s,0,9600,Haus
+2,13,s,0,9600,Heizung

1,77070100100700ff@1,Aktuell,W,Power_curr,2
1,77070100010800FF@100000000,Zählerstand Verb.,kWh,Verbrauch,2
1,77070100020800FF@100000000,Zählerstand Einsp.,kWh,Einspeisung,2
1,77070100000009ff@#,Meter Nr,,Meter_number,0
2,=h==================
2,77070100100700ff@1,Aktuell,W,Power_curr,2
2,77070100010800FF@1000,Zählerstand Verb.,kWh,Gesamt,2
2,77070100010801FF@1000,Zählerstand Nachttarif,kWh,Nachttarif,2
2,77070100010802FF@1000,Zählerstand Tagtarif,kWh,Tagtarif,2
2,77070100000009ff@#,Meter Nr,,Meter_number,0
#
```
Reader 1 is: eBZ GmbH - DD3 2R06 DTA SMZ1 
Reader 2 is: EMH ED300L

Problem occurs with both smartmeters
**(Please, remember to close the issue when the problem has been addressed)**

matzefisi, 2021-04-05:
A first idea is, that the type-length field is not beeing considered correctly. I had a similar issue in my fhem perl scripts (https://github.com/matzefisi/fhem-70_SMLUSB/blob/master/70_SMLUSB.pm)
-
https://www.bsi.bund.de/SharedDocs/Downloads/DE/BSI/Publikationen/TechnischeRichtlinien/TR03109/TR-03109-1_Anlage_Feinspezifikation_Drahtgebundene_LMN-Schnittstelle_Teilb.pdf?__blob=publicationFile&v=1 page 42

gemu2015, 2021-04-06:
this looks exactly like hardware glitches. did you try with median filter ?

\>M 2
+1,12,s,16,9600,Haus
+2,13,s,16,9600,Heizung

if it was a programming error in the decoder it would occur always and not occasionally.

also did you check using hardware serial ?

when using 2 software serial lines there might be an interference between the 2 irq routines.

matzefisi, 2021-04-06:
Hi @gemu2015 , thanks for the feedback. Switched the port 12 to 3 (according to PIN layout of nodemcu 0.9 GPIO 3 and 13 are serial ports) and will check the results.  In the console it said "19:27:02.922 SNS: Hardware Serial" so I assume, that this is a confirmation that it runs now with hardware serial port. If the problem still occurs I will also check the median filter and report back here.

I had similar issues in the past with my perl scripts, which I do not use anymore (tasmota + sml is more relaiable ;-) ), and there the meter occasionally switched  the value types (signed / unsigned and so on)

matzefisi, 2021-04-07:
Hi again,
it looks like that the switch to the hardware port fixed already my issue. I will monitor further, but until now, no further glitches appeard. 
Thanks for the help!

closed 2021-04-07 by matzefisi