arendst/Tasmota #11606: Diolamp A609WIFI brightness lowered after tasmota upgrade

costakis79, 2021-04-06:
### PROBLEM DESCRIPTION
I have some Diolamp A609WIFI which I have successfully upgraded to Tasmota 9.3.1 using tuya-convert.
After upgrading 6 of them (and was on my way to the 7th) I noticed that the lamp, before upgrading (still on Tuya), was much much brighter on the white channel. I think that using using then the RGB leds are turned of in Tasmota.

I am using so37 30

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Diolamp A609WIFI
- [x] Tasmota binary firmware version number used: 9.3.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: tuya-convert
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
08:18:44.472 RSL: stat/tasmota_06E378/RESULT = {"NAME":"Diolamp A609WIFI","GPIO":[0,0,0,0,4032,0,0,0,0,0,4064,0,0,0],"FLAG":0,"BASE":18}
08:18:44.679 RSL: stat/tasmota_06E378/RESULT = {"Module":{"0":"Diolamp A609WIFI"}}
08:18:44.886 RSL: stat/tasmota_06E378/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"4032":"SM2135 Clk"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"4064":"SM2135 Dat1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
08:19:55.453 RSL: stat/tasmota_06E378/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_06E378","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":5,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
08:19:55.483 RSL: stat/tasmota_06E378/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:04:51","StartupUTC":"2021-04-06T07:15:04","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"2021-04-06T07:37:28","SaveCount":34,"SaveAddress":"FA000"}}
08:19:55.517 RSL: stat/tasmota_06E378/STATUS2 = {"StatusFWR":{"Version":"9.3.1(tasmota)","BuildDateTime":"2021-03-09T16:12:28","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"395/699"}}
08:19:55.539 RSL: stat/tasmota_06E378/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["50101Devices",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C800011E060000005A0A000000000000","00000080","00006000","00000000"]}}
08:19:55.568 RSL: stat/tasmota_06E378/STATUS4 = {"StatusMEM":{"ProgramSize":593,"Free":408,"Heap":27,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
08:19:55.601 RSL: stat/tasmota_06E378/STATUS5 = {"StatusNET":{"Hostname":"tasmota_06E378-0888","IPAddress":"10.10.200.45","Gateway":"10.10.200.1","Subnetmask":"255.255.255.0","DNSServer":"10.10.200.1","Mac":"24:62:AB:06:E3:78","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
08:19:55.623 RSL: stat/tasmota_06E378/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_06E378","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
08:19:55.647 RSL: stat/tasmota_06E378/STATUS7 = {"StatusTIM":{"UTC":"2021-04-06T07:19:55","Local":"2021-04-06T08:19:55","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:16","Sunset":"19:27"}}
08:19:55.665 RSL: stat/tasmota_06E378/STATUS10 = {"StatusSNS":{"Time":"2021-04-06T08:19:55"}}
08:19:55.675 RSL: stat/tasmota_06E378/STATUS11 = {"StatusSTS":{"Time":"2021-04-06T08:19:55","Uptime":"0T00:04:51","UptimeSec":291,"Heap":27,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":100,"Color":"000000FF00","HSBColor":"0,0,0","White":100,"CT":153,"Channel":[0,0,0,100,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"50101Devices","BSSId":"48:8F:5A:AA:6E:07","Channel":1,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

08:22:22.602 RSL: stat/tasmota_06E378/RESULT = {"WebLog":4}
08:22:26.078 HTP: Main Menu
08:22:26.942 CFG: Saved to flash at F9, Count 35, Bytes 4096
08:22:27.204 SRC: WebGui from 192.168.2.101
08:22:27.207 CMD: Group 0, Index 1, Command "DIMMER", Data "59"
08:22:27.213 RSL: stat/tasmota_06E378/RESULT = {"POWER":"ON","Dimmer":59,"Color":"0000009600","HSBColor":"0,0,0","White":59,"CT":153,"Channel":[0,0,0,59,0]}
08:22:27.611 SRC: WebGui from 192.168.2.101
08:22:27.614 CMD: Group 0, Index 1, Command "DIMMER", Data "100"
08:22:27.620 RSL: stat/tasmota_06E378/RESULT = {"POWER":"ON","Dimmer":100,"Color":"000000FF00","HSBColor":"0,0,0","White":100,"CT":153,"Channel":[0,0,0,100,0]}

```

### TO REPRODUCE
Just try a lamp before upgrading it, you can't event look at it.

### EXPECTED BEHAVIOUR
 After upgrade the brightness is lowered

### SCREENSHOTS

### ADDITIONAL CONTEXT

arendst, 2021-04-06:
Play with the indexes in GPIO14 ``SM2135 Dat`` 1, 3 or 5. This increases currents but make sure your bulb does not get too hot!!!

costakis79, 2021-04-06:
Indeed if I change the DAT then the light becomes brighter, but I lose RGB color selection

arendst, 2021-04-07:
In that case I expect it draws too much current so you'll need to select a different option.

costakis79, 2021-04-07:
Like what ? Can you help ? I cab give you remote access to a lamp if you like.

ryrzy, 2021-04-20:
+1 
rgb and white should be separated settings with current

Giorgakis, 2021-04-21:
+1 

I have the same problem. When using DAT=1 the brightness is very low. When using the other values I cannot use the RGB colors.

closed 2021-05-21 by github-actions[bot]