arendst/Tasmota #11623: Tasmota generates Github issues with @april in the description

april, 2021-04-07:
### PROBLEM DESCRIPTION
Tasmota has now generated multiple tickets with @april in the description. It usually looks like:

> * [x]  Self-compiled @april 6 2021 17h45

If this could be changed to be something like "at april 6" or "@ april 6", that would be great. It would really help my notification spam.

Thanks so much!

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [x] Self-compiled @april 6 2021 17h45
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
N/A

### EXPECTED BEHAVIOUR
"at april 6" or "@ april 6" would be a lot better for my notification spam.

### SCREENSHOTS
N/A

### ADDITIONAL CONTEXT
N/A

**(Please, remember to close the issue when the problem has been addressed)**

april, 2021-04-07:
The original ticket that caused this issue:
https://github.com/arendst/Tasmota/discussions/11621

sfromis, 2021-04-07:
I can fully understand how annoying that can be. However, the "original ticket" you link has some user choosing to enter a date with month name after an @ sign, not something that can be avoided in the template used.

april, 2021-04-07:
Oh, got'cha. I had assumed that since there was both #11621 and #11613, that they were being generated programmatically.

If that's not the case, no big deal at all, and I can just unsubscribe from the few random issues.

Thanks for getting back to me so quickly! I really appreciate it.

closed 2021-04-07 by april

ascillato2, 2021-04-07:
Deleted the incorrectly used @ that the user have added into the description of his issue #11621

You should not receive any further notification. Sorry for the inconvenience.