arendst/Tasmota #11627: Rule: Subcribe a multisensor topic

GVollaro, 2021-04-08:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I have a device that publishes the following topic:
asp/termo_display/state/sensors => {"Temp":"23.08","Hum":"30.80","Pres":"980.65","PresSL":"1025.88"}

I defined the following rule to generate MQTT events on publication:
Rule1 on mqtt#connected do backlog 
Subscribe DisplayTemp, hasp/termo_display/state/sensors, Temp;
Subscribe DisplayHum, hasp/termo_display/state/sensors, Hum; 
Subscribe DisplayPres, hasp/termo_display/state/sensors, PresSL endon

Another rule allows me to publish the intercepted values ​​on an OpenHASP display:
Rule2 on Event#DisplayTemp do publish hasp/termo_display/command/p[1].b[20].txt %value% endon
on Event#DisplayHum do publish hasp/termo_display/command/p[1].b[21].txt %value% endon
on Event#DisplayPres do publish hasp/termo_display/command/p[1].b[23].txt %value% endon

The problem encountered is that only the first subscribed parameter generates the publish event, while the next two are ignored.
Swapping the order of definition in the first rule also changes the published parameter which is always the first in the order of definition

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodeMCU8266
- [x] Tasmota binary firmware version number used: Tasmota 9.2.0.4
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
 
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
CMD: Backlog Rule1; Rule2; Rule3
14:50:32.943 RUL: Rule1 on mqtt#connected do backlog  Subscribe DisplayTemp, hasp/termo_display/state/sensors, Temp; Subscribe DisplayHum, ha...
14:50:32.948 RUL: Rule1 +Subscribe AcquaRiscSet, hasp/termo_display/state/p2b5, text; Subscribe AcquaSanSet, hasp/termo_display/state/p2b3, t...
14:50:32.965 MQT: stat/OpenTherm1/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Length":751,"Free":269,"Rules":"on mqtt#connected do backlog  Subscribe DisplayTemp, hasp/termo_display/state/sensors, Temp; Subscribe DisplayHum, hasp/termo_display/state/sensors, Hum;  Subscribe DisplayPres, hasp/termo_display/state/sensors, PresSL; Subscribe DisplayPag, hasp/termo_display/state/page; Subscribe DisplayDim, hasp/termo_display/state/dim; Subscribe PulsanteAcc, hasp/termo_display/state/p1b14, val; Subscribe CursoreLum, hasp/termo_display/state/p1b15, val; Subscribe TermostatoSet, hasp/termo_display/state/p2b1, text; Subscr..."}
14:50:33.175 RUL: Rule2 on Event#DisplayTemp do publish hasp/termo_display/command/p[1].b[20].txt %value% endon on Event#DisplayHum do publis...
14:50:33.180 RUL: Rule2 +Event#AcquaRiscSet do ot_tboiler %value% endon on Event#AcquaSanSet do ot_twater %value% endon on Event#DisplayPag d...
14:50:33.196 MQT: stat/OpenTherm1/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Length":785,"Free":250,"Rules":"on Event#DisplayTemp do publish hasp/termo_display/command/p[1].b[20].txt %value% endon on Event#DisplayHum do publish hasp/termo_display/command/p[1].b[21].txt %value% endon on Event#DisplayPres do publish hasp/termo_display/command/p[1].b[23].txt %value% endon on Event#PulsanteAcc do POWER1 %value% endon on Event#CursoreLum do backlog var6=100-%value%; event setvar6=1 endon  on Event#setvar6 do publish hasp/termo_display/command/dim %var6% endon on Event#TermostatoSet do mem10 %value% endon on Event#Acqua..."}
14:50:33.006 MQT: stat/OpenTherm1/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
MQT: stat/OpenTherm1/STATUS = {"Status":{"Module":0,"DeviceName":"OpenTherm1","FriendlyName":["OpenTherm1"],"Topic":"OpenTherm1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

I expect that in response to the subscription of the three parameters Temp, Hum, PresSL, the three events od rule2
are activated

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

lalo-uy, 2021-04-08:
I have come to this problem before.
To have 2 diferent values from a json msg you need to use 2 separated rules.

GVollaro, 2021-04-08:
Thanks @lalo-uy for your answer but, if I understood what you are saying, I think I have already done so by defining in rule 1 an MQTT subsribe event for each of the parameters contained in the Json msg

sfromis, 2021-04-08:
You are doing all 3 subscribes in just one rule. Try following the suggestion about separate rules, to see if you get better results then.

GVollaro, 2021-04-08:
Even if splitting the subscriptions into different rules does not make much sense to me as if I had 4 parameters the 3 rules of tasmota would not be enough for me, I still tried to divide the subscription of the 3 parameters between rule1, rule2 and rule3 but the result was that only for the parameter defined in rule2 has the publication event been activated, for the other two nothing

sfromis, 2021-04-08:
You can still have several rules in one rule set. One rule is from `on` to `endon`. Why it still does not work for you is another matter.

closed 2021-04-08 by ascillato2