arendst/Tasmota #11628: software restarting/crashing? un commanded relay switch off's

scroatamus, 2021-04-08:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
NodeMCU with 3 temperature sensors & 4 relays to semi automate beer making.
relays control pumps valves & heating elements. All has been working fine for year or so.
Recently added PZEM0004 to monitor energy use. ( also working fine independently)

During the boil cycle using 2 relays & one temp sensor, at a certain point the system malfunctions, and commanded relay 3 on switches are immediately followed with un-commanded relay switch off to all four.

Removing the PZEM00XX TX configuration, restores normal operation.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
6:09:56 CMD: Backlog Template; Module; GPIO 255
16:09:56 SRC: WebConsole from 192.168.1.88
16:09:56 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
16:09:57 SRC: Backlog
16:09:57 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
16:09:57 MQT: stat/nodemcu2/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
16:09:57 SRC: Backlog
16:09:57 CMD: Group 0, Index 1, Command "MODULE", Data ""
16:09:57 MQT: stat/nodemcu2/RESULT = {"Module":{"18":"Generic"}}
16:09:57 SRC: Backlog
16:09:57 CMD: Group 0, Index 1, Command "GPIO", Data "255"

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
16:11:19 CMD: Status 0
16:11:19 SRC: WebConsole from 192.168.1.88
16:11:19 CMD: Group 0, Index 1, Command "STATUS", Data "0"
16:11:19 MQT: stat/nodemcu2/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2","Tasmota3","Tasmota4"],"Topic":"nodemcu2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:11:19 MQT: stat/nodemcu2/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:10:45","StartupUTC":"2021-04-08T15:00:34","Sleep":50,"CfgHolder":4617,"BootCount":25,"BCResetTime":"2021-02-02T10:41:34","SaveCount":916,"SaveAddress":"F8000"}}
16:11:19 MQT: stat/nodemcu2/STATUS2 = {"StatusFWR":{"Version":"9.2.0(sensors)","BuildDateTime":"2020-12-21T15:04:56","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"359/699"}}
16:11:19 MQT: stat/nodemcu2/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BTHub5-8HPW",""],"TelePeriod":15,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00006000","00000000"]}}
16:11:19 MQT: stat/nodemcu2/STATUS4 = {"StatusMEM":{"ProgramSize":638,"Free":364,"Heap":18,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164068","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7FFBFCF","01DA9BC4","64367CC7","00084052","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,51,52,55,56,58,59,62,64,66,67,74"}}
16:11:19 MQT: stat/nodemcu2/STATUS5 = {"StatusNET":{"Hostname":"nodemcu2-7003","IPAddress":"192.168.1.250","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"CC:50:E3:35:1B:5B","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
16:11:19 MQT: stat/nodemcu2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.119","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_351B5B","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
16:11:19 MQT: stat/nodemcu2/STATUS7 = {"StatusTIM":{"UTC":"2021-04-08T15:11:19","Local":"2021-04-08T16:11:19","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:12","Sunset":"19:30"}}
16:11:19 MQT: stat/nodemcu2/STATUS10 = {"StatusSNS":{"Time":"2021-04-08T16:11:19","DS18B20-1":{"Id":"030797794C61","Temperature":23.8},"DS18B20-2":{"Id":"031997792B29","Temperature":10.0},"DS18B20-3":{"Id":"03209779D81E","Temperature":98.7},"TempUnit":"C"}}
16:11:19 MQT: stat/nodemcu2/STATUS11 = {"StatusSTS":{"Time":"2021-04-08T16:11:19","Uptime":"0T00:10:45","UptimeSec":645,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"BTHub5-8HPW","BSSId":"78:DD:12:EA:85:40","Channel":6,"RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:09"}}}
16:11:21 MQT: tele/nodemcu2/STATE = {"Time":"2021-04-08T16:11:21","Uptime":"0T00:10:47","UptimeSec":647,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"BTHub5-8HPW","BSSId":"78:DD:12:EA:85:40","Channel":6,"RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:09"}}
16:11:21 MQT: tele/nodemcu2/SENSOR = {"Time":"2021-04-08T16:11:21","DS18B20-1":{"Id":"030797794C61","Temperature":23.7},"DS18B20-2":{"Id":"031997792B29","Temperature":10.0},"DS18B20-3":{"Id":"03209779D81E","Temperature":98.7},"TempUnit":"C"}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
16:15:18 MQT: tele/nodemcu2/STATE = {"Time":"2021-04-08T16:15:18","Uptime":"0T00:01:56","UptimeSec":116,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"BTHub5-8HPW","BSSId":"78:DD:12:EA:85:40","Channel":6,"RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:05"}}
16:15:18 MQT: tele/nodemcu2/SENSOR = {"Time":"2021-04-08T16:15:18","ENERGY":{"TotalStartTime":"2021-02-02T10:41:34","Total":510.761,"Yesterday":6.601,"Today":14.063,"Period":3,"Power":620,"ApparentPower":696,"ReactivePower":315,"Factor":0.89,"Frequency":50,"Voltage":248,"Current":2.805},"DS18B20-1":{"Id":"030797794C61","Temperature":23.3},"DS18B20-2":{"Id":"031997792B29","Temperature":10.1},"DS18B20-3":{"Id":"03209779D81E","Temperature":98.4},"TempUnit":"C"}
16:15:19 DMP: 01 04 14 09 B0 0A F5 00 00 18 39 00 00 54 7F 00 08 01 F4 00 59 00 00 90 2B
[... 22 lines cut ...]
16:15:26 CFG: Saved to flash at F5, Count 943, Bytes 4096
16:15:26 DMP: 01 04 14 09 AF 0D 78 00 00 1C 51 00 00 54 80 00 08 01 F4 00 55 00 00 59 28
16:15:27 DMP: 01 04 14 09 AF 0D 78 00 00 1C 51 00 00 54 80 00 08 01 F4 00 55 00 00 59 28
16:15:27 WIF: Checking connection...
16:15:28 DMP: 01 04 14 09 A9 14 2F 00 00 31 4A 00 00 54 81 00 08 01 F4 00 63 00 00 6C 73
16:15:29 DMP: 01 04 14 09 B3 0A F1 00 00 18 35 00 00 54 81 00 08 01 F4 00 59 00 00 CF 06
16:15:30 DMP: 01 04 14 09 B4 0A EF 00 00 18 31 00 00 54 81 00 08 01 F4 00 59 00 00 1D 50

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Add PZEM0004 to configuration & run a beer boil cycle.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
would be nice to run both concurrently.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-04-08:
Do you use hardware serial for your PZEM004T module? You did not provide your GPIO config.

scroatamus, 2021-04-08:
yes using hardware serial.

Sent from Mailspring (https://getmailspring.com/), the best free email app for work

Jason2866, 2021-04-08:
Without a precise complete description of your setup there is no help possible

scroatamus, 2021-04-08:
with the PZEM in the configuration, I command relay POWER2 ON 2 secs later all 4 relays are set to off.  that is not requested by me. remove PZEM operates as normal relays under my control.
I wondered what these weblogs are
16:15:25 NRG: GlobTemp 98.4
16:15:25 SRC: Overtemp

Jason2866, 2021-04-08:
We still need a crystal ball to help. Provide details. Whole config and logs.

scroatamus, 2021-04-08:
thought I had done that following the template..
did I miss bits in the copy paste?
bit hard to do when the screen fills so quickly.

Sent from Mailspring (https://getmailspring.com/), the best free email app for work

arendst, 2021-04-08:
If you add the pzem the enrgy monitoring gets precedent. It reacts on the system temperature and if too high it shutsdown all power. That is what happens here.

You'll jeed to raise the temperature threshold to a level that it win't turn off all power. Look for a setoption comand to do this.

scroatamus, 2021-04-08:
I had a feeling it was temperature related. 
I don't quite understand  what is system temperature?

scroatamus, 2021-04-08:
SetOption42 ?

arendst, 2021-04-08:
Yes that's it. Try to set it to 250. That's about the max in the 8-bit register.

closed 2021-04-08 by ascillato2