arendst/Tasmota #11643: [Zigbee Docs]

BusinessClaes, 2021-04-09:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
In Zigbee Docs T2Z, low-level commands, the second example has a "_" but it should have a "!"
ZbSend {"Device":"0x1234","Send":"0008_04/800A00"}
ZbSend {"Device":"0x1234","Send":"0008!04/800A00"}

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____SONOFF ZBBridge 6
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Get the docs updated :)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

tammo0, 2021-04-09:
Hi @BusinessClaes 
thanks for reporting but you can change it too :)
Please take a look at https://tasmota.github.io/docs/Contributing/

ascillato2, 2021-04-09:
Please, help us and do the editions in the docs. Thanks.

closed 2021-04-09 by ascillato2

BusinessClaes, 2021-04-10:
Thank you @tammo0 and @ascillato2, I should have read that myself at the contributing info 😅
I have now edited the docs and made a pull request.