arendst/Tasmota #11649: Telegram chat_id of incorrect size

jcmadrioso, 2021-04-09:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Command tmsend fails when chat_id is greater than int32 (it is truncated to 32bits value)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): nodemcu v1.0
- [x ] Tasmota binary firmware version number used: 9.3.1.2
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x ] Flashing tools used: esptool.py v2.8 (arduino SDK)
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
18:03:51.157 CMD: Backlog Template; Module; GPIO 255
18:03:51.186 MQT: stat/sensor3/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
18:03:51.415 MQT: stat/sensor3/RESULT = {"Module":{"0":"Generic"}}
18:03:51.671 MQT: stat/sensor3/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"6656":"TM1637 DIO"},"GPIO5":{"6624":"TM1637 CLK"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"0":"None"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
CMD: Backlog Rule1; Rule2; Rule3
18:07:04.223 MQT: stat/sensor3/RESULT = {"Rule1":{"State":"ON","Once":"ON","StopOnError":"OFF","Length":135,"Free":376,"Rules":"on time#minute>1350 do displaydimmer 10 endon on time#minute=480 do displaydimmer 99 endon on time#minute<479 do displaydimmer 10 endon"}}
18:07:04.451 MQT: stat/sensor3/RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":161,"Free":350,"Rules":"on rules#timer=1 do backlog tmsend %topic% Sigo vivo ; ruletimer1 %mem3% endon on system#init do backlog tmstate 1 ; ruletimer1 1 ; tmsend %topic% Resucito endon"}}
18:07:04.703 MQT: stat/sensor3/RESULT = {"Rule3":{"State":"ON","Once":"ON","StopOnError":"OFF","Length":41,"Free":470,"Rules":"on time#minute=539 do ruletimer1 60 endon"}}

```
- [x ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
CMD: status 0
18:07:44.992 MQT: stat/sensor3/STATUS = {"Status":{"Module":0,"DeviceName":"Sensor3","FriendlyName":["Sensor3"],"Topic":"sensor3","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
18:07:45.013 MQT: stat/sensor3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://nb30.madrioso.org/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T02:14:29","StartupUTC":"2021-04-09T13:53:16","Sleep":50,"CfgHolder":4617,"BootCount":12,"BCResetTime":"2021-04-08T21:27:02","SaveCount":34,"SaveAddress":"FD000"}}
18:07:45.037 MQT: stat/sensor3/STATUS2 = {"StatusFWR":{"Version":"9.3.1.2(tasmota)","BuildDateTime":"2021-04-09T15:51:11","Boot":31,"Core":"2_7_4","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"482/699"}}
18:07:45.059 MQT: stat/sensor3/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxxx1","xxxxxxx2"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
18:07:45.087 MQT: stat/sensor3/STATUS4 = {"StatusMEM":{"ProgramSize":789,"Free":1256,"Heap":13,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640C8","FlashFrequency":40,"FlashMode":2,"Features":["00000809","8FFAC787","003683A0","000004CF","010013C0","C000F981","0000C024","40001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,13,16,18,19,20,21,22,24,26,27,29,30,35,37,40,45","Sensors":"1,2,3,4,5,6,9"}}
18:07:45.112 MQT: stat/sensor3/STATUS5 = {"StatusNET":{"Hostname":"sensor3-0857","IPAddress":"192.168.1.113","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"80.58.61.250","Mac":"60:01:94:xx:xx:xx","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
18:07:45.135 MQT: stat/sensor3/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.xxxxxxxx.org","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_732359","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:07:45.159 MQT: stat/sensor3/STATUS7 = {"StatusTIM":{"UTC":"2021-04-09T16:07:45","Local":"2021-04-09T18:07:45","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"07:44","Sunset":"20:46"}}
18:07:45.186 MQT: stat/sensor3/STATUS10 = {"StatusSNS":{"Time":"2021-04-09T18:07:45","BME280":{"Temperature":23.9,"Humidity":33.7,"DewPoint":7.0,"Pressure":949.1},"PressureUnit":"hPa","TempUnit":"C"}}
18:07:45.209 MQT: stat/sensor3/STATUS11 = {"StatusSTS":{"Time":"2021-04-09T18:07:45","Uptime":"0T02:14:29","UptimeSec":8069,"Heap":13,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxxxxxxx","BSSId":"D8:FB:5E:xx:xx:xx","Channel":1,"RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:01"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
tmsend hello
18:14:25.552 SRC: WebConsole from 0.0.0.0
18:14:25.555 CMD: Group 0, Index 1, Command "TMSEND", Data "hello"
18:14:25.559 TGM: sendMessage
18:14:25.562 TGM: Cmnd bot425652e98:ABHxx7R-LjBv68_cJFHF6DwqtJRoNsiUp05/sendMessage?chat_id=-2147483648&text=hello
18:14:26.607 TGM: Connected in 1081 ms, max ThunkStack used 4516
18:14:26.914 MQT: stat/sensor3/RESULT = {"TmSend":"Failed"}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Set  bot token with tmtoken.
Set chat_id with tmchatid, with chat_id greater than 2147483648 (absolute value)
tmsend hello.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Message sent.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
The chat_id showed is always truncated to 2^31, although it is correctly stored, as can be seen with command tmchatid.
Changing the chat_id variable type from int32_t to String solves de problem. I will have a pull request if issue is confirmed.

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-04-10:
I suggest you have a look at #9831 why this is an int32 now instead of a uint32.

arendst, 2021-04-10:
I think I'll move it to String anyway. Let me do some tests.

arendst, 2021-04-10:
Pls try and report back.

ascillato2, 2021-04-10:
@jcmadrioso

Please, try the latest changes made by Theo. Thanks.

closed 2021-04-10 by ascillato2

jcmadrioso, 2021-04-10:
I can now confirm that the modification is working perfectly. 
Thank you very much.