arendst/Tasmota #11677: Alexa dont find Tasmota HUE Emulation device

Sub7evenHH, 2021-04-12:
### PROBLEM DESCRIPTION
Tasmota device with HUE emulation is not found on Alexa. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): ESP-01, Wemos D1, RGB-Controller WS2812, Echo Dot (1x 2.Generation, 1x 3.Generation, 1x Echo Show)
- [X ] Tasmota binary firmware version number used: 9.3.1
  - [ X] Pre-compiled
  - [ ] Self-compiled
- [ X] Flashing tools used: Tasmotizer
- [ X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
`00:57:35.144 CMD: Backlog Template; Module; GPIO 255
00:57:35.146 SRC: WebConsole from 192.168.178.21
00:57:35.149 CMD: Gruppe 0, Index 1, Befehl "BACKLOG", Daten "Template; Module; GPIO 255"
00:57:35.215 SRC: Backlog
00:57:35.218 CMD: Gruppe 0, Index 1, Befehl "TEMPLATE", Daten ""
00:57:35.223 RSL: stat/tasmota_7EA1AE/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
00:57:35.428 SRC: Backlog
00:57:35.431 CMD: Gruppe 0, Index 1, Befehl "MODULE", Daten ""
00:57:35.435 RSL: stat/tasmota_7EA1AE/RESULT = {"Module":{"18":"Generic"}}
00:57:35.679 SRC: Backlog
00:57:35.682 CMD: Gruppe 0, Index 1, Befehl "GPIO", Daten "255"
00:57:35.689 RSL: stat/tasmota_7EA1AE/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"1376":"WS2812"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
00:57:42.049 WIF: Prüfe Verbindung...`

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [X ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
`00:58:12.165 CMD: Gruppe 0, Index 1, Befehl "STATUS", Daten "0"
00:58:12.171 RSL: stat/tasmota_7EA1AE/STATUS = {"Status":{"Module":18,"DeviceName":"abert","FriendlyName":["abert"],"Topic":"tasmota_7EA1AE","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
00:58:12.200 RSL: stat/tasmota_7EA1AE/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:45:34","StartupUTC":"2021-04-11T23:12:38","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"2021-04-12T00:11:51","SaveCount":27,"SaveAddress":"F9000"}}
00:58:12.235 RSL: stat/tasmota_7EA1AE/STATUS2 = {"StatusFWR":{"Version":"9.3.1(tasmota)","BuildDateTime":"2021.03.09 16:14:55","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"385/699"}}
00:58:12.255 RSL: stat/tasmota_7EA1AE/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MyHome WIFI-1",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
00:58:12.287 RSL: stat/tasmota_7EA1AE/STATUS4 = {"StatusMEM":{"ProgramSize":594,"Free":408,"Heap":25,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
00:58:12.323 RSL: stat/tasmota_7EA1AE/STATUS5 = {"StatusNET":{"Hostname":"tasmota_7EA1AE-0430","IPAddress":"192.168.178.69","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"48:3F:DA:7E:A1:AE","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
00:58:12.344 RSL: stat/tasmota_7EA1AE/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7EA1AE","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
00:58:12.366 RSL: stat/tasmota_7EA1AE/STATUS7 = {"StatusTIM":{"UTC":"2021-04-11T23:58:12","Local":"2021-04-12T00:58:12","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:04","Sunset":"19:36"}}
00:58:12.386 RSL: stat/tasmota_7EA1AE/STATUS10 = {"StatusSNS":{"Time":"2021-04-12T00:58:12"}}
00:58:12.394 RSL: stat/tasmota_7EA1AE/STATUS11 = {"StatusSTS":{"Time":"2021-04-12T00:58:12","Uptime":"0T00:45:34","UptimeSec":2734,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"MqttCount":0,"POWER":"OFF","Dimmer":100,"Color":"00FF33","HSBColor":"132,100,100","Channel":[0,100,20],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"MyHome WIFI-1","BSSId":"F0:B0:14:D3:52:70","Channel":1,"RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:03"}}}`

```
- [ X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

`00:14:21.751 WIF: Prüfe Verbindung...
00:14:33.260 UDP: Packet (119)
00:14:33.263 UDP: HUE
00:14:33.271 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.27:50000
00:14:33.274 UDP: Packet (119)
00:14:33.277 UDP: Packet (119)
00:14:33.282 UDP: Packet (119)
00:14:33.285 UDP: Packet (119)
00:14:33.287 UDP: Packet (119)
00:14:33.525 HTP: Hue-Setup
00:14:34.254 HTP: Hue-API () from 192.168.178.27
00:14:34.257 HTP: Hue POST-Argumente ({"devicetype": "Echo"})
00:14:34.263 HTP: Hue Authentication Result ([{"success":{"username":"7ea1ae"}}])
00:14:34.301 HTP: Hue-API (/7ea1ae/lights) from 192.168.178.27
00:14:34.312 HTP: Hue Result ({"132782817":{"state":{"on":true,"bri":254,"colormode":"hs","xy":[0.73500,0.26500],"hue":0,"sat":254,"alert":"n...`
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
- Flash Tasmota Device
- Use Template Generic(18) with RGB-Controller and ESP-01 and set GPIO2 to WS2812
- Activate HUE Emulation
- Search for new smart home devices on alexa (web, direct with echo dot)

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
- Open ports for devices
- delete all smart home devices and groups in alexa and start new search
- Try with older Tasmota Firmware (from 7.0 - 9.3.1)
- Wemo Emulation. That works without problems
- Search Tasmota Device in a seperate WIFI (mobile Hotspot) with Echo Dot (2.Generation and 3.Generation)
- Try SetOption109
- search the tasmota hue emulation device in an alternative app (all 4 hue). With this app i found and control the device without problems. Only in alexa it not works

Logfile with all 4 hue app (android)
`00:38:13.552 HTP: Hue-API (/7ea1ae) from 192.168.178.20
00:38:13.651 HTP: Hue-API (/7ea1ae) from 192.168.178.20
00:38:21.767 WIF: Prüfe Verbindung...
00:38:22.067 UDP: Packet (101)
00:38:22.070 UDP: HUE
00:38:22.078 UPP: Hue 3 Antwortpakete gesendet zu 192.168.178.20:1901
00:38:29.195 HTP: Hue-API (/config) from 192.168.178.20
00:38:41.099 HTP: Hue-API (/config) from 192.168.178.20
00:38:41.784 WIF: Prüfe Verbindung...
00:38:42.189 HTP: Hue-API (/) from 192.168.178.20
00:38:42.191 HTP: Hue POST-Argumente ()
00:38:42.194 HTP: Hue POST-Argumente ({ "devicetype" : "all 4 hue#PCT-L29" })
00:38:42.199 HTP: Hue Authentication Result ([{"success":{"username":"7ea1ae"}}])
00:38:42.287 HTP: Hue-API (/7ea1ae) from 192.168.178.20
00:38:46.068 HTP: Hue-API (/7ea1ae) from 192.168.178.20
00:38:46.649 UDP: Packet (119)
00:38:46.652 UDP: HUE
00:38:47.649 UDP: Packet (119)
00:38:47.652 UDP: HUE
00:38:47.654 UDP: Packet (119)
00:38:48.703 HTP: Hue-API (/7ea1ae) from 192.168.178.20
00:38:48.725 UDP: Packet (119)
00:38:48.728 UDP: HUE
00:38:48.804 HTP: Hue-API (/7ea1ae) from 192.168.178.20
00:38:49.697 UDP: Packet (119)
00:38:49.700 UDP: HUE
00:38:52.692 UDP: Packet (119)
00:38:52.695 UDP: HUE`

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
6 months ago this worked with the same configurations with another device. It goes until today. 

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2021-04-12:
Fixed in 9.3.1.2
Use the latest development version from http://ota.tasmota.com/tasmota/

Sub7evenHH, 2021-04-12:
I try it with 9.3.1.2 and it works :)
Thank you very much!

closed 2021-04-12 by Sub7evenHH