arendst/Tasmota #11680: SML json SENSOR invalid

OCIN91, 2021-04-12:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
JSON Template not Valid by use SML Meter
{"Time":"2021-04-12T09:37:42","SML  ":{"Total_in":720.2593,"Total_out":1091.3074,"Power_curr":-821  ,"daily_consumption":0.00,"daily_feed":0.00  }

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): ESP2866 -Wemos D1
- [x ] Tasmota binary firmware version number used: 9.3.1.2
  - [ ] Pre-compiled
  - [ x] Self-compiled
- [ x] Flashing tools used: Visual Studio with Plattform IO
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
 Expecting ending } in STATUS JSON 

Error:  [Code 22, Structure 27]

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2021-04-12:
this was a bug in a previous version. please use last dev version.

ascillato2, 2021-04-12:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

OCIN91, 2021-04-13:
> this was a bug in a previous version. please use last dev version.

I have Compiled the last dev yesterday, same issure

gemu2015, 2021-04-13:
this is indeed a bug. it is caused by your meter number not being updated from your meter.
if you delete the last decoder line the problem should be fixed.

but you may also try this version. it should fix the bug also.

[xsns_53_sml.ino.zip](https://github.com/arendst/Tasmota/files/6304999/xsns_53_sml.ino.zip)

closed 2021-04-15 by ascillato2