arendst/Tasmota #11682: ADC Button causes reset

nonix, 2021-04-12:
### PROBLEM DESCRIPTION
The Wemos D1 mini resets to default when ADC Button is configured. Seems to be working well with with ADC Input (std. analog IN).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 Mini
- [x] Tasmota binary firmware version number used: 9.3.1.2
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
10:50:04.593 CMD: Backlog Template; Module; GPIO 255
10:50:04.666 MQT: stat/tester/RESULT = {"NAME":"Gosund 112v3.4","GPIO":[320,0,321,0,2656,2720,0,0,2624,257,224,0,0,4800],"FLAG":0,"BASE":18}
10:50:04.884 MQT: stat/tester/RESULT = {"Module":{"0":"Gosund 112v3.4"}}
10:50:05.138 MQT: stat/tester/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"321":"Led_i2"},"GPIO3":{"0":"None"},"GPIO4":{"2656":"HLWBL CF1"},"GPIO5":{"2720":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"257":"Relay_i2"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"4800":"ADC Button"}}
```

- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
10:53:13.938 MQT: stat/tester/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
10:53:13.943 MQT: stat/tester/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
10:53:13.955 MQT: stat/tester/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
```

- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
10:52:57.724 MQT: stat/tester/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["tester","Tasmota2"],"Topic":"tester","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
```

- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:00:00.041 CFG: Loaded from flash at F6, Count 6
00:00:00.047 QPC: Count 1
00:00:00.054 Project tasmota Tasmota Version 9.3.1.2(tasmota)-2_7_4_9(2021-04-11T20:05:24)
00:00:00.562 WIF: Connecting to AP1 0325125441 in mode 11n as tester-5982...
00:00:05.005 WIF: Connected
00:00:05.259 HTP: Web server active on tester-5982 with IP address 192.168.13.130
10:57:21.277 MQT: Attempting connection...
10:57:21.299 MQT: Connected
10:57:21.302 MQT: tele/tester/LWT = Online (retained)
10:57:21.304 MQT: cmnd/tester/POWER = 
10:57:21.318 MQT: tele/tester/INFO1 = {"Info1":{"Module":"Gosund 112v3.4","Version":"9.3.1.2(tasmota)","FallbackTopic":"cmnd/DVES_E3B75E_fb/","GroupTopic":"cmnd/tasmotas/"}}
10:57:21.326 MQT: tele/tester/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tester-5982","IPAddress":"192.168.13.130"}}
10:57:21.337 MQT: tele/tester/INFO3 = {"Info3":{"RestartReason":"Software/System restart"}}
10:57:21.345 MQT: stat/tester/RESULT = {"POWER1":"OFF"}
10:57:21.349 MQT: stat/tester/POWER1 = OFF
10:57:21.352 MQT: stat/tester/RESULT = {"POWER2":"OFF"}
10:57:21.357 MQT: stat/tester/POWER2 = OFF
10:57:21.703 QPC: Reset
10:57:25.681 MQT: tele/tester/STATE = {"Time":"2021-04-12T10:57:25","Uptime":"0T00:00:12","UptimeSec":12,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"0325125441","BSSId":"34:31:C4:CA:7D:E4","Channel":1,"RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:06"}}
10:57:25.709 MQT: tele/tester/SENSOR = {"Time":"2021-04-12T10:57:25","ENERGY":{"TotalStartTime":"2021-04-12T10:56:02","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}
10:57:45.409 CMD: WebLog 4
10:57:45.416 MQT: stat/tester/RESULT = {"WebLog":4}
10:57:45.844 CFG: Saved to flash at FA, Count 10, Bytes 4096
10:57:59.686 MQT: stat/tester/RESULT = {"Reset":"Reset and Restarting"}
10:57:59.777 CFG: Use defaults
```

### TO REPRODUCE
Using Wemos D1 mini (no other circuitry) connected and powered via USB, flash the latest dev. fw. and apply template:
{"NAME":"Gosund 112v3.4","GPIO":[320,0,321,0,2656,2720,0,0,2624,257,224,0,0,4800],"FLAG":0,"BASE":18}

### EXPECTED BEHAVIOUR
Not to reset after a short while

### SCREENSHOTS
none

### ADDITIONAL CONTEXT
It could be related to this discussion: [Error! in new Wemos d1 mini !!!](https://github.com/arendst/Tasmota/discussions/11615)
The issue could also be related to this talk: [ESP8266 Analog read interferes with wifi?](https://arduino.stackexchange.com/questions/19787/esp8266-analog-read-interferes-with-wifi) or WDT will trigger reset when not refreshed for a period during analog read sampling.

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-04-12:
It has nothing to do with any Additional Context issues.

As you only configured one button it automagically becomes Button 1. When button 1 is pressed for over 40 seconds (default) it will reset your device stated by the message:
```
10:57:59.686 MQT: stat/tester/RESULT = {"Reset":"Reset and Restarting"}
10:57:59.777 CFG: Use defaults
```
Apparently you wired the button in such a way that it is detected as always on. Tasmota then triggers a reset after 40 seconds. 

Just rethink your wiring.

NB. Enable ``seriallog 4`` for more information telling you the button is initiating the reset.

nonix, 2021-04-12:
Yes, you are right. 'm sorry to waste your time.

closed 2021-04-12 by nonix