arendst/Tasmota #11712: 3 shutters detected on a Sonoff 4chpro configured for 2. Using HA and new Tasmota integration.

JackTalisker, 2021-04-13:
### PROBLEM DESCRIPTION
3 shutters detected on a Sonoff 4chpro configured for 2. Using HA and new Tasmota integration.
Seems that Tasmota discovery topic sends null values after Shutter1 and Shutter2. After that a useless "Shutter3" appears.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: 9.3.1.2
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
17:32:40.979 MQT: stat/4CHpro_serr/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
17:32:41.189 MQT: stat/4CHpro_serr/RESULT = {"Module":{"23":"Sonoff 4CH Pro"}}
17:32:41.443 MQT: stat/4CHpro_serr/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"226":"Relay3"},"GPIO5":{"225":"Relay2"},"GPIO9":{"33":"Button2"},"GPIO10":{"34":"Button3"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"35":"Button4"},"GPIO15":{"227":"Relay4"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
17:34:00.422 MQT: stat/4CHpro_serr/STATUS = {"Status":{"Module":23,"DeviceName":"4CHpro_serr","FriendlyName":["SerrSala1_1","SerrSala1_2","SerrSala2_1","SerrSala2_2"],"Topic":"4CHpro_serr","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
17:34:00.428 MQT: stat/4CHpro_serr/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:23:53","StartupUTC":"2021-04-13T16:10:07","Sleep":50,"CfgHolder":4617,"BootCount":38,"BCResetTime":"2021-01-17T15:21:43","SaveCount":1432,"SaveAddress":"FA000"}}
17:34:00.435 MQT: stat/4CHpro_serr/STATUS2 = {"StatusFWR":{"Version":"9.3.1.2(tasmota)","BuildDateTime":"2021-04-13T15:02:47","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"407/699"}}
17:34:00.440 MQT: stat/4CHpro_serr/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TaliskerUp",""],"TelePeriod":180,"Resolution":"558180C0","SetOption":["0000C009","2805C8000100060000005A0A000000000000","C0000080","00006000","00000000"]}}
17:34:00.449 MQT: stat/4CHpro_serr/STATUS4 = {"StatusMEM":{"ProgramSize":598,"Free":404,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
17:34:00.456 MQT: stat/4CHpro_serr/STATUS5 = {"StatusNET":{"Hostname":"4CHpro_serr-1529","IPAddress":"192.168.0.80","Gateway":"192.168.0.3","Subnetmask":"255.255.255.0","DNSServer":"208.67.222.222","Mac":"60:01:94:AC:E5:F9","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
17:34:00.463 MQT: stat/4CHpro_serr/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.20","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_ACE5F9","MqttUser":"dueller","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
17:34:00.470 MQT: stat/4CHpro_serr/STATUS7 = {"StatusTIM":{"UTC":"2021-04-13T16:34:00","Local":"2021-04-13T17:34:00","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"06:02","Sunset":"19:38"}}
17:34:00.478 MQT: stat/4CHpro_serr/STATUS10 = {"StatusSNS":{"Time":"2021-04-13T17:34:00","Shutter1":{"Position":88,"Direction":0,"Target":88},"Shutter2":{"Position":100,"Direction":0,"Target":100}}}
17:34:00.486 MQT: stat/4CHpro_serr/STATUS11 = {"StatusSTS":{"Time":"2021-04-13T17:34:00","Uptime":"0T00:23:53","UptimeSec":1433,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":39,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"TaliskerUp","BSSId":"C8:54:4B:CC:22:15","Channel":3,"RSSI":54,"Signal":-73,"LinkCount":1,"Downtime":"0T00:00:03"}}}
17:34:00.491 MQT: stat/4CHpro_serr/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":295,"Close":285,"50perc":68,"Delay":0,"Opt":"0100","Calib":[2130,377,4637,900,1000],"Mode":"0"},"SHT1":{"Relay1":3,"Relay2":4,"Open":151,"Close":150,"50perc":68,"Delay":0,"Opt":"0100","Calib":[1090,193,2373,900,1000],"Mode":"0"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
17:38:54.044 CMD: weblog 4
17:38:54.051 MQT: stat/4CHpro_serr/RESULT = {"WebLog":4}
17:38:54.905 CFG: Saved to flash at F9, Count 1433, Bytes 4096
17:38:59.040 WIF: Checking connection...
17:39:15.737 CMD: setoption19 0
17:39:15.739 SRC: WebConsole from 192.168.0.173
17:39:15.742 CMD: Group 0, Index 19, Command "SETOPTION", Data "0"
17:39:15.748 MQT: stat/4CHpro_serr/RESULT = {"SetOption19":"OFF"}
17:39:16.604 MQT: homeassistant/binary_sensor/ACE5F9_SW_3/config =  (retained)
17:39:16.607 MQT: homeassistant/device_automation/ACE5F9_SW_4_TOGGLE/config =  (retained)
17:39:16.610 MQT: homeassistant/device_automation/ACE5F9_SW_4_HOLD/config =  (retained)
17:39:16.613 MQT: homeassistant/binary_sensor/ACE5F9_SW_4/config =  (retained)
17:39:16.617 MQT: homeassistant/device_automation/ACE5F9_SW_5_TOGGLE/config =  (retained)
17:39:16.621 MQT: homeassistant/device_automation/ACE5F9_SW_5_HOLD/config =  (retained)
17:39:16.624 MQT: homeassistant/binary_sensor/ACE5F9_SW_5/config =  (retained)
17:39:16.629 MQT: homeassistant/device_automation/ACE5F9_SW_6_TOGGLE/config =  (retained)
17:39:16.632 MQT: homeassistant/device_automation/ACE5F9_SW_6_HOLD/config =  (retained)
17:39:16.635 MQT: homeassistant/binary_sensor/ACE5F9_SW_6/config =  (retained)
17:39:16.639 MQT: homeassistant/device_automation/ACE5F9_SW_7_TOGGLE/config =  (retained)
17:39:16.642 MQT: homeassistant/device_automation/ACE5F9_SW_7_HOLD/config =  (retained)
17:39:16.647 MQT: homeassistant/binary_sensor/ACE5F9_SW_7/config =  (retained)
17:39:16.651 MQT: homeassistant/device_automation/ACE5F9_SW_8_TOGGLE/config =  (retained)
17:39:16.655 MQT: homeassistant/device_automation/ACE5F9_SW_8_HOLD/config =  (retained)
17:39:16.660 MQT: homeassistant/binary_sensor/ACE5F9_SW_8/config =  (retained)
[... 21 lines cut ...]
17:39:16.746 LOG: Home Assistant MQTT Discovery disabled.
17:39:16.757 MQT: tasmota/discovery/600194ACE5F9/config = {"ip":"192.168.0.80","dn":"4CHpro_serr","fn":["SerrSala1_1","SerrSala1_2","SerrSala2_1","SerrSala2_2",null,null,null,null],"hn":"4CHpro_serr-1529","mac":"600194ACE5F9","md":"Sonoff 4CH Pro","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"9.3.1.2","t":"4CHpro_serr","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[3,3,3,3,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":1,"lt_st":0,"sho":[4,4,0,0],"ver":1} (retained)
17:39:16.767 MQT: tasmota/discovery/600194ACE5F9/sensors = {"sn":{"Time":"2021-04-13T17:39:16","Shutter1":{"Position":88,"Direction":0,"Target":88},"Shutter2":{"Position":100,"Direction":0,"Target":100}},"ver":1} (retained)
17:39:19.290 WIF: Checking connection...
17:39:39.572 WIF: Checking connection...

```

### TO REPRODUCE
Setup the Tasmota HA integration with a Sonoff 4chpro configured for 2 shutters.

[... cut ...]
{"ip":"192.168.0.80","dn":"4CHpro_serr","fn":["SerrSala1_1","SerrSala1_2","SerrSala2_1","SerrSala2_2",null,null,null,null],"hn":"4CHpro_serr-1529","mac":"600194ACE5F9","md":"Sonoff 4CH Pro","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"9.3.1.2","t":"4CHpro_serr","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[3,3,3,3,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":1,"lt_st":0,"sho":[4,4,0,0],"ver":1}
```

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2021-04-30:
Can you give more insight WHY HA detects 3 shutters. What is the wrong value in the discovery? Are the "0" the problem? Why it does not detect 4 in this case? I assume "sho":[4,4,0,0] is the part for the shutter in HA?

ascillato2, 2021-05-16:
Hi,

When you have some time, please provide the requested information so as to properly help you. Thanks.

closed 2021-05-16 by ascillato2

JackTalisker, 2021-05-17:
> Can you give more insight WHY HA detects 3 shutters. What is the wrong value in the discovery? 

I don't know the discovery process. The only thing I suppose could be wrong is the "fn" value:
"fn":["SerrSala1_1","SerrSala1_2","SerrSala2_1","SerrSala2_2",null,null,null,null], 
where SerrSala1_1 is the first shutter going up (relay 1), SerrSala1_2 is the first shutter going down  (relay 2),
SerrSala2_1 is the second shutter going up (relay 3) and SerrSala2_2 is the second shutter going down (relay 4);
after there are those values "null,null,null,null" that could be wrong. I really don't know.  

> Are the "0" the problem? Why it does not detect 4 in this case? I assume "sho":[4,4,0,0] is the part for the shutter in HA?
Don't know.

If I could help providing other data just ask.
Thank you

stefanbode, 2021-05-17:
Let me do a small patch and get rid of the null,null. Have you checked how the message look like if there is only ONE shutter?

JackTalisker, 2021-05-17:
I have another 4chpro configured with 2 relays for a single shutter and the other two realy for two switch (for 2 bulbs).

I gave a "weblog 4" and  a "SetOption19 0" in the console and the null values are still there but this time the discovery is right, one shutter and two switches:

`
06:34:54.592 CMD: weblog 4
06:34:54.599 MQT: stat/4CHpro_corr/RESULT = {"WebLog":4}
06:34:54.926 CFG: Saved to flash at F6, Count 307, Bytes 4096
06:34:55.081 WIF: Checking connection...
06:35:08.702 CMD: setoption19 0
06:35:08.705 SRC: WebConsole from 192.168.0.20
06:35:08.708 CMD: Group 0, Index 19, Command "SETOPTION", Data "0"
06:35:08.713 MQT: stat/4CHpro_corr/RESULT = {"SetOption19":"OFF"}
06:35:09.624 MQT: homeassistant/device_automation/FB72D7_SW_3_HOLD/config =  (retained)
06:35:09.628 MQT: homeassistant/binary_sensor/FB72D7_SW_3/config =  (retained)
06:35:09.633 MQT: homeassistant/device_automation/FB72D7_SW_4_TOGGLE/config =  (retained)
06:35:09.638 MQT: homeassistant/device_automation/FB72D7_SW_4_HOLD/config =  (retained)
06:35:09.642 MQT: homeassistant/binary_sensor/FB72D7_SW_4/config =  (retained)
06:35:09.647 MQT: homeassistant/device_automation/FB72D7_SW_5_TOGGLE/config =  (retained)
06:35:09.651 MQT: homeassistant/device_automation/FB72D7_SW_5_HOLD/config =  (retained)
06:35:09.655 MQT: homeassistant/binary_sensor/FB72D7_SW_5/config =  (retained)
06:35:09.660 MQT: homeassistant/device_automation/FB72D7_SW_6_TOGGLE/config =  (retained)
06:35:09.665 MQT: homeassistant/device_automation/FB72D7_SW_6_HOLD/config =  (retained)
06:35:09.669 MQT: homeassistant/binary_sensor/FB72D7_SW_6/config =  (retained)
06:35:09.674 MQT: homeassistant/device_automation/FB72D7_SW_7_TOGGLE/config =  (retained)
06:35:09.678 MQT: homeassistant/device_automation/FB72D7_SW_7_HOLD/config =  (retained)
06:35:09.683 MQT: homeassistant/binary_sensor/FB72D7_SW_7/config =  (retained)
06:35:09.687 MQT: homeassistant/device_automation/FB72D7_SW_8_TOGGLE/config =  (retained)
06:35:09.691 MQT: homeassistant/device_automation/FB72D7_SW_8_HOLD/config =  (retained)
06:35:09.696 MQT: homeassistant/binary_sensor/FB72D7_SW_8/config =  (retained)
06:35:09.701 MQT: homeassistant/cover/FB72D7_SHT_1/config =  (retained)
06:35:09.705 MQT: homeassistant/cover/FB72D7_SHT_2/config =  (retained)
06:35:09.709 MQT: homeassistant/cover/FB72D7_SHT_3/config =  (retained)
06:35:09.713 MQT: homeassistant/cover/FB72D7_SHT_4/config =  (retained)
06:35:09.718 MQT: homeassistant/light/FB72D7_LI_1/config =  (retained)
06:35:09.722 MQT: homeassistant/switch/FB72D7_RL_1/config =  (retained)
06:35:09.726 MQT: homeassistant/light/FB72D7_LI_2/config =  (retained)
06:35:09.731 MQT: homeassistant/switch/FB72D7_RL_2/config =  (retained)
06:35:09.735 MQT: homeassistant/light/FB72D7_LI_3/config =  (retained)
06:35:09.739 MQT: homeassistant/switch/FB72D7_RL_3/config =  (retained)
06:35:09.744 MQT: homeassistant/light/FB72D7_LI_4/config =  (retained)
06:35:09.748 MQT: homeassistant/switch/FB72D7_RL_4/config =  (retained)
06:35:09.752 MQT: homeassistant/light/FB72D7_LI_5/config =  (retained)
06:35:09.756 MQT: homeassistant/switch/FB72D7_RL_5/config =  (retained)
06:35:09.759 MQT: homeassistant/light/FB72D7_LI_6/config =  (retained)
06:35:09.763 MQT: homeassistant/switch/FB72D7_RL_6/config =  (retained)
06:35:09.767 MQT: homeassistant/light/FB72D7_LI_7/config =  (retained)
06:35:09.771 MQT: homeassistant/switch/FB72D7_RL_7/config =  (retained)
06:35:09.776 MQT: homeassistant/light/FB72D7_LI_8/config =  (retained)
06:35:09.779 MQT: homeassistant/switch/FB72D7_RL_8/config =  (retained)
06:35:09.784 MQT: homeassistant/sensor/FB72D7_status/config =  (retained)
06:35:09.786 LOG: Home Assistant MQTT Discovery disabled.
06:35:09.797 MQT: tasmota/discovery/40F520FB72D7/config = {"ip":"192.168.0.82","dn":"4CHpro_corr","fn":["corridoio_su","corridoio_giu","corridoio_luce","corridoio_luce_cass",null,null,null,null],"hn":"4CHpro_corr-4823","mac":"40F520FB72D7","md":"Sonoff 4CH Pro","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"9.4.0.3","t":"4CHpro_corr","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[3,3,1,1,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":1,"lt_st":0,"sho":[4,0,0,0],"ver":1} (retained)
06:35:09.807 MQT: tasmota/discovery/40F520FB72D7/sensors = {"sn":{"Time":"2021-05-17T06:35:09","Shutter1":{"Position":100,"Direction":0,"Target":100}},"ver":1} (retained)

`
[image]