arendst/Tasmota #11716: PowerDelta calculation lacks precision

isorin, 2021-04-13:
### PROBLEM DESCRIPTION
Using "WattRes 1" so power is reported with 1 decimal precision.
and PowerDelta "15" which should trigger at 15% but looks like differences between 6.6W and 7.1W are enough to trigger it.
Looking at the logs looks like the delta calculation is done using only integers.

```
8:31:30.804 CMD: PowerDelta 15
18:31:30.813 MQT: stat/tasmota_0329F7/RESULT = {"PowerDelta1":15}
18:32:13.552 MQT: tele/tasmota_0329F7/MARGINS = {"MARGINS":{"PowerDelta":1}}
18:32:13.560 MQT: tele/tasmota_0329F7/SENSOR = {"Time":"2021-04-13T18:32:13","ENERGY":{"TotalStartTime":"2021-03-01T11:31:01","Total":16.33415,"Yesterday":0.40627,"Today":0.28024,"Power":7.1,"ApparentPower":15.6,"ReactivePower":13.9,"Factor":0.45,"Voltage":241,"Current":0.065}}
18:32:14.552 MQT: tele/tasmota_0329F7/MARGINS = {"MARGINS":{"PowerDelta":-1}}
18:32:14.561 MQT: tele/tasmota_0329F7/SENSOR = {"Time":"2021-04-13T18:32:14","ENERGY":{"TotalStartTime":"2021-03-01T11:31:01","Total":16.33415,"Yesterday":0.40627,"Today":0.28024,"Power":6.6,"ApparentPower":15.9,"ReactivePower":14.5,"Factor":0.42,"Voltage":241,"Current":0.066}}
18:32:43.549 MQT: tele/tasmota_0329F7/MARGINS = {"MARGINS":{"PowerDelta":1}}
18:32:43.557 MQT: tele/tasmota_0329F7/SENSOR = {"Time":"2021-04-13T18:32:43","ENERGY":{"TotalStartTime":"2021-03-01T11:31:01","Total":16.33420,"Yesterday":0.40627,"Today":0.28029,"Power":7.0,"ApparentPower":15.7,"ReactivePower":14.0,"Factor":0.45,"Voltage":241,"Current":0.065}}
18:32:44.551 MQT: tele/tasmota_0329F7/MARGINS = {"MARGINS":{"PowerDelta":-1}}
18:32:44.559 MQT: tele/tasmota_0329F7/SENSOR = {"Time":"2021-04-13T18:32:44","ENERGY":{"TotalStartTime":"2021-03-01T11:31:01","Total":16.33421,"Yesterday":0.40627,"Today":0.28030,"Power":6.7,"ApparentPower":15.7,"ReactivePower":14.2,"Factor":0.43,"Voltage":241,"Current":0.065}}
18:33:43.541 MQT: tele/tasmota_0329F7/MARGINS = {"MARGINS":{"PowerDelta":1}}
18:33:43.549 MQT: tele/tasmota_0329F7/SENSOR = {"Time":"2021-04-13T18:33:43","ENERGY":{"TotalStartTime":"2021-03-01T11:31:01","Total":16.33432,"Yesterday":0.40627,"Today":0.28041,"Power":7.0,"ApparentPower":15.6,"ReactivePower":14.0,"Factor":0.45,"Voltage":240,"Current":0.065}}
18:33:44.544 MQT: tele/tasmota_0329F7/MARGINS = {"MARGINS":{"PowerDelta":-1}}
18:33:44.552 MQT: tele/tasmota_0329F7/SENSOR = {"Time":"2021-04-13T18:33:44","ENERGY":{"TotalStartTime":"2021-03-01T11:31:01","Total":16.33432,"Yesterday":0.40627,"Today":0.28041,"Power":6.6,"ApparentPower":15.7,"ReactivePower":14.2,"Factor":0.42,"Voltage":241,"Current":0.065}}
18:34:13.544 MQT: tele/tasmota_0329F7/MARGINS = {"MARGINS":{"PowerDelta":1}}
18:34:13.551 MQT: tele/tasmota_0329F7/SENSOR = {"Time":"2021-04-13T18:34:13","ENERGY":{"TotalStartTime":"2021-03-01T11:31:01","Total":16.33437,"Yesterday":0.40627,"Today":0.28046,"Power":7.1,"ApparentPower":15.6,"ReactivePower":13.9,"Factor":0.45,"Voltage":241,"Current":0.065}}
18:34:14.543 MQT: tele/tasmota_0329F7/MARGINS = {"MARGINS":{"PowerDelta":-1}}
18:34:14.553 MQT: tele/tasmota_0329F7/SENSOR = {"Time":"2021-04-13T18:34:14","ENERGY":{"TotalStartTime":"2021-03-01T11:31:01","Total":16.33437,"Yesterday":0.40627,"Today":0.28046,"Power":6.8,"ApparentPower":15.9,"ReactivePower":14.4,"Factor":0.43,"Voltage":241,"Current":0.066}}
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
WattRes 1
PowerDelta 15

### EXPECTED BEHAVIOUR
Expected PowerDelta to take into account the precision of measurement and calculate the delta percentage accurately.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-04-13:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

isorin, 2021-04-14:
I have updated the template.
I can provide more info if needed.

arendst, 2021-04-14:
As designed.

The energy monitoring hardware is not able to consistently report power useage to the decimal.

The feature is supposed to be used for decent loads anyway.

sfromis, 2021-04-14:
I'd say that PowerDelta mainly make sense with watt-based deltas, **not** for percents at moderate loads. As a starting point, I'd suggest using `PowerDelta 110` for reasonably reliable operation.

closed 2021-04-14 by ascillato2