arendst/Tasmota #11774: Sonoff POW R2 Energy Total set to nonsense number after some time

Dehumanizer77, 2021-04-18:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Sonoff Pow R2 Energy Total set to a nonsense number always after some time, for example the daily energy usage on my device is 0.2-1 kWh and after resetting energy total to 0, after 1 or 2 days the energy total jumps to over 300 kWh.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Pow R2
- [x] Tasmota binary firmware version number used: 9.3.1, same problem with 9.1.0 previously
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
11:13:38.607 MQT: stat/budaheat/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
11:13:38.807 MQT: stat/budaheat/RESULT = {"Module":{"43":"Sonoff Pow R2"}}
11:13:39.012 MQT: stat/budaheat/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"3072":"CSE7766 Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3104":"CSE7766 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
11:16:00.891 MQT: stat/budaheat/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}
11:16:01.097 MQT: stat/budaheat/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
11:17:46.185 MQT: stat/budaheat/STATUS = {"Status":{"Module":43,"DeviceName":"Kurenie v bude","FriendlyName":["Kurenie v bude"],"Topic":"budaheat","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
11:17:46.193 MQT: stat/budaheat/STATUS1 = {"StatusPRM":{"Baudrate":4800,"SerialConfig":"8E1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Hardware Watchdog","Uptime":"0T00:36:45","StartupUTC":"2021-04-18T08:41:01","Sleep":50,"CfgHolder":4617,"BootCount":1369,"BCResetTime":"2021-01-27T18:58:24","SaveCount":2311,"SaveAddress":"F4000"}}
11:17:46.199 MQT: stat/budaheat/STATUS2 = {"StatusFWR":{"Version":"9.3.1(tasmota)","BuildDateTime":"2021-03-09T16:12:28","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"405/699"}}
11:17:46.205 MQT: stat/budaheat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["2600_mqtt",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100068000005A0A000000000000","00000280","00006000","00000000"]}}
11:17:46.217 MQT: stat/budaheat/STATUS4 = {"StatusMEM":{"ProgramSize":593,"Free":408,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
11:17:46.224 MQT: stat/budaheat/STATUS5 = {"StatusNET":{"Hostname":"budaheat-1393","IPAddress":"192.168.128.16","Gateway":"192.168.128.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.128.1","Mac":"98:F4:AB:E2:E5:71","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
11:17:46.230 MQT: stat/budaheat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.128.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E2E571","MqttUser":"tasmota","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:17:46.238 MQT: stat/budaheat/STATUS7 = {"StatusTIM":{"UTC":"2021-04-18T09:17:46","Local":"2021-04-18T11:17:46","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"06:52","Sunset":"20:45"}}
11:17:46.244 MQT: stat/budaheat/STATUS9 = {"StatusPTH":{"PowerDelta":
![Untitled](https://user-images.githubusercontent.com/1876451/115140689-d8bfe700-a038-11eb-836e-a0664d177d73.png)
[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
11:17:46.253 MQT: stat/budaheat/STATUS10 = {"StatusSNS":{"Time":"2021-04-18T11:17:46","ENERGY":{"TotalStartTime":"2021-04-15T00:00:00","Total":337.370,"Yesterday":0.780,"Today":0.221,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
11:17:46.261 MQT: stat/budaheat/STATUS11 = {"StatusSTS":{"Time":"2021-04-18T11:17:46","Uptime":"0T00:36:45","UptimeSec":2205,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"2600_mqtt","BSSId":"92:2A:A8:94:EB:59","Channel":6,"RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
n/a

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
-reset Energy Total to 0 with "EnergyReset3 0"
-wait 24-48 hours 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Energy Total reading is correct.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-04-24:
Hi,

Please, update to latest version and try again. This has been solved. Thanks.

If you experience this again. Please, provide the console log when this happens but with weblog set to 4. Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-04-24 by ascillato2

Dehumanizer77, 2021-05-06:
Ok it seems it was working for a while, but it just happened today. I have turned off external syslog logging to 4 and will update the issue once it happens again.