arendst/Tasmota #11816: Gosund SP111 dies after manual toggle with Tasmota 9.3.1

TungstenE2, 2021-04-20:
### PROBLEM DESCRIPTION
I have 4 Gosund SP111 with Tasmota 9.3.1

2 of them recently died after manual toggle while a small 4w consuming pump was attached and turned on.

First time I thought this was a bug, but after the same happend again with similar situation this seems to be reproducable.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic):Gosund SP111
- [ x] Tasmota binary firmware version number used: 9.3.1
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [x ] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:  not possible as device no longer responding

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here: 

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here: not possible as device no longer responding

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:  not possible as device no longer responding

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Gosund SP111 with  Tasmota 9.3.1
Template {"NAME":"Gosund SP111 2","GPIO":[56,0,57,0,132,134,0,0,131,17,0,21,0],"FLAG":0,"BASE":18}
plugin small consumer eg with 4w
press button to switch of pluged in consumer device
try to press button again after 1 hr
device now longer working, webserver not responding, no reset possible

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Device should still be working after button is pressed

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

not possible as device no longer responding

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

TungstenE2, 2021-04-21:
@ascillato2 to which ticket is this duplicated?

barbudor, 2021-04-22:
@TungstenE2 
I have 8 Gosund SP111 plugs working very nicely for around 1 year with all Tasmota versions since 8.1.0. No bug so far.
Are you sure you didn't faced a configuration reset due to button pressed more than 40 seconds ? No tasmota-XXXX AP visible ?
You should join the Discord server for easier discussion.

TungstenE2, 2021-04-22:
yes, I am sure I did not trigger reset. Button was pressed for 1 sec. and plug switched off.

No AP is visible.

I have several other Tasmota devices working fine, so this is really strange.

With the first device I was thinking about a faulty SP111, but the second died exactly the same way.

Button pressed for 1 sec, while 4w consumer was plugged in. After that the SP111 is dead with no AP visible or tasmota device being logged in to the router.

barbudor, 2021-04-22:
I now remember that I had 1 plug that failed a few months ago (but without any specific action) and I had to reflash it.
I would open the plug and hook the serial adapter first to see what happens at boot time, and second to reflash it.

TungstenE2, 2021-04-22:
Of course I will try flashing, but this ticket is about issue itself. This should not have happend. As it happend on two differnt devices with similar precondictions I consider this a software bug, which should be reported in order to be followed up and fixed.

ryrzy, 2021-04-24:
i have one blitzwolf shp6 15A (similar as sp111) https://templates.blakadder.com/blitzwolf_SHP6-15A.html that sometimes make some strange noises after turn on by automatization - then it dosent switch with tasmota/home assistant switch but when i switch with button on cover it works ok and noises are gone, and tasmota/HA switch is working again. realy strange. i have think that relay is for change but it should not work then with cover switch hmm

Jason2866, 2021-04-24:
> Of course I will try flashing, but this ticket is about issue itself. This should not have happend. As it happend on two differnt devices with similar precondictions I consider this a software bug, which should be reported in order to be followed up and fixed.

And how is a hardware failure related to Tasmota?

TungstenE2, 2021-04-24:
> And how is a hardware failure related to Tasmota?

that is a good question. I would have agreed if it was one hardware failure. But 2 with similar conditions, I doubt that. Can you proove different? Happy to assist.

TungstenE2, 2021-04-24:
@ryrzy nice story, but how is this related to the issue?

Jason2866, 2021-04-24:
> that is a good question. I would have agreed if it was one hardware failure. But 2 with similar conditions, I doubt that. Can you proove different? Happy to assist.

I cant help, i can not reproduce this behaviour with my SP111.

ascillato2, 2021-04-24:
Closing this issue can not be reproduced as seems to be a hardware problem. Sorry.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-04-24 by ascillato2

mtw20, 2022-11-05:
I faced the same issue, two sockets already dead.

TungstenE2, 2022-11-06:
> I faced the same issue, two sockets already dead.

this is so sad, but at least good to know I was not the only one facing this issue

mlkoch, 2023-05-04:
Two of my four SP111 died during OTA from 8.4 to 12.5 as well. 
first step: Update to "minimal firmware" worked well
then they died
can not even flash with wires