arendst/Tasmota #11835: Regression with Tasmota 9.4.0: cam stream no longer visible with ESP32CAM

AlphaXMines, 2021-04-22:
### PROBLEM DESCRIPTION
Hello,
There is a regression with the new release 9.4.0 on ESP32CAM module, using AITHINKERCAM: the webcam no longer works while the cam stream was correctly visible with 9.3.1

Step to reproduce:

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32-CAM_____
- [x] Tasmota binary firmware version number used: http://ota.tasmota.com/tasmota32/release/tasmota32-webcam.bin _____
- [x] Flashing tools used: OTA upgrade_____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
12:17:59.608 RSL: RESULT = {"NAME":"AITHINKER CAM","GPIO":[4992,1,672,1,416,5088,1,1,1,6720,736,704,1,1,5089,5090,0,5091,5184,5152,0,5120,5024,5056,0,0,0,0,4928,576,5094,5095,5092,0,0,5093],"FLAG":0,"BASE":18}
12:17:59.819 RSL: RESULT = {"Module":{"0":"AITHINKER CAM"}}
12:18:00.081 RSL: RESULT = {"GPIO0":{"4992":"CAM_XCLK"},"GPIO1":{"0":"None"},"GPIO2":{"672":"SPI MISO"},"GPIO3":{"0":"None"},"GPIO4":{"416":"PWM1"},"GPIO5":{"5088":"CAM_DATA"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"6720":"SDCard CS"},"GPIO14":{"736":"SPI CLK"},"GPIO15":{"704":"SPI MOSI"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"5089":"CAM_DATA"},"GPIO19":{"5090":"CAM_DATA"},"GPIO20":{"0":"None"},"GPIO21":{"5091":"CAM_DATA"},"GPIO22":{"5184":"CAM_PCLK"},"GPIO23":{"5152":"CAM_HREF"},"GPIO24":{"0":"None"},"GPIO25":{"5120":"CAM_VSYNC"}}
12:18:00.094 RSL: RESULT = {"GPIO26":{"5024":"CAM_SIOD"},"GPIO27":{"5056":"CAM_SIOC"},"GPIO28":{"0":"None"},"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"4928":"CAM_PWDN"},"GPIO33":{"576":"LedLink_i"},"GPIO34":{"5094":"CAM_DATA"},"GPIO35":{"5095":"CAM_DATA"},"GPIO36":{"5092":"CAM_DATA"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"5093":"CAM_DATA"}}

```
- [x] no rule

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
12:18:37.571 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"ESP32CAM1","FriendlyName":["ESP32CAM1"],"Topic":"tasmota_EA6618","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
12:18:37.578 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"RTC Watch dog reset digital core and rtc module","Uptime":"0T00:45:34","StartupUTC":"2021-04-22T10:33:03","Sleep":50,"CfgHolder":4617,"BootCount":18,"BCResetTime":"2021-04-19T15:05:06","SaveCount":149}}
12:18:37.585 RSL: STATUS2 = {"StatusFWR":{"Version":"9.4.0(tasmota)","BuildDateTime":"2021-04-21T12:52:37","Core":"1_0_6","SDK":"v3.3.5-1-g85c43024c","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6","CR":"363/699"}}
12:18:37.589 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxx",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
12:18:37.604 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1472,"Free":1856,"Heap":209,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":3,"Features":["00000809","9FDAC787","00152001","B7F7BFCF","05DA9BC0","60360DC7","00084052","20200000","0000002D"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,17,20,21,24,26,27,29,34,50,52","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,87"}}
12:18:37.610 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_EA6618-1560","IPAddress":"192.168.1.231","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"9C:9C:1F:EA:66:18","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
12:18:37.614 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_EA6618","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
12:18:37.621 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-04-22T11:18:37","Local":"2021-04-22T12:18:37","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"05:45","Sunset":"19:51"}}
12:18:37.631 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-04-22T12:18:37","ESP32":{"Temperature":32.8},"TempUnit":"C"}}
12:18:37.639 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-04-22T12:18:37","Uptime":"0T00:45:34","UptimeSec":2734,"Heap":208,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"SFR5D78b","BSSId":"D8:0D:17:DE:A4:14","Channel":6,"RSSI":54,"Signal":-73,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
* Prepare a ESP32CAM with tasmota 9.3.1 (tasmota32-webcam.bin)
** customize TASMOTA to use the configuration AITHINKER CAM as defined here: https://tasmota.github.io/docs/ESP32/
*** menu Configuration
*** menu Configure other
*** past the template string of AITHINKER CAM in template field
*** save
The cam stream is visible in Tasmota (and you can also control the LED of the device)

* Update your module to use tasmota 9.4.0
** chose _Firmware Upgrade_
** then _Start upgrade_ button, to use the OTA update from the latest 9.4.0 release
The cam stream is no longer visible in Tasmota  (but you can still control the LED present on the device)

* Reupload tasmota 9.3.1 on your device
The cam stream is back

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The cam stream should be visible as it is with Tasmota 9.3.1

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-04-22:
Cannot reproduce.

If you load the wrong binary as noted in your ``otaurl`` ("OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin") you won't see the webcam.

The correct binary is ``http://ota.tasmota.com/tasmota32/release/tasmota32-webcam.bin`` and works as expected.

ascillato2, 2021-04-22:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-04-22 by ascillato2

AlphaXMines, 2021-04-22:
Hello Arendst.
Thanks a lot, you're right, with the correct binary it works. Sorry for that.