arendst/Tasmota #11851: Rule with several conditions stops on first condition (on 9.4.0.1 was working on 9.3.1)

gerardovf, 2021-04-22:
### PROBLEM DESCRIPTION
_I'm using the 'display' version_
On every message from a SI7021 a rule is executed to show Temperature/Humidity/DewPoint

**RULE1
ON TELE-SI7021#TEMPERATURE DO DISPLAYTEXT [s1l1c1T][s1l1c14t] [s1l3c1]TEMP.: %VALUE%C ENDON
ON TELE-SI7021#HUMIDITY DO DISPLAYTEXT [s1l4c1]HUM.: %VALUE%% ENDON
ON TELE-SI7021#DEWPOINT DO DISPLAYTEXT [s1l5c1]DEW PT.: %VALUE%C ENDON**

On version 9.3.1 it's shown in console:
20:34:18.176 RUL: SYSTEM#BOOT performs "DISPLAYTEXT [s1l1c1T][s1l1c14t]"
20:34:18.187 MQT: gvf/cega/stat/salonBlinds/RESULT = {"DisplayText":"[s1l1c1T][s1l1c14t]"}
20:34:22.425 MQT: gvf/cega/tele/salonBlinds/STATE = {"Time":"2021-04-22T20:34:22","Uptime":"0T00:00:09","UptimeSec":9,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"ON","Wifi":{"AP":1,"SSId":"VIRGIN-telco_63D6","BSSId":"B8:D5:26:0C:63:D6","Channel":10,"RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:03"}}
20:34:22.463 MQT: gvf/cega/tele/salonBlinds/SENSOR = {"Time":"2021-04-22T20:34:22","SI7021":{"Temperature":21.4,"Humidity":43.4,"DewPoint":8.4},"TempUnit":"C"}
20:34:22.477 RUL: TELE-SI7021#TEMPERATURE performs "DISPLAYTEXT [s1l1c1T][s1l1c14t] [s1l3c1]TEMP.: 21.4C"
20:34:22.489 MQT: gvf/cega/stat/salonBlinds/RESULT = {"DisplayText":"[s1l1c1T][s1l1c14t] [s1l3c1]TEMP.: 21.4C"}
20:34:22.501 RUL: TELE-SI7021#HUMIDITY performs "DISPLAYTEXT [s1l4c1]HUM.: 43.4%"
20:34:22.514 MQT: gvf/cega/stat/salonBlinds/RESULT = {"DisplayText":"[s1l4c1]HUM.: 43.4%"}
20:34:22.526 RUL: TELE-SI7021#DEWPOINT performs "DISPLAYTEXT [s1l5c1]DEW PT.: 8.4C"
20:34:22.537 MQT: gvf/cega/stat/salonBlinds/RESULT = {"DisplayText":"[s1l5c1]DEW PT.: 8.4C"}

On version 9.4.0.1 it's shown in console:
21:08:36.187 RUL: SYSTEM#BOOT performs "DISPLAYTEXT [s1l1c1T][s1l1c14t]"
21:08:36.224 MQT: gvf/cega/stat/salonBlinds/RESULT = {"DisplayText":"[s1l1c1T][s1l1c14t]"}
21:08:40.442 MQT: gvf/cega/tele/salonBlinds/STATE = {"Time":"2021-04-22T21:08:40","Uptime":"0T00:00:09","UptimeSec":9,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","Wifi":{"AP":1,"SSId":"VIRGIN-telco_63D6","BSSId":"B8:D5:26:0C:63:D6","Channel":10,"RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:03"}}
21:08:40.480 MQT: gvf/cega/tele/salonBlinds/SENSOR = {"Time":"2021-04-22T21:08:40","SI7021":{"Temperature":20.9,"Humidity":44.4,"DewPoint":8.3},"TempUnit":"C"}
21:08:40.496 RUL: TELE-SI7021#TEMPERATURE performs "DISPLAYTEXT [s1l1c1T][s1l1c14t] [s1l3c1]TEMP.: 20.9C"
21:08:40.536 MQT: gvf/cega/stat/salonBlinds/RESULT = {"DisplayText":"[s1l1c1T][s1l1c14t] [s1l3c1]TEMP.: 20.9C"}

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1
- [ ] Tasmota binary firmware version number used: 9.4.0.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
21:12:10.191 CMD: Backlog Template; Module; GPIO 255
21:12:10.234 MQT: gvf/cega/stat/salonBlinds/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
21:12:10.452 MQT: gvf/cega/stat/salonBlinds/RESULT = {"Module":{"18":"Generic"}}
21:12:10.705 MQT: gvf/cega/stat/salonBlinds/RESULT = {"GPIO0":{"225":"Relay2"},"GPIO1":{"0":"None"},"GPIO2":{"224":"Relay1"},"GPIO3":{"0":"None"},"GPIO4":{"608":"I2C SCL"},"GPIO5":{"640":"I2C SDA"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"227":"Relay4"},"GPIO13":{"36":"Button5"},"GPIO14":{"1248":"SI7021"},"GPIO15":{"226":"Relay3"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
21:10:37.008 CMD: Backlog Rule1; Rule2; Rule3
21:10:37.033 MQT: gvf/cega/stat/salonBlinds/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":230,"Free":281,"Rules":"ON TELE-SI7021#TEMPERATURE DO DISPLAYTEXT [s1l1c1T][s1l1c14t] [s1l3c1]TEMP.: %VALUE%C ENDON ON TELE-SI7021#HUMIDITY DO DISPLAYTEXT [s1l4c1]HUM.: %VALUE%% ENDON ON TELE-SI7021#DEWPOINT DO DISPLAYTEXT [s1l5c1]DEW PT.: %VALUE%C ENDON"}}
21:10:37.275 MQT: gvf/cega/stat/salonBlinds/RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":151,"Free":360,"Rules":"ON SYSTEM#BOOT DO DISPLAYTEXT [s1l1c1T][s1l1c14t] ENDON ON BUTTON5#STATE DO BACKLOG POWER5 ON; RULETIMER1 60 ENDON ON RULES#TIMER=1 DO POWER5 OFF ENDON"}}
21:10:37.524 MQT: gvf/cega/stat/salonBlinds/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
21:11:13.821 CMD: Status 0
21:11:13.830 MQT: gvf/cega/stat/salonBlinds/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2","Tasmota3","Tasmota4",""],"Topic":"salonBlinds","ButtonTopic":"0","Power":16,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
21:11:13.879 MQT: gvf/cega/stat/salonBlinds/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota-display.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:02:42","StartupUTC":"2021-04-22T19:08:31","Sleep":50,"CfgHolder":4617,"BootCount":23,"BCResetTime":"2021-04-22T17:25:27","SaveCount":31,"SaveAddress":"FB000"}}
21:11:13.923 MQT: gvf/cega/stat/salonBlinds/STATUS2 = {"StatusFWR":{"Version":"9.4.0.1(display)","BuildDateTime":"2021-04-22T16:14:46","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"414/699"}}
21:11:13.948 MQT: gvf/cega/stat/salonBlinds/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["VIRGIN-telco_63D6",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
21:11:13.983 MQT: gvf/cega/stat/salonBlinds/STATUS4 = {"StatusMEM":{"ProgramSize":588,"Free":412,"Heap":27,"ProgramFlashSize":1024,"FlashSize":16384,"FlashChipId":"1840EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0FAA858E","04049FA1","000000C3","00000000","00004080","00000020","401F8000","00000000"],"Drivers":"1,2,4,5,8,9,10,13,16","Sensors":"1,2,5,6"}}
21:11:14.013 MQT: gvf/cega/stat/salonBlinds/STATUS5 = {"StatusNET":{"Hostname":"salonBlinds-4361","IPAddress":"192.168.1.193","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.155","Mac":"5C:CF:7F:57:71:09","Webserver":2,"WifiConfig":0,"WifiPower":17.0}}
[... 12 lines cut ...]
21:13:36.441 DHT: Timeout waiting for start signal low pulse
21:13:40.402 MQT: gvf/cega/tele/salonBlinds/STATE = {"Time":"2021-04-22T21:13:40","Uptime":"0T00:05:09","UptimeSec":309,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"ON","Wifi":{"AP":1,"SSId":"VIRGIN-telco_63D6","BSSId":"B8:D5:26:0C:63:D6","Channel":10,"RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:03"}}
21:13:40.442 MQT: gvf/cega/tele/salonBlinds/SENSOR = {"Time":"2021-04-22T21:13:40","SI7021":{"Temperature":20.8,"Humidity":44.4,"DewPoint":8.2},"TempUnit":"C"}
21:13:40.457 RUL: TELE-SI7021#TEMPERATURE performs "DISPLAYTEXT [s1l1c1T][s1l1c14t] [s1l3c1]TEMP.: 20.8C"
21:13:40.460 SRC: Rule
21:13:40.463 CMD: Group 0, Index 1, Command "DISPLAYTEXT", Data "[s1l1c1T][s1l1c14t] [s1l3c1]TEMP.: 20.8C"
21:13:40.502 MQT: gvf/cega/stat/salonBlinds/RESULT = {"DisplayText":"[s1l1c1T][s1l1c14t] [s1l3c1]TEMP.: 20.8C"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-04-22:
Using a similar rule, I'm seeing the same issue, also when I split one rule into a separate rule set. Only first trigger is executed, also if they are from different sensors in the same tele payload.

arendst, 2021-04-23:
Thx. Verified too.

I did change something around teleperiod rules so this must be easy to find. Working on it.

closed 2021-04-23 by ascillato2