arendst/Tasmota #11867: On 9.4.0 Skripts are not backuped in *.dmp

lolax80, 2021-04-24:
### PROBLEM DESCRIPTION
if I make a backup of the configuration and on a new device with same bin and I reinstall with *.dmp the script isn't there

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 mini
- [x] Tasmota binary firmware version number used: 9.4.0
  - [ x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Esp-flasher
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
 {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
{"Module":{"0":"Generic"}}
{"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"640":"I2C SDA"},"GPIO5":{"608":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
14:23:08.457 MQT: stat/tasmota_9BCFB7/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_9BCFB7","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
14:23:08.489 MQT: stat/tasmota_9BCFB7/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:06:47","StartupUTC":"2021-04-24T13:16:21","Sleep":50,"CfgHolder":4617,"BootCount":2,"BCResetTime":"2021-03-28T10:02:42","SaveCount":10,"SaveAddress":"FC000"}}
14:23:08.526 MQT: stat/tasmota_9BCFB7/STATUS2 = {"StatusFWR":{"Version":"9.4.0(tasmota)","BuildDateTime":"2021.04.24 12:47:08","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"361/699"}}
14:23:08.546 MQT: stat/tasmota_9BCFB7/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Unifi",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
14:23:08.579 MQT: stat/tasmota_9BCFB7/STATUS4 = {"StatusMEM":{"ProgramSize":695,"Free":1352,"Heap":10,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":3,"Features":["00000407","00A0458E","042E0181","69FE37C3","1A6C0004","D03EF989","0009385C","00201000","00000000"],"Drivers":"1,2,4,6,8,10,13,18,19,22,24,26,27,29,30,35,37,45,50","Sensors":"1,2,4,5,6,7,8,9,10,11,14,15,16,17,18,19,20,21,24,26,27,31,38,39,41,42,44,46,47,53,55,56,57,58,59,63,65,66,67,70,71,72,73,74"}}
14:23:08.623 MQT: stat/tasmota_9BCFB7/STATUS5 = {"StatusNET":{"Hostname":"tasmota_9BCFB7-4023","IPAddress":"192.168.69.84","Gateway":"192.168.69.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.69.254","Mac":"E8:DB:84:9B:CF:B7","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
14:23:08.647 MQT: stat/tasmota_9BCFB7/STATUS6 = {"StatusMQT":{"MqttHost":"xxx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9BCFB7","MqttUser":"xxx","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:23:08.669 MQT: stat/tasmota_9BCFB7/STATUS7 = {"StatusTIM":{"UTC":"2021-04-24T13:23:08","Local":"2021-04-24T14:23:08","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00"}}
14:23:08.690 MQT: stat/tasmota_9BCFB7/STATUS10 = {"StatusSNS":{"Time":"2021-04-24T14:23:08","BME280":{"Temperature":22.6,"Humidity":32.7,"DewPoint":5.3,"Pressure":999.5},"PressureUnit":"hPa","TempUnit":"C"}}
14:23:08.707 MQT: stat/tasmota_9BCFB7/STATUS11 = {"StatusSTS":{"Time":"2021-04-24T14:23:08","Uptime":"0T00:06:47","UptimeSec":407,"Heap":10,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Unifi","BSSId":"18:E8:29:6A:BD:96","Channel":10,"RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-04-24:
Rules and scripts are NOT backuped in *.dmp. Not a issue, it is by design.

lolax80, 2021-04-24:
but on the 9.3.1 it was backuped

Jason2866, 2021-04-24:
If so, not intended.

closed 2021-04-24 by ascillato2