arendst/Tasmota #11869: CJMCU-8128 Temp too high

lolax80, 2021-04-24:
### PROBLEM DESCRIPTION
i tested two sensors
first bme280. temp 22
then CJMCU-8128, SI7921 TEMP=25.4 and BMP280=24.9

the sensor is on 3v on the Wemos d1.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 mini
- [x] Tasmota binary firmware version number used: 9.4.0
  - [ x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Esp-flasher
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
 {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
{"Module":{"0":"Generic"}}
{"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"640":"I2C SDA"},"GPIO5":{"608":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
6:19:40.630 MQT: stat/tasmota_9BCFB7/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_9BCFB7","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
16:19:40.664 MQT: stat/tasmota_9BCFB7/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:07:34","StartupUTC":"2021-04-24T15:12:06","Sleep":50,"CfgHolder":4617,"BootCount":21,"BCResetTime":"2021-03-28T10:02:42","SaveCount":31,"SaveAddress":"3FB000"}}
16:19:40.697 MQT: stat/tasmota_9BCFB7/STATUS2 = {"StatusFWR":{"Version":"9.4.0(tasmota)","BuildDateTime":"2021.04.24 12:47:08","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"361/699"}}
16:19:40.718 MQT: stat/tasmota_9BCFB7/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Unifi",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
16:19:40.754 MQT: stat/tasmota_9BCFB7/STATUS4 = {"StatusMEM":{"ProgramSize":695,"Free":1352,"Heap":9,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":3,"Features":["00000407","00A0458E","042E0181","69FE37C3","1A6C0004","D03EF989","0009385C","00201000","00000000"],"Drivers":"1,2,4,6,8,10,13,18,19,22,24,26,27,29,30,35,37,45,50","Sensors":"1,2,4,5,6,7,8,9,10,11,14,15,16,17,18,19,20,21,24,26,27,31,38,39,41,42,44,46,47,53,55,56,57,58,59,63,65,66,67,70,71,72,73,74"}}
16:19:40.801 MQT: stat/tasmota_9BCFB7/STATUS5 = {"StatusNET":{"Hostname":"tasmota_9BCFB7-4023","IPAddress":"192.168.69.84","Gateway":"192.168.69.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.69.254","Mac":"E8:DB:84:9B:CF:B7","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
16:19:40.824 MQT: stat/tasmota_9BCFB7/STATUS6 = {"StatusMQT":{"MqttHost":"xxx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9BCFB7","MqttUser":"xxx","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
16:19:40.846 MQT: stat/tasmota_9BCFB7/STATUS7 = {"StatusTIM":{"UTC":"2021-04-24T15:19:40","Local":"2021-04-24T16:19:40","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00"}}
16:19:40.870 MQT: stat/tasmota_9BCFB7/STATUS10 = {"StatusSNS":{"Time":"2021-04-24T16:19:40","SI7021":{"Temperature":25.1,"Humidity":37.1,"DewPoint":9.4},"BMP280":{"Temperature":24.7,"Pressure":1000.3},"CCS811":{"eCO2":1007,"TVOC":92},"PressureUnit":"hPa","TempUnit":"C"}}
16:19:40.892 MQT: stat/tasmota_9BCFB7/STATUS11 = {"StatusSTS":{"Time":"2021-04-24T16:19:40","Uptime":"0T00:07:34","UptimeSec":454,"Heap":9,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Unifi","BSSId":"18:E8:29:6A:BD:96","Channel":10,"RSSI":100,"Signal":-38,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2021-04-24:
Tasmota "only" reads the data which is digital transmitted via i2c.
So if sensors readings are off the sensor is a bad one.

sfromis, 2021-04-24:
Be careful about where the temperature sensor is placed. The ESP8266 module acts as a small radiator, with the risk of sensor heating if it is close, or having heat conduction via airflow or materials. Worst is if you have a temperature shield on top of the D1 Mini.

lolax80, 2021-04-24:
The Sensor is 20cm above.
On Same cable with only a bme280 i get the right temp

Jason2866, 2021-04-24:
Trash the sensor it is defect.

ascillato2, 2021-04-24:
Closing this issue as it has been answered.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-04-24 by ascillato2

sfromis, 2021-04-24:
With only 0.5 degrees difference between two sensors, it may just be how things are. It is not unusual for sensor specs to document an expected max deviation from "true" temperature, which very well may be 0.5 degrees, depending on sensor.