arendst/Tasmota #11875: MCP230xx MQTT uniq_id when USE_MCP230xx_OUTPUT is too long and truncate to nonuniq id.

nngg, 2021-04-25:
### PROBLEM DESCRIPTION
MCP230xx MQTT uniq_id when USE_MCP230xx_OUTPUT is too long and truncate to nonuniq id.
this prevents MQTT autoconfig of multiple out pins

04:25:12.856 MQT: homeassistant/sensor/78E658_MCP230XX_MCP230_OUT_OU/config = {"name":"Tasmota MCP230XX MCP230_OUT OUT_D14","stat_t":"tele/t78E658/SENSOR","avty_t":"tele/t78E658/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"78E658_MCP230XX_MCP230_OUT_OU","dev":{"ids":["78E658"]},"unit_of_meas":" ","ic":"mdi:eye","frc_upd":true,"val_tpl":"{{value_json['MCP230XX']['MCP230_OUT']['OUT_D14']}}"} (retained)
04:25:12.894 MQT: homeassistant/sensor/78E658_MCP230XX_MCP230_OUT_OU/config = {"name":"Tasmota MCP230XX MCP230_OUT OUT_D15","stat_t":"tele/t78E658/SENSOR","avty_t":"tele/t78E658/LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"78E658_MCP230XX_MCP230_OUT_OU","dev":{"ids":["78E658"]},"unit_of_meas":" ","ic":"mdi:eye","frc_upd":true,"val_tpl":"{{value_json['MCP230XX']['MCP230_OUT']['OUT_D15']}}"} (retained)

line to change in order to resolve (replace MCP230_OUT with 3 chars to fix):

if (outputcount) {
uint16_t gpiototal = ((uint16_t)gpiob << 8) | gpio;
ResponseAppend_P(PSTR(","MCP230_OUT":{"));
char stt[7];
for (uint32_t pinx = 0; pinx < mcp230xx_pincount; pinx++) {
if (Settings.mcp230xx_config[pinx].pinmode >= 5) {
sprintf(stt, ConvertNumTxt(((gpiototal>>pinx)&1), Settings.mcp230xx_config[pinx].pinmode));
ResponseAppend_P(PSTR(""OUT_D%i":"%s","), pinx, stt);
}
}
ResponseAppend_P(PSTR(""END":1}"));

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- x[ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): generic
- [ x] Tasmota binary firmware version number used: latest
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x ] Flashing tools used: ota
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
  Configuration output here:
06:25:59.179 MQT: stat/tasmota_78E658/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
06:25:59.389 MQT: stat/tasmota_78E658/RESULT = {"Module":{"0":"Generic"}}
06:25:59.644 MQT: stat/tasmota_78E658/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"608":"I2C SCL"},"GPIO2":{"0":"None"},"GPIO3":{"640":"I2C SDA"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
06:26:53.848 MQT: stat/tasmota_78E658/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2"],"Topic":"tasmota_78E658","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
06:26:53.856 MQT: stat/tasmota_78E658/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:28:31","StartupUTC":"2021-04-25T04:58:22","Sleep":50,"CfgHolder":4617,"BootCount":6,"BCResetTime":"2021-04-25T05:53:21","SaveCount":21,"SaveAddress":"F7000"}}
06:26:53.863 MQT: stat/tasmota_78E658/STATUS2 = {"StatusFWR":{"Version":"9.3.1(tasmota)","BuildDateTime":"2021-04-24T20:53:31","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"350/699"}}
06:26:53.870 MQT: stat/tasmota_78E658/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Xiaomi_983F",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A0A000000000000","00000280","00006000","00000000"]}}
06:26:53.881 MQT: stat/tasmota_78E658/STATUS4 = {"StatusMEM":{"ProgramSize":598,"Free":404,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","000013F1","C000F981","00004004","00001000","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6,29"}}
06:26:53.889 MQT: stat/tasmota_78E658/STATUS5 = {"StatusNET":{"Hostname":"tasmota_78E658-1624","IPAddress":"10.0.0.17","Gateway":"10.0.0.138","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"2C:F4:32:78:E6:58","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
06:26:53.895 MQT: stat/tasmota_78E658/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.37","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_78E658","MqttUser":"g","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
06:26:53.904 MQT: stat/tasmota_78E658/STATUS7 = {"StatusTIM":{"UTC":"2021-04-25T05:26:53","Local":"2021-04-25T06:26:53","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"05:39","Sunset":"19:56"}}
06:26:53.915 MQT: stat/tasmota_78E658/STATUS10 = {"StatusSNS":{"Time":"2021-04-25T06:26:53","MCP230XX":{"D0":0,"D1":0,"D2":0,"D3":0,"D4":1,"D5":1,"D6":0,"D7":0,"D8":0,"D9":0,"D10":0,"D11":0,"D12":0,"D13":0,"D14":1,"D15":1,"NG":{"OUT_D14":"ON","OUT_D15":"ON","END":1}}}}
06:26:53.925 MQT: stat/tasmota_78E658/STATUS11 = {"StatusSTS":{"Time":"2021-04-25T06:26:53","Uptime":"0T00:28:31","UptimeSec":1711,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"Xiaomi_983F","BSSId":"88:C3:97:47:98:40","Channel":4,"RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-04-25:
Please, can you provide a PR with your proposed change?

nngg, 2021-04-25:
What is PR?

ascillato, 2021-04-25:
A PR is a pull request. It is a proposed code modification.

You can do that here in tasmota's github:

1- Go to the file you want to modify.
2- Click on the pencil (top right of the page)
3- Do the modifications you want to the file.
4- click on the green button on the bottom
5- click on propose pull request and you are done

Then, your code will be automatically tested and we will review it.

ascillato, 2021-04-28:
Hi,

Please, can you help us and make a PR for this?

Thanks

ascillato2, 2021-04-28:
Moving to Ideas in Discussions for further analysis.

closed 2021-04-28 by ascillato2