arendst/Tasmota #11883: Shelly 1 web interface unresponsive after 9.4.0

fmarzocca, 2021-04-25:
After upgrade to 9.4.0 my 2 shelly1 devices are no more reachable through web interface. Mqtt is working but I can’t reach the configuration page. All other devices (including a sonoff mini) are correctly working.

closed 2021-04-25 by close-issue-app[bot]

ascillato, 2021-04-25:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md)

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

fmarzocca, 2021-04-25:
Hi Adrian,

thanks for your answer. I had already reported this in the “lounge”, but couldn’t get any useful help. As the devices were working fine with the previous version, I thought it is a bug related to the specific Shelly1 devices, as the other are correctly working,