arendst/Tasmota #11884: lost cam with 9.4.0 to aithinker CAM

alpha13fr, 2021-04-25:
Upgrade from 9.3.1 to 9.4.0 and no more cam
good template install

closed 2021-04-25 by close-issue-app[bot]

ascillato, 2021-04-25:
Hi,

Duplicated issue. Please check that you have upgraded to the correct firmware and not just tasmota32.bin that don't have cam support.

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md)

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato, 2021-04-25:
https://github.com/arendst/Tasmota/issues/11835