arendst/Tasmota #11891: EZO, MQTT, and sensor78 commands return unknow

stevewull, 2021-04-26:
### PROBLEM DESCRIPTION
Using the Atlas Scientific EC, pH, and RTD probes and three PMPs. Huzzah is flashed with Tasmota 9.3.1 sensors. Works great. I can send sensor78 commands from the console or over http. But when I send a sensor78 command over MQTT it returns Command : Unknown.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Adafruit Feather HUZZAH 
- [x] Tasmota binary firmware version number used: 9.3.1
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: Compiled with Gitpod, flashed with Tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
01:56:21.668 MQT: stat/Hydro1_Station/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
01:56:21.879 MQT: stat/Hydro1_Station/RESULT = {"Module":{"18":"Generic"}}
01:56:22.134 MQT: stat/Hydro1_Station/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"640":"I2C SDA"},"GPIO5":{"608":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"3872":"Output Lo"},"GPIO13":{"3872":"Output Lo"},"GPIO14":{"3872":"Output Lo"},"GPIO15":{"3840":"Output Hi"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
0

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
Not using rules

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
01:57:19.798 MQT: stat/Hydro1_Station/STATUS = {"Status":{"Module":18,"DeviceName":"Hydro1 Station","FriendlyName":["Hydro1 Station"],"Topic":"Hydro1_Station","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
01:57:19.806 MQT: stat/Hydro1_Station/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T02:19:49","StartupUTC":"2021-04-25T22:37:30","Sleep":50,"CfgHolder":4617,"BootCount":27,"BCResetTime":"2021-04-18T22:40:41","SaveCount":44,"SaveAddress":"F8000"}}
01:57:19.812 MQT: stat/Hydro1_Station/STATUS2 = {"StatusFWR":{"Version":"9.3.1(tasmota)","BuildDateTime":"2021-04-18T21:36:44","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"432/699"}}
01:57:19.819 MQT: stat/Hydro1_Station/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Lares2",""],"TelePeriod":20,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
01:57:19.830 MQT: stat/Hydro1_Station/STATUS4 = {"StatusMEM":{"ProgramSize":531,"Free":472,"Heap":28,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0FE0458E","04260001","00000003","00000000","C000E981","10000004","000017FF","00000000"],"Drivers":"1,2,4,6,8,9,10,12,13,18,19,22,24,26,27,29,30,35,37,45","Sensors":"1,2,78"}}
01:57:19.836 MQT: stat/Hydro1_Station/STATUS5 = {"StatusNET":{"Hostname":"Hydro1_Station","IPAddress":"10.0.10.168","Gateway":"10.0.10.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.10.1","Mac":"84:CC:A8:84:26:1B","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
01:57:19.843 MQT: stat/Hydro1_Station/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.10.41","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_84261B","MqttUser":"mosquitto_things","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
01:57:19.852 MQT: stat/Hydro1_Station/STATUS7 = {"StatusTIM":{"UTC":"2021-04-26T00:57:19","Local":"2021-04-26T01:57:19","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"05:37","Sunset":"19:57"}}
01:57:19.861 MQT: stat/Hydro1_Station/STATUS10 = {"StatusSNS":{"Time":"2021-04-26T01:57:19","EZO-1":{"pH":5.90},"EZO-2":{"EC":2123.000},"EZO-3":{"Temperature":18.9},"EZO-4":{"Pumped":0.0},"EZO-5":{"Pumped":0.0},"EZO-6":{"Pumped":0.0},"EZO-7":{"Pumped":0.0},"TempUnit":"C"}}
01:57:19.869 MQT: stat/Hydro1_Station/STATUS11 = {"StatusSTS":{"Time":"2021-04-26T01:57:19","Uptime":"0T02:19:49","UptimeSec":8389,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Lares2","BSSId":"C8:03:F5:4E:B9:19","Channel":8,"RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
01:58:28.926 CMD: Group 0, Index 1, Command "SENSOR78-", Data "i"
01:58:28.932 MQT: stat/Hydro1_Station/RESULT = {"Command":"Unknown"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
see the screen shots below
1. Compile tasmota-sensors.bin. Include the EZO sensors
2. Set to Generic Module (18) and GPIO pins as in screenshot below
3. Configure MQTT (topic in my example is Hydro1_Station)
4. Using an MQTT client, publish the command sensor78-1 i
5. It returns:
{
  "Command" : "Unknown"
}

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Is should return sensor type and firmware version of the EZO sensor at index 1. For example:
{"Sensor":"?I,pH,2.13"}

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
Module config:
[image]

issuing the Module command works:
[image]
[image]

the sensor78 commands do not work
[image]
[image]

To confirm the commands, here is the result when issued from the consult:
[image]

Also, works using http:
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-04-26:
No bug here. There is no command "SENSOR78-1". Correct syntax for MQTT is to have the command name (Sensor78) in the topic, and the argument (-1) in the payload. It may be less confusing if you always use the command name Backlog in the topic, and then put your desired command into the payload.

Either
Topic `cmnd/Hydro1_Station/Backlog` payload `sensor78-1`
or
Topic `cmnd/Hydro1_Station/Sensor78` payload `-1`

stevewull, 2021-04-26:
Hi sfromis. Thanks for the clarification.  My mistake.

closed 2021-04-26 by stevewull