arendst/Tasmota #11896: TempRes should not be affecting Humidity resolution in Tuya MCU

blakadder, 2021-04-26:
### PROBLEM DESCRIPTION
TempRes command is affecting humidity reading resolution in TuyaMCU after #11513 

Having both temperature and humidty on the same device with different resolutions makes this unusuable

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): TUYAMCU
- [x] Tasmota binary firmware version number used: 9.4.0.2
  - [x] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: PiO
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
TuyaMCU 71,105; TuyaMCU 73,106;
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
```
- [x] Provide the output of this command: `Status 0`:
```lua
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
```

### TO REPRODUCE
Use TempRes on a TuyaMCU device with both temp and humidity sensors configured

### EXPECTED BEHAVIOUR
TempRes should affect temperature resolution and HumRes the humidity if needed. Please revert commit

arendst, 2021-04-26:
Pls retry. I don't have Tuya so can't chk.

blakadder, 2021-04-26:
nope, did not help

JSON and webUI still show the same and HumRes has no effect

arendst, 2021-04-26:
What happens if you set tempres to 0 and humres to 0?

Pls show some JSON output here.

arendst, 2021-04-26:
Pls retry latest changes.

blakadder, 2021-04-26:
backlog tempres 0 ;humres 0 : `{"Time":"2021-04-26T17:41:21","TuyaSNS":{"Temperature":245,"Humidity":44},"TempUnit":"C"}`

backlog tempres 1 ;humres 0 : {"Time":"2021-04-26T17:42:11","TuyaSNS":{"Temperature":24.5,"Humidity":4.1},"TempUnit":"C"}

arendst, 2021-04-26:
OK. I think latest change should solve this.

blakadder, 2021-04-26:
spoke a bit too soon:

```
17:58:20.711 MQT: tele/siren/SENSOR = {"Time":"2021-04-26T17:58:20","TuyaSNS":{"Temperature":23.0,"Humidity":44},"TempUnit":"C"}
17:58:24.774 MQT: tele/siren/SENSOR = {"TuyaSNS":{"Humidity":4.3}}
```

one topic still reports the broken way

closed 2021-04-26 by blakadder

arendst, 2021-04-27:
And another try....

blakadder, 2021-04-27:
solved now! 👍