arendst/Tasmota #11898: DS18B20 randomly report CRC error -> Incompete STATUS message

stefanbode, 2021-04-26:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
on ESP32 precompiled build I got randomly CRC errors. I try to fix this on the hardware side, but also the software side may need an enhancement. I assume that there are magnetic problems though the motors that even a shielded cable does not help. Already tried 3.3V and 5V and different GPIO without getting rid of the hardware problem.

My problem is, that the status message just leave out the sensor, also if the problem to read the sensor in nearly all cases works the second time. For me it would help if the status reports the last value or at least try 3 times before skipping the sensor. There are 3 DS18B20 on onewire. Cable length is about 2m.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): ESP32 mini
- [ x] Tasmota binary firmware version number used: _9.2.1.2
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
```
14:33:32.955 MQT: hm/status/ESP_F50A98/RESULT = {"WebLog":4}
14:33:33.621 CFG: Saved, Count 2404, Bytes 4096
14:33:34.926 DSB: Sensor CRC error
14:33:34.927 SNS: DS18B20-3 missed 1
14:33:36.482 WIF: Checking connection...
14:33:36.745 DSB: Sensor CRC error
14:33:36.913 DSB: Sensor CRC error
14:33:36.915 SNS: DS18B20-2 missed 1
14:33:36.929 DSB: Sensor CRC error
14:33:36.931 SNS: DS18B20-3 missed 1
14:33:38.917 DSB: Sensor CRC error
14:33:38.919 SNS: DS18B20-2 missed 2
14:33:40.847 DSB: Sensor CRC error
14:33:44.933 DSB: Sensor CRC error
14:33:44.934 SNS: DS18B20-3 missed 1
14:33:44.950 DSB: Sensor CRC error
14:33:46.932 DSB: Sensor CRC error
14:33:46.933 SNS: DS18B20-3 missed 1
14:33:49.076 DSB: Sensor CRC error
14:33:49.904 MQT: hm/status/ESP_F50A98/STATE = {"Time":"2021-04-26T14:33:49","Uptime":"0T00:07:44","UptimeSec":464,"Heap":212,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":100,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","POWER8":"OFF","POWER9":"OFF","POWER10":"OFF","POWER11":"OFF","POWER12":"OFF","POWER13":"ON","POWER14":"ON","POWER15":"ON","POWER16":"OFF","POWER17":"OFF","Channel17":100,"POWER18":"ON","Channel18":60,"Color":"FF99","Wifi":{"AP":1,"SSId":"BodeNet_Keller","BSSId":"38:10:D5:14:E6:70","Channel":6,"RSSI":42,"Signal":-79,"LinkCount":1,"Downtime":"0T00:00:06"}}
14:33:50.003 DSB: Sensor CRC error
14:33:50.015 MQT: hm/status/ESP_F50A98/SENSOR = {"Time":"2021-04-26T14:33:49","DS18B20-1":{"Id":"051760A050FF","Temperature":33.2},"DS18B20-2":{"Id":"0316B55F59FF","Temperature":25.6},"Shutter1":{"Position":93,"Direction":0,"Target":93},"Shutter2":{"Position":0,"Direction":0,"Target":0},"TempUnit":"C"}
14:33:50.929 DSB: Sensor CRC error
14:33:50.930 SNS: DS18B20-3 missed 3
14:33:53.215 DSB: Sensor CRC error
14:33:53.217 SNS: DS18B20-1 missed 1
14:33:53.534 DSB: Sensor CRC error
14:33:53.548 DSB: Sensor CRC error
14:33:54.931 DSB: Sensor CRC error
14:33:54.932 SNS: DS18B20-3 missed 2
```

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2021-04-27:
The 100nF did not make the job. Next will be the resistor.

stefanbode, 2021-04-30:
@arendst : Theo, is there a change "possible" that on an error it retries 3 times before coming back with an error? Currently, a CRC Error is just reported.

arendst, 2021-04-30:
I think so yes. The ESP32 uses a different driver anyway as the esp8266 driver failed misserably on the esp32.

arendst, 2021-04-30:
Oh BTW what happens if you enable mean calculation with command ``SO126 1``. It might filter the CRC errors out.

stefanbode, 2021-05-03:
i tried setoption127, but still there in the SENSOR message, one of the three sensors is missing from time to time. Anyhow good idea... And maybe I'm wrong but for me, it feels like the error came less often.

```
10:27:33.003 MQT: hm/status/ESP_F50A98/RESULT = {"WebLog":4}
10:27:33.584 CFG: Saved, Count 2757, Bytes 4096
10:27:42.598 DSB: Sensor CRC error
10:27:42.599 SNS: DS18B20-2 missed 1
10:27:42.613 DSB: Sensor CRC error
10:27:42.613 SNS: DS18B20-3 missed 1
10:27:44.594 DSB: Sensor CRC error
10:27:44.595 SNS: DS18B20-2 missed 2
10:27:48.396 WIF: Checking connection...
10:27:48.607 DSB: Sensor CRC error
10:27:48.608 SNS: DS18B20-3 missed 1
10:27:50.606 DSB: Sensor CRC error
10:27:50.607 SNS: DS18B20-3 missed 1
10:27:51.586 MQT: hm/status/ESP_F50A98/STATE = {"Time":"2021-05-03T10:27:51","Uptime":"5T05:51:12","UptimeSec":453072,"Heap":221,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":11,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","POWER8":"OFF","POWER9":"OFF","POWER10":"OFF","POWER11":"OFF","POWER12":"OFF","POWER13":"ON","POWER14":"ON","POWER15":"ON","POWER16":"OFF","POWER17":"OFF","Channel17":1,"POWER18":"ON","Channel18":73,"Color":"03BA","Wifi":{"AP":1,"SSId":"BodeNet_Keller","BSSId":"38:10:D5:14:E6:70","Channel":6,"RSSI":46,"Signal":-77,"LinkCount":11,"Downtime":"0T00:01:37"}}
10:27:51.649 MQT: hm/status/ESP_F50A98/SENSOR = {"Time":"2021-05-03T10:27:51","DS18B20-1":{"Id":"051760A050FF","Temperature":33.7},"DS18B20-2":{"Id":"0316B55F59FF","Temperature":36.9},"DS18B20-3":{"Id":"051760AF75FF","Temperature":33.9},"Shutter1":{"Position":0,"Direction":0,"Target":0},"Shutter2":{"Position":0,"Direction":0,"Target":0},"TempUnit":"C"}
10:27:54.408 DSB: Sensor CRC error
10:27:54.436 DSB: Sensor CRC error
10:27:54.610 DSB: Sensor CRC error
10:27:54.611 SNS: DS18B20-3 missed 2
10:28:02.654 DSB: Sensor CRC error
```

stefanbode, 2021-05-03:
Lowering the 4.7k resistor is still on open topic.

ascillato2, 2021-05-03:
> The error only occurs if I switch on the pump that is controlled by the frequency converter. So, it is 100% an interference issue.

Great to know that you have found the source of this issue !

closed 2021-05-03 by ascillato2

rt400, 2021-05-28:
i have the same issue but i don't have any pump like @stefanbode ...

rt400, 2021-06-03:
i fix the problem by replace the ressistor to 600ohm