arendst/Tasmota #11930: Prometheus metrics parse error on DS18x20

twollweber, 2021-04-28:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When using USE_PROMETHEUS, scraping fails with an error message

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 mini
- [x] Tasmota binary firmware version number used: 9.4.0.2
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
17:21:02.908 MQT: stat/tasmota_20C1FB/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
17:21:03.139 MQT: stat/tasmota_20C1FB/RESULT = {"Module":{"18":"Generic"}}
17:21:03.392 MQT: stat/tasmota_20C1FB/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"32":"Button1"},"GPIO5":{"1312":"DS18x20"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"608":"I2C SCL"},"GPIO13":{"640":"I2C SDA"},"GPIO14":{"33":"Button2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1. Compile with '#define USE_PROMETHEUS'.
2. Configure and attach DS18x20, making sure that it appears in SENSOR data.
3. Check that http://ip.of.your.tasmota/metrics contains a metric like 
"# TYPE tasmota_sensors_id_ gauge
tasmota_sensors_id_{sensor="ds18b20-1"} 0000064B4013"
4. Configure Prometheus to scape http://ip.of.your.tasmota/metrics.
5. Find error message in Prometheus similar to "strconv.ParseFloat: parsing „0000064B4013“: invalid syntax"

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Prometheus shall process the DS18x20 ID (which is the culprit) either as a label or not at all.
In addition, the type shall be set to "untyped".
Therefore, the metrics output shall contain a metric "# TYPE tasmota_sensors_id_untyped gauge
tasmota_sensors_id_untyped{sensor="ds18b20-1",id="0000064B4013"} 1"

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
The DS18x20 ID can be treated like the existing pseudo-metrics (e.g. Wi-Fi). This will keep the information of the hardware ID but avoid treating the hex serial number as a float.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-04-28:
Hi,

Please, can you elaborate a little more your issue?

How is the data your Prometheus is receiving and How it should be?

twollweber, 2021-04-28:
As I put in "TO REPRODUCE" and "EXPECTED BEHAVIOUR":
Prometheus scrapes
> # TYPE tasmota_sensors_id_ gauge
> tasmota_sensors_id_{sensor="ds18b20-1"} 0000064B4013

It should be
> # TYPE tasmota_sensors_id_untyped gauge
> tasmota_sensors_id_untyped{sensor="ds18b20-1",id="0000064B4013"} 1

I have modified the code already in my fork. It does exactly this.
Currently, I am learning how to use Git to provide that PR.

ascillato2, 2021-04-28:
Thanks for providing a PR. Very appreciated.

> without the ' – how do I format the lines not to be 'header text"?

using 3 times the symbol ' like in the code snippets in the template.

PR #11931 (2021-04-28): Fix Prometheus metrics parse error on DS18x20 and align other metrics
## Description:

Fix Prometheus metrics parse error on DS18x20
Align tasmotaglobal metrics for Prometheus to other metrics' syntax

**Related issue (if applicable):** fixes #11930, #11933

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works w [... cut]
closed 2021-04-28 by ascillato2