arendst/Tasmota #11942: Tasmota 9.4 SSID page won't connect / deletes details and goes back to being AP 192.168.4.1

1Drains, 2021-04-29:
Hello all of you smart people.
I am not a tech savy user but love the Tasmota world, Thanks Theo! (and all contributers.)

I have started reflashing all of my 35 Tasmota Flashed Sonoffs, and ModeMCU's etc with 9.4.
(Why you ask ??? because I have started a new Home Assistant and wanted to declutter my config pages by taking advantage of the auto integrations etc.)

Problem 1.
What I have noticed is that with NEW flashed devices, when connecting to the AP of Tasmota-2356 I than go to 192.168.4.1 when I can select my wifi and enter the password...
It repeatedly comes back saying failed to connect...   Please don't ask if my details are correct! lol yes they are...
(THIS HAS NEVER WORKED FOR ME...)

The work around has been to flash 9.1 and then do OTA to 9.4...

Problem 2.
I have had a D1Mini Pro with 9.4 disconnect from the wifi during the night and then it would only reconnect when I changed the wifi channel in the router from auto to say channel 2 or 3... either worked getting it reconnected, then I changed the router back to auto channel and all was fine. (It controls the heating... I got in trouble with my wife for the cold house in the morning!)
It also has happened to a couple of sonoffs and nodeMCU's / Wemos D1 Mini's which the changing of the settings in the router brings them back...

However I have a Sonoff Basic controlling my Garage lights, with a relay controlling the garage rollor door, via a relay
on the TX pin
a DHT11 on GPIO 14
and a reed switch on RX pin    (However I haven't actually physically connected the reed switch yet other than testing after config.)

When I came home yesterday the Sonoff had gone back to AP mode and I can't enter my credentials, as per above...
So I removed it and installed another Sonoff basic in its place with the same config (I'm getting good at soldering!)
Came home today and same again.

Have read as much as I could on High And Low pins on boot but couldn't find any for my set up to indicate an issue.

Not sure what to do?
Thanks if you can help.
Adrian

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): __2___
- [X ] Tasmota binary firmware version number used: _Tasmota Bin 9.4____
  - [ X] Pre-compiled
  - [ ] Self-compiled
- [ X] Flashing tools used: _FlashESP8266   ESP Home flasher for windows... I use both____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
  Not available...

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2021-04-29:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-04-29 by ascillato2

1Drains, 2021-04-29:
Hi, thanks,
I have removed the silly stuff. ( should have know to do to Discord or Community...)
However as the 9.4 wifi config page is an issue I would have thought this would be where to raise the issue?

Should I post this on Support Chat?
Thanks  for the help