arendst/Tasmota #11953: V9xx unresponsive on vlan

ardenking, 2021-04-29:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I moved all my 20+ sonoffs to their own vlan without internet access as they only need to see my Mqtt broker on HA

All the older ones on different levels v 8 worked perfectly but the ones in the 9xx steam became unresponsive / very slow

Moved them back to the main network worked fine downgraded to 8.5 and working fine on vlan

Any idea what could cause this on newer levels

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): ____sonoff basic_
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled precompiled 
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-04-29:
Hi,

In order to help you, more information is needed. Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

sfromis, 2021-04-29:
If you have no NTP server available, that could cause trouble.

ascillato2, 2021-04-30:
Hi,

When you have some time, please provide more information or address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4).

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2021-04-30 by ascillato2

pessorrusso, 2021-05-27:
I had the same issue. I fixed it by providing a local NTP, when there is no NTP available the device becomes extremely slow.

ascillato, 2021-05-27:
Such function already exists.

You can disable NTP by using command `time 0` as explained in the docs.

tomose, 2021-07-01:
I am curious why was the issue closed. Seems like the optimal solution is that the devices should not become slow or non-functional just because they can't reach an NTP server. They should also not spam a network with NTP requests. This still happens on v9.5.