arendst/Tasmota #11965: Telegram Notificat: tmstate gets resetted on reboot

Shadouw, 2021-04-30:
### PROBLEM DESCRIPTION
I've compiled my own Tasmota version with Telegram support enabled: `#define USE_TELEGRAM`
Next I've configured it as described in [a tickets comment|https://github.com/arendst/Tasmota/issues/8619#issuecomment-744418530]. `tmsend` works fine and I get Telegram notification.

I've noticed that `tmstate` gets resetted whenever the Tasmota device restarts. 

To be honest: I'm working on Tasmota 9.3.1 and have not tested on development branch. But I had a quick look at the development source code:

```
enum SettingsTextIndex { SET_OTAURL,
...
                         SET_DEVICENAME,
                         SET_TELEGRAM_TOKEN, SET_TELEGRAM_CHATID,
``` 
There is no `SET_TELEGRAM_TMSTATE` setting and `void CmndTmState(void)` in `xdrv_40_telegram.ino` doesn't call `bool SettingsUpdateText(uint32_t index, const char* replace_me)`.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 Mini
- [x] Tasmota binary firmware version number used: 9.3.1
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
```
- [ ] Provide the output of this command: `Status 0`:
```lua
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
```

### TO REPRODUCE
Compile your firmware with `#define USE_TELEGRAM`
Configure `tmtoken` and `tmchatid`
Set `tmstate 1` to enable sending telegram messages.
Try `tmsend Testmessage` and check if your configuration works.

Reboot the Tasmota device.

Again try `tmsend Another test message`. You will not receive a message and in console you get a send failure. 
Check `tmstate` and you will notice:
```
{"TmState":{"Send":"OFF","Receive":"OFF","Echo":"OFF"}}
```

### EXPECTED BEHAVIOUR
`tmtoken` and `tmchatid` are persistent. I would expect that `tmstate` is as well persistent.

### SCREENSHOTS

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-04-30:
As designed.

Learn to use rules to enable telegram on reboot.

Shadouw, 2021-04-30:
Thanks for the quick response. I've enabled a rule to workaround the issue. Sorry that I forgot to mention this workaround in the description.

ascillato2, 2021-04-30:
Closing as this request has been added by Theo. Thanks.

closed 2021-04-30 by ascillato2