arendst/Tasmota #11986: ESP32 WS2812 ledX unreliable

marc-gist, 2021-05-01:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Using the latest build (as of yesterday git pull).  ESP32 ws2812 works fine. However, issuing `led10 004400` or the like commands to turn on a specific Led to a specific color sometimes works, sometimes doesn't.  when it doesn't work, what appears to happen is some leds turn off (don't receive correct data).  usually random 1-3 leds.  issuing the same command often "fixes" it, sometime just moves the off leds.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): _ESP32 dev board____
- [ d] Tasmota binary firmware version number used: __9.4.0.3___
  - [ ] Pre-compiled
  - [x ] Self-compiled
- [ x] Flashing tools used: _VC platformio____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
{"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}

{"Module":{"1":"ESP32-DevKit"}}

{"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"1376":"WS2812"},"GPIO18":{"544":"LedLink"},"GPIO19":{"288":"Led1"},"GPIO20":{"0":"None"},"GPIO21":{"640":"I2C SDA1"},"GPIO22":{"608":"I2C SCL1"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO28":{"0":"None"}}
{"GPIO29":{"0":"None"},"GPIO30":{"0":"None"},"GPIO31":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
ot/tasmota/stat/tasmota_370E24/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_370E24","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
19:01:08.032 MQT: iot/tasmota/stat/tasmota_370E24/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:02:52","StartupUTC":"2021-05-01T22:58:16","Sleep":50,"CfgHolder":4617,"BootCount":13,"BCResetTime":"2021-05-01T08:53:31","SaveCount":91}}
19:01:08.065 MQT: iot/tasmota/stat/tasmota_370E24/STATUS2 = {"StatusFWR":{"Version":"9.4.0.3(tasmota-mdr)","BuildDateTime":"2021-05-01T08:52:32","Core":"1_0_6","SDK":"v3.3.5-1-g85c43024c","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6","CR":"399/699"}}
19:01:08.087 MQT: iot/tasmota/stat/tasmota_370E24/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IOTWAN",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008008","0C05C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
19:01:08.123 MQT: iot/tasmota/stat/tasmota_370E24/STATUS4 = {"StatusMEM":{"ProgramSize":1458,"Free":1984,"Heap":209,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashFrequency":40,"FlashMode":2,"Features":["00000809","9FCAC587","00052001","B7F7BFCF","45DA9BF1","68360DC7","00084052","20200000","0000002D"],"Drivers":"1,2,3,4,5,8,9,10,11,12,14,17,24,26,27,29,34,50,52","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,29,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,87"}}
19:01:08.170 MQT: iot/tasmota/stat/tasmota_370E24/STATUS5 = {"StatusNET":{"Hostname":"tasmota_370E24-3620","IPAddress":"10.20.1.199","Gateway":"10.20.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.20.1.1","Mac":"7C:9E:BD:37:0E:24","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
19:01:08.193 MQT: iot/tasmota/stat/tasmota_370E24/STATUS6 = {"StatusMQT":{"MqttHost":"10.20.1.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_370E24","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
19:01:08.216 MQT: iot/tasmota/stat/tasmota_370E24/STATUS7 = {"StatusTIM":{"UTC":"2021-05-01T23:01:08","Local":"2021-05-01T19:01:08","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":99,"Sunrise":"06:11","Sunset":"20:21"}}
19:01:08.243 MQT: iot/tasmota/stat/tasmota_370E24/STATUS10 = {"StatusSNS":{"Time":"2021-05-01T19:01:08","ESP32":{"Temperature":70.6},"TempUnit":"C"}}
19:01:08.256 MQT: iot/tasmota/stat/tasmota_370E24/STATUS11 = {"StatusSTS":{"Time":"2021-05-01T19:01:08","Uptime":"0T00:02:52","UptimeSec":172,"Heap":209,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":25,"MqttCount":1,"POWER":"OFF","Dimmer":36,"Color":"000E5C","HSBColor":"231,100,36","Channel":[0,5,36],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"IOTWAN","BSSId":"FC:EC:DA:A1:6F:B5","Channel":1,"RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

19:02:01.669 CMD: power1 1
19:02:01.671 SRC: WebConsole from 10.2.1.2
19:02:01.673 CMD: Group 0, Index 1, Command "POWER", Data "1"
19:02:01.694 MQT: iot/tasmota/stat/tasmota_370E24/RESULT = {"POWER":"ON"}
19:02:01.699 MQT: iot/tasmota/stat/tasmota_370E24/POWER = ON
19:02:07.817 WIF: Checking connection...
19:02:10.914 CMD: led1 800000
19:02:10.916 SRC: WebConsole from 10.2.1.2
19:02:10.918 CMD: Group 0, Index 1, Command "LED", Data "800000"
19:02:10.940 MQT: iot/tasmota/stat/tasmota_370E24/RESULT = {"Led1":"800000"}
19:02:20.043 CMD: led2 800000
19:02:20.045 SRC: WebConsole from 10.2.1.2
19:02:20.047 CMD: Group 0, Index 2, Command "LED", Data "800000"
19:02:20.068 MQT: iot/tasmota/stat/tasmota_370E24/RESULT = {"Led2":"800000"}
19:02:23.415 CMD: led10 800000
19:02:23.417 SRC: WebConsole from 10.2.1.2
19:02:23.419 CMD: Group 0, Index 10, Command "LED", Data "800000"
19:02:23.443 MQT: iot/tasmota/stat/tasmota_370E24/RESULT = {"Led10":"800000"}
19:02:27.818 WIF: Checking connection...
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Connect a strip of WS2812 leds (i have 16 leds) and try to set individual leds to a color.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

led changes to set color

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-05-02:
Hi,

Seems to be a power supply issue. Can you use a bigger power supply and re check wiring?

marc-gist, 2021-05-02:
> Hi,
> 
> Seems to be a power supply issue. Can you use a bigger power supply and re check the wirings?

I have a 3A supply, so not that :(
Works fine using the FastLED library and MCP23017 in a custom code, so maybe interference/timing issue with other routines that tasmota does is my guess.

s-hadinger, 2021-05-02:
Can you make sure you have defined `#define USE_WS2812_RMT  0`

Default configuration is to use hardware RMT channel 0. If not, it defaults to bitbang which is not reliable on ESP32 due to interrupts

marc-gist, 2021-05-02:
> Can you make sure you have defined `#define USE_WS2812_RMT 0`
> 
> Default configuration is to use hardware RMT channel 0. If not, it defaults to bitbang which is not reliable on ESP32 due to interrupts

I had that set... so that doesn't appear to be the issue.  I am driving it with 3.3v data (no level shifter), but the cable is like 6 inches, and never had issues driving with 3.3v data with FastLED, or esp8266's and tasmota before.

marc-gist, 2021-05-02:
I did more testing. Powered the strip from 3.3v, still have the same intermittent issues using tasmota :(

s-hadinger, 2021-05-03:
I tried extensively with a M5Stack Atom that has 20 leds and the RMT version is rock solid, no glitch. However big banging has random issues (not often but still occurs). It's maybe not a timing issue, but another bug.

Do you have a logic analyzer at hand to record what is sent by the ESP32?

marc-gist, 2021-05-03:
> 
> Do you have a logic analyzer at hand to record what is sent by the ESP32?

I wish... can you recommend an affordable one? I would defiantly consider getting one to help out in these instances.

From my reading.  The ESP32 using NeoPixelBus should be using the RMT method on the dev kit.  What pin does the Atom use for the WS2812 (I'll look up and see if that changes anything)...  Also could be that they use the 2812C, where I think mine are generic or 2812B.

marc-gist, 2021-05-03:
Doing more testing. Got a fresh WS2812B strip factory ends etc, just to make sure.  Tasmota on 3 different ESP32 from 3 different manufactures (D1 ESP32, TTGO Esp32, generic Dev4 board).  All exibit the same behaviour, with and without a level shifter (74AHCT125).

Here is my custom compile config

```
#ifndef _USER_CONFIG_OVERRIDE_H_
#define _USER_CONFIG_OVERRIDE_H_

// Shrink Bin Size; good for bulbs
// remove things like web timer editing, 
//    sunrise, shelly dimmer, sonoff rf, web server advertise
//#define SHRINK

// ENABLE custom sensor selection
#define SENSOR_CUSTOM
#define MCP_23017_INOUT

// ENABLE WS2812 Led Support
#define WSLED
//used with ESP32 for non bit-bang led control
#define USE_WS2812_RMT 0

// https://tasmota.github.io/docs/Berry-Scripting/
#ifdef ESP32
  #ifdef USE_BERRY
    #undef USE_BERRY
  #endif
  #define USE_BERRY
#endif

#ifdef KEY_HOLD_TIME
  #undef KEY_HOLD_TIME
#endif
#define KEY_HOLD_TIME 12

#ifdef WIFI_IP_ADDRESS
  #undef WIFI_IP_ADDRESS
#endif
#define WIFI_IP_ADDRESS    ""

#ifdef WIFI_SUBNETMASK
  #undef WIFI_SUBNETMASK
#endif
#define WIFI_SUBNETMASK    ""

#ifdef WIFI_GATEWAY
  #undef WIFI_GATEWAY
#endif
#define WIFI_GATEWAY    ""

#ifdef WIFI_DNS
  #undef WIFI_DNS
#endif
#define WIFI_DNS    ""

#ifdef USE_EMULATION_WEMO
  #undef USE_EMULATION_WEMO
#endif

#ifdef USE_EMULATION_HUE
  #undef USE_EMULATION_HUE
#endif

#ifdef USE_MHZ19
  #undef USE_MHZ19
#endif

#ifdef USE_SENSEAIR
  #undef USE_SENSEAIR
#endif
[... 288 lines cut ...]
#endif
#define APP_TIMEZONE 99

#ifdef SHRINK
  
  #ifdef USE_TIMERS_WEB
    #undef USE_TIMERS_WEB
  #endif
  #ifdef USE_SUNRISE
    #undef USE_SUNRISE
  #endif
  #ifdef USE_SONOFF_RF
    #undef USE_SONOFF_RF
  #endif
  #ifdef USE_SHELLY_DIMMER
    #undef USE_SHELLY_DIMMER
  #endif
  #ifdef WEBSERVER_ADVERTISE
    #undef WEBSERVER_ADVERTISE
  #endif
#endif

#endif
//end first ifndef
```

marc-gist, 2021-05-03:
More information.  I went back version to see if any were more stable.  v9.1.0 is way more stable.  v9.3.0 won't even compile as it gets a 404 error grabbing he esp32 framework i think., v9.2.0 same issues.
Particularly when changing the color/leds quickly it gets worse fast.

not sure what changed between v9.1 and 9.2 but it seems that's when stability issues started.

s-hadinger, 2021-05-03:
I was using 9.3 and upgrade to 9.4, it's working flawlessly with Atom. No single glitch. I tried with your settings too, no issue.
GPIO is 27.

You can find cheap Saleae clone of logic analyzers at 10€

marc-gist, 2021-05-04:
> I was using 9.3 and upgrade to 9.4, it's working flawlessly with Atom. No single glitch. I tried with your settings too, no issue.
> GPIO is 27.
> 
> You can find cheap Saleae clone of logic analyzers at 10€

i would suspect that is your WS2812C chips that handle it fine.  do you have any ws2812b and esp32 dev boards? I I have tried many and the issue persists. FastLED works perfect on all devices and strips i have, WLED as well, which I think is based on fastled, or some other library.

s-hadinger, 2021-05-04:
We are using this library: https://github.com/Makuna/NeoPixelBus

You can try to reach the author, there might be an issue in the timings.

ascillato2, 2021-05-16:
Hi, any news on this?

marc-gist, 2021-05-17:
Unfortunately I have not had the time to test the library directly, outside of Tasmota.

closed 2021-05-26 by github-actions[bot]