arendst/Tasmota #12008: Sonoff L1 problem from 9.3.1

djpa3k, 2021-05-03:
### PROBLEM DESCRIPTION
Sonoff L1 does not listen to any commands except on/off. The device responds to sound when turned on and colors can be set with the remote control.  
Using RC (change color) can be seen in the console but not on color pallete .  
Same with 9.4.0 and 9.3.1
With 9.2.0 is everything OK. 
Another older L1 with 9.3.1 works perfectly 

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff L1
- [x] Tasmota binary firmware version number used: 9.4.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
21:19:37.735 CMD: Backlog Template
21:19:37.765 MQT: stat/led_spaln1_2E8CDC/RESULT = {"NAME":"Generic","GPIO":[1,3200,0,3232,1,0,0,0,0,320,1,0,0,0],"FLAG":0,"BASE":70}
21:19:58.061 CMD: Module
21:19:58.068 MQT: stat/led_spaln1_2E8CDC/RESULT = {"Module":{"70":"Sonoff L1"}}
21:20:13.480 CMD: GPIO 255
21:20:13.491 MQT: stat/led_spaln1_2E8CDC/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"3200":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3232":"Serial Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
21:05:27.121 CMD: status 0
21:05:27.124 SRC: WebConsole from 192.168.1.108
21:05:27.127 CMD: Group 0, Index 1, Command "STATUS", Data "0"
21:05:27.134 MQT: stat/led_spaln1_2E8CDC/STATUS = {"Status":{"Module":70,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"led_spaln1_2E8CDC","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
21:05:27.143 MQT: stat/led_spaln1_2E8CDC/STATUS1 = {"StatusPRM":{"Baudrate":19200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:05:12","StartupUTC":"2021-05-03T20:00:15","Sleep":50,"CfgHolder":4617,"BootCount":4,"BCResetTime":"2021-05-03T20:57:21","SaveCount":28,"SaveAddress":"F5000"}}
21:05:27.153 MQT: stat/led_spaln1_2E8CDC/STATUS2 = {"StatusFWR":{"Version":"9.4.0(tasmota)","BuildDateTime":"2021-04-23T10:07:22","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"352/699"}}
21:05:27.160 MQT: stat/led_spaln1_2E8CDC/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["pa3k",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
21:05:27.171 MQT: stat/led_spaln1_2E8CDC/STATUS4 = {"StatusMEM":{"ProgramSize":599,"Free":404,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
21:05:27.182 MQT: stat/led_spaln1_2E8CDC/STATUS5 = {"StatusNET":{"Hostname":"led_spaln1_2E8CDC-3292","IPAddress":"192.168.1.117","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"91.221.110.18","Mac":"C4:DD:57:2E:8C:DC","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
21:05:27.189 MQT: stat/led_spaln1_2E8CDC/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.111","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2E8CDC","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:05:27.200 MQT: stat/led_spaln1_2E8CDC/STATUS7 = {"StatusTIM":{"UTC":"2021-05-03T20:05:27","Local":"2021-05-03T21:05:27","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"05:25","Sunset":"20:07"}}
21:05:27.209 MQT: stat/led_spaln1_2E8CDC/STATUS10 = {"StatusSNS":{"Time":"2021-05-03T21:05:27"}}
21:05:27.219 MQT: stat/led_spaln1_2E8CDC/STATUS11 = {"StatusSTS":{"Time":"2021-05-03T21:05:27","Uptime":"0T00:05:12","UptimeSec":312,"Heap":24,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":100,"Color":"002AFF","HSBColor":"230,100,100","Channel":[0,16,100],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"pa3k","BSSId":"60:32:B1:1F:3D:3A","Channel":11,"RSSI":100,"Signal":-41,"LinkCount":1,"Downtime":"0T00:00:05"}}}
21:05:33.178 CMD: weblog 2

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
21:01:12.608 SRC: WebGui from 192.168.1.108
21:01:12.611 CMD: Group 0, Index 1, Command "HSBCOLOR", Data "114"
21:01:12.619 MQT: stat/led_spaln1_2E8CDC/RESULT = {"POWER":"ON","Dimmer":100,"Color":"19FF00","HSBColor":"114,100,100","Channel":[10,100,0]}
21:01:12.644 SL1: Send AT+UPDATE="sequence":"1620075672320","switch":"on","light_type":1,"colorR":2,"colorG":255,"colorB":0,"bright":100,"mode":1,"sensitive":5,"speed":50
21:01:12.770 SL1: Rcvd 
21:01:12.772 SL1: Send AT+SEND=ok
21:01:13.584 CFG: Saved to flash at F6, Count 27, Bytes 4096
21:01:14.041 HTP: Console

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-05-04:
I cannot reproduce as mine works fine.

Execute command ``hsbcolor 114`` from console:
```
11:47:13.838 CMD: hsbcolor 114
11:47:13.841 SRC: WebConsole from 192.168.2.1
11:47:13.844 CMD: Group 0, Index 1, Command "HSBCOLOR", Data "114"
11:47:13.851 MQT: stat/l1/RESULT = {"POWER":"ON","Dimmer":100,"Color":"19FF00","HSBColor":"114,100,100","Channel":[10,100,0]}
11:47:13.912 SL1: Send AT+UPDATE="sequence":"1620128833280","switch":"on","light_type":1,"colorR":25,"colorG":255,"colorB":0,"bright":100,"mode":1,"sensitive":5,"speed":50
11:47:14.014 SL1: Rcvd AT+RESULT="sequence":"1620128833280"
11:47:14.016 SL1: Send AT+SEND=ok
11:47:14.610 CFG: Saved to flash at FB, Count 1579, Bytes 4096
```
Press button on RC:
```
11:49:12.957 SL1: Rcvd AT+UPDATE="switch":"on","light_type":1,"colorR":255,"colorG":63,"colorB":0,"bright":100,"mode":1,"speed":50,"sensitive":5
11:49:12.960 SRC: Remote
11:49:12.963 CMD: Group 0, Index 2, Command "COLOR", Data "ff3f00"
11:49:12.971 MQT: stat/l1/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF3F00","HSBColor":"15,100,100","Channel":[100,25,0]}
11:49:12.973 SL1: Send AT+SEND=ok
11:49:13.615 CFG: Saved to flash at FA, Count 1580, Bytes 4096
```
Notice the received data from the L1 represented by ``SL1: Rcvd ...``.

Do you receive anything from the L1? In you log it appears it dos not receive anything.

djpa3k, 2021-05-04:
`SL1: Rcvd` is empty (9.4.0, 9.3.1, 9.2.0) but 9.2.0 works OK with empty `SL1: Rcvd`
> 
> Do you receive anything from the L1? In you log it appears it dos not receive anything.

djpa3k, 2021-05-08:
I try compile 9.4.0 with xlgt_05_sonoff_l1.ino before [ac8d2698](https://github.com/arendst/Tasmota/commit/ac8d26982d220b15d8e8c089c26942f12798408b) - color change works OK but `SL1: Rcvd` still empty.

I found another problem: `speed` and `speed2` command does not woks. Same with another piece of L1 where is not problem with color change after 9.3.1 

This output from another piece of L1 with not modified 9.4.0 but speed2/speed problem (transitions):
```
11:30:37.165 CMD: power 0
11:30:37.168 SRC: WebConsole from 192.168.1.108
11:30:37.171 CMD: Group 0, Index 1, Command "POWER", Data "0"
11:30:37.177 MQT: stat/led_tv_E7DDB0/RESULT = {"POWER":"OFF"}
11:30:37.183 MQT: stat/led_tv_E7DDB0/POWER = OFF
11:30:41.033 WIF: Checking connection...
11:30:41.893 CMD: speed2 10
11:30:41.896 SRC: WebConsole from 192.168.1.108
11:30:41.899 CMD: Group 0, Index 2, Command "SPEED", Data "10"
11:30:41.906 MQT: stat/led_tv_E7DDB0/RESULT = {"Speed2":10}
11:30:44.447 CMD: power 1
11:30:44.450 SRC: WebConsole from 192.168.1.108
11:30:44.453 CMD: Group 0, Index 1, Command "POWER", Data "1"
11:30:44.458 SL1: Send AT+UPDATE="sequence":"1620473444203","switch":"on","light_type":1,"colorR":0,"colorG":0,"colorB":255,"bright":100,"mode":1,"sensitive":5,"speed":50
11:30:44.542 MQT: stat/led_tv_E7DDB0/RESULT = {"POWER":"ON"}
11:30:44.547 MQT: stat/led_tv_E7DDB0/POWER = ON
11:30:44.606 SL1: Rcvd AT+RESULT="sequence":"1620473444203"
11:30:44.609 SL1: Send AT+SEND=ok
```

DDock, 2021-05-22:
I was having the same thing after just getting into the Tasmota ecosystem. I downgraded from 9.4.0 to 9.2.0 and it is working properly. I was having issues where I could only control on / off but not color. I didn't see this mentioned but the L1 Lite was always stuck in MUSIC mode, where if you tap the box it would constantly change color. Not sure if the issue with the firmware is due to not honoring music mode toggle. I did attempt to send the L1MusicSync command to turn the mode on / off but neither fixed the issue.

martinsefcik, 2021-05-25:
I can confirm there are multiple issues introduced in newer versions (9.4.0). 9.2.0 is working fine. I guess they were introduced by new L1MusicSync command.
List of issues:
- after plugging power adapter it always starts in music sync mode. Even in cases it was in power off state before unplugging power adapter
- as it was mentioned it is not possible to control brightness and color from web UI
- brightness buttons on remote control does not work correctly. I can change brightness only once and after first press brightness is changed but automatically music mode is enabled and then second press of brightness button does not work I guess due to enabled music mode, so I have to change color on remote control and then I again can once press brightness button and it is working but only after first press so I have to again press any color to disable music mode and again I can use brightness button and so on ...

DDock, 2021-06-19:
Has this issue been fixed. Don't want this to become "stale" if it isn't fixed yet. Don't have any sort of notifications here that is is being worked on or fixed.

khacduy1978, 2021-06-23:
My Sonoff L1 Lite is on 9.5.0 firmware and the above issues are the same.

mhamzahkhan, 2021-06-27:
Also having this issue with my Sonoff L1  on 9.5.0 :(

martinsefcik, 2021-07-22:
How we can help to resolve this issue ?

mdiazgoncalves, 2021-08-02:
> I can confirm there are multiple issues introduced in newer versions (9.4.0). 9.2.0 is working fine. I guess they were introduced by new L1MusicSync command.
> List of issues:
>
>     * after plugging [...]

I have the same issues with Sonoff L1. Can't we disable L1MusicSync command?

arendst, 2021-08-03:
Try latest development version available after T + 2 hours.

martinsefcik, 2021-08-04:
Tested it with 9.5.0.5 | 20210804 18:17 and I guess all described issues are still present.

khacduy1978, 2021-08-06:
> I tested it with 9.5.0.5 | 20210804 18:17 and I guess all described issues are still present.

I can confirm these issues do not resolve after testing the latest development version.

mediarath, 2021-08-23:
Same with the 9.5.0.7
3x of „new“ Sonoff SpiderZ did not work.
A older one works in the newer versions above 9.2 properly.

arendst, 2021-08-23:
Now we're getting somewhere.

I'll revisit the changes made between the two versions and try to come up with a solution.

arendst, 2021-08-25:
Fixed in latest dev branch.

See discussion here https://github.com/arendst/Tasmota/discussions/12930#discussioncomment-1233209