arendst/Tasmota #12009: Framework for custom "plugins"

bilogic, 2021-05-04:
Hi,

I have a "problem", but it isn't a bug with Tasmota, thought I make this clear first.
I was also concerned it might get lost in Discord, that's the reason why I'm creating this issue here.

### PROBLEM DESCRIPTION
I have custom Tasmota code and find it a challenge to rebase my changes on to the latest version of Tasmota.

2 of my main customizations are:

1. Custom RF `xdrv_17_rcswitch.ino` to support both receiving RF315 and RF433 concurrently
2. Control of normal servos to open and close lids/covers

The reason I need my own code is because they can be quite hard to configure in the official Tasmota. And in the case of servos, I feel, the complexity controlling a server should not be "programmed" through configuration.

I'm aware it is always a triage to decide which features to cram into the ESP, so, having a framework to support custom code should eliminate this "fight". To someone like me, Tasmota's value lies in its Wifi, MQTT and reporting mechanisms which interfaces very well with software like HomeAssistant and NodeRed, something which I would like to leverage on.

So, I hope to suggest a way to make integrating customizations much more easier:

1. My idea is to pre-allocate a number of blank `xdrv_*`, `xsns_*` etc for customizations
2. In the main branch, they will be left blank with maybe just some example code
3. This way, once there is a new Tasmota release, we can just replace these files and automate our custom compilation

I came up with the idea only last night and have not scrutinized through every detail.
Open to hearing any opinions on this. Thank you.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2021-05-04:
I suppose you've read the documentation regarding API https://tasmota.github.io/docs/API/ and https://tasmota.github.io/docs/Sensor-API/

The idea was that users could easily add their drivers or sensors using an ID from 90 to 97.

So to add seamlessly you just need to select an unused driver or sensor id in that range.

bilogic, 2021-05-04:
No, I have not. Let me have a read. Thank you.

ascillato2, 2021-05-04:
Moving to discussions as this is a question and not a bug report

closed 2021-05-04 by ascillato2