arendst/Tasmota #12024: Tcpbridge: modbus 32bit checksum error

nikito7, 2021-05-05:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I use tasmota as TCP Bridge for modbus

Rtu over TCP

16bits registers are ok.

32bits I got crc error at the end

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
13:32:52.502 TCP: to MCU/1: 010400160002900F
13:32:52.700 TCP: from MCU: 010408004E3B710000007E72A9
13:32:53.952 TCP: to MCU/1: 010400160002900F
13:32:54.201 TCP: from MCU: 010408004E3B720000007E36A9
13:32:55.361 TCP: to MCU/1: 010400160002900F
13:32:55.559 TCP: from MCU: 010408004E3B720000007E36A9
13:32:56.810 TCP: to MCU/1: 010400160002900F
13:32:57.008 TCP: from MCU: 010408004E3B720000007E36A9
13:32:58.270 TCP: to MCU/1: 010400160002900F

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

ESP32

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2021-05-05:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

nikito7, 2021-05-05:
Done

nikito7, 2021-05-05:
Tasmota strip crc bytes I think

Tasmota: from MCU: 010404004E430FEAA7

Python: [pymodbus.transaction] RECV: 0x1 0x4 0x4 0x0 0x4e 0x43 0xf

nikito7, 2021-05-07:
2bytes register count 1 work
2bytes register count 7 work
4bytes register count 1 not

Is needed special mode?

@gemu2015

nikito7, 2021-05-08:
So checksum in log it's correct.
Using wireshark show same hex.

Closing, not tasmota problem

closed 2021-05-08 by nikito7