arendst/Tasmota #12030: Script status (>U) not receiving JSON payload.

leocomitale, 2021-05-06:
### PROBLEM DESCRIPTION
The >U status update section in scripting is not receiving its JSON payload data.  All the data returned appears to be blank.  I tried this with network status (see the included sample script) and power monitoring status with the same result.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff S31
- [ ] Tasmota binary firmware version number used:
  - [ ] Pre-compiled
  - [X] Self-compiled from latest development branch
- [X] Flashing tools used: Web interface.
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
15:21:58.062 CMD: Backlog Template; Module; GPIO 255
15:21:58.065 SRC: WebConsole from 192.168.1.50
15:21:58.068 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
15:21:58.090 SRC: Backlog
15:21:58.093 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
15:21:58.098 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
15:21:58.296 SRC: Backlog
15:21:58.300 CMD: Group 0, Index 1, Command "MODULE", Data ""
15:21:58.304 RSL: RESULT = {"Module":{"41":"Sonoff S31"}}
15:21:58.502 SRC: Backlog
15:21:58.505 CMD: Group 0, Index 1, Command "GPIO", Data "255"
15:21:58.512 RSL: RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"3072":"CSE7766 Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3104":"CSE7766 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```lua
15:23:43.365 CMD: status 0
15:23:43.368 SRC: WebConsole from 192.168.1.50
15:23:43.371 CMD: Group 0, Index 1, Command "STATUS", Data "0"
15:23:43.377 RSL: STATUS = {"Status":{"Module":41,"DeviceName":"EspressoMachine","FriendlyName":["Tasmota"],"Topic":"tasmota_067C93","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
15:23:43.383 RSL: STATUS1 = {"StatusPRM":{"Baudrate":4800,"SerialConfig":"8E1","GroupTopic":"tasmotas","OtaUrl":"https://github.com/tasmota/binaries/raw/master/tasmota-scripting.bin","RestartReason":"Software/System restart","Uptime":"0T00:13:40","StartupUTC":"2021-05-06T19:10:03","Sleep":50,"CfgHolder":4617,"BootCount":61,"BCResetTime":"2020-10-14T22:14:56","SaveCount":1414,"SaveAddress":"F8000"}}
15:23:43.389 RSL: STATUS2 = {"StatusFWR":{"Version":"9.4.0.3(tasmota)","BuildDateTime":"2021-05-06T15:07:34","Boot":31,"Core":"2_7_4_3","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"458/699"}}
15:23:43.395 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Redacted",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8001","2805C8000100068000005A0A000000000000","00000281","00006000","00000000"]}}
15:23:43.404 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":615,"Free":388,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","87DAC787","043E8001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
15:23:43.410 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_067C93-7315","IPAddress":"192.168.1.66","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.161","Mac":"C8:2B:96:06:7C:93","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
15:23:43.418 RSL: STATUS7 = {"StatusTIM":{"UTC":"2021-05-06T19:23:43","Local":"2021-05-06T15:23:43","StartDST":"2021-03-29T02:00:00","EndDST":"2021-03-29T02:00:00","Timezone":99,"Sunrise":"00:20","Sunset":"15:12"}}
15:23:43.423 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
15:23:43.430 RSL: STATUS10 = {"StatusSNS":{"Time":"2021-05-06T15:23:43","ENERGY":{"TotalStartTime":"2020-10-15T03:55:24","Total":70.759,"Yesterday":0.362,"Today":0.548,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
15:23:43.436 RSL: STATUS11 = {"StatusSTS":{"Time":"2021-05-06T15:23:43","Uptime":"0T00:13:40","UptimeSec":820,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Redacted","BSSId":"B0:6E:BF:DE:A8:40","Channel":11,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
15:37:03.214 SRC: WebGui from 192.168.1.50
15:37:03.217 SRC: Rule
15:37:03.220 CMD: Group 0, Index 1, Command "STATUS", Data "5"
15:37:03.225 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota_067C93-7315","IPAddress":"192.168.1.66","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.161","Mac":"C8:2B:96:06:7C:93","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
15:37:03.230 hey there, the gateway is 
15:37:03.236 RSL: STATE = {"Time":"2021-05-06T15:37:03","Uptime":"0T00:27:00","UptimeSec":1620,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"MonkeyDoodle","BSSId":"B0:6E:BF:DE:A8:40","Channel":11,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:05"}}
15:37:03.241 RSL: RESULT = {"POWER":"ON"}
15:37:03.244 RSL: POWER = ON
15:37:03.321 CFG: Saved to flash at F4, Count 1418, Bytes 4096
15:37:03.329 DMP: F2 5A 03 07 00 00 06 3A 00 42 1D 04 5C 14 53 8A B8 A6 2D BB 61 0F 7B 2B
15:37:03.366 DMP: F2 5A 03 07 00 00 06 3A 00 42 1D 04 5C 14 53 8A B8 A7 C8 1F 61 0F 7B 2B
15:37:03.470 DMP: F2 5A 03 07 00 00 06 3A 00 42 1D 04 5C 14 53 8A B8 A9 61 84 61 0F 7B 2B
15:37:03.574 DMP: F2 5A 03 07 00 00 06 3A 00 42 1D 04 5C 14 53 8A B8 AA D7 A9 61 0F 7B C7
15:37:03.678 DMP: F2 5A 03 07 00 00 06 3A 00 42 1D 04 5C 14 53 8A B8 AA D7 A9 61 0F 7B C7
[... 246 lines cut ...]
>D
gw=""

; Request Status information. The response will trigger the `U` section
>P
+>status 5

; Read the status JSON payload
>U
gw=StatusNET#Gateway
print hey there, the gateway is %gw%
```
- toggle power and check the console log for the gateway to be printed
- currently the script prints "hey there, the gateway is ", indicating the variable was not updated during the status call.

### EXPECTED BEHAVIOUR
The script should print the gateway network setting to the console with the message "hey there, the gateway is 192.168.1.1".

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
The last working commit I found was 4c1011d0aaff66381a245f7192667cda339d3d21. 

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-05-06:
I've also seen this doing testing, and the `>P` section does get executed with the `Status 5` JSON payload appearing in the console. It just does not get picked up in the `>U` section like it did on previous Tasmota versions. My testing was done building with old and current source, with exactly the same setup for both tests.

gemu2015, 2021-05-07:
checked this and 1. result is 
script section >U only gets 1. part of message {"StatusNET
this means that it already has been processed by another part of tasmota with a JSON parser. the JSON parser destroys the JSON string. this is not scripter related.
will try to check where the message gets lost.

gemu2015, 2021-05-07:
ok, found the reason.  
the change was for rules.
this means the status is now passed to >E  (FUNC_RULES_PROCESS) and later to section >U.
now 2 options are open.  
1. prevent status messages to be passed to >E (needs code change)
2. or nor longer use section >U (USE_SCRIPT_STATUS obsolete, don't define, no code change needed)

with the current version you can evaluate all status messages in >E.

any suggestions ?

sfromis, 2021-05-07:
That should not depend on if you are using a broker or not. As gemu2015 mentioned, it should already be working if you just use the `>E` section where you now use `>U`, with the complication that other payloads are also arriving there. At least that worked for the small example script you posted, also when not having a MQTT connection.

leocomitale, 2021-05-09:
Just to follow up, my preference would be to restore the >U functionality.  >E does work, but to me it feels like I am relying on a side effect of other functionality that could easily stop working in the future.

ascillato2, 2021-05-16:
>my preference would be to restore the >U functionality. >E does work, but to me it feels like I am relying on a side effect of other functionality that could easily stop working in the future.

Moving this issue to discussion as it is now a feature request.

closed 2021-05-16 by ascillato2