arendst/Tasmota #12031: TLS Failing since commit 63c7e5f7bfb886c6ce4318d744e0191f78a77204

ksaye, 2021-05-06:
### PROBLEM DESCRIPTION
Ever since the TLS changes in 63c7e5f7bfb886c6ce4318d744e0191f78a77204, it fails on NodeMCU.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): _NodeMCU ESP8266EX _
- [X] Tasmota binary firmware version number used: _9.4.0.3__
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: _Visual Studio Code____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
1. Clone the repo
2. change to the Tasmota directory
3. Run the command ```Git checkout -f 63c7e5f7bfb886c6ce4318d744e0191f78a77204```
4. Add the ```#define USE_MQTT_TLS``` to your user_config_override.h file
5. Compile and flash your NodeMCU
6. Configure the WiFi
7. Go into Configuration --> Configure MQTT --> Define an MQTT Host, select use TSL, set the port to 8883 and configure userid and password.
8. After it reboots, you will see a similar message below:
```
00:00:00.002 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at FA, Count 10
00:00:00.055 FRC: Some settings have been reset (2)
00:00:00.062 Project tasmota Tasmota Version 9.4.0.3(tasmota)-2_7_4_9(2021-05-06T16:03:24)
00:00:00.551 WIF: Connecting to AP1 iotdevices Channel 3 BSSId FE:92:BF:64:58:1E in mode 11n as tasmota_126366-0870...
00:00:01.754 WIF: Connected
00:00:02.005 HTP: Web server active on tasmota_126366-0870 with IP address 192.168.15.45
22:09:31.035 MQT: Attempting connection...

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Abort called

>>>stack>>>

ctx: sys
sp: 3fff7d88 end: 3fffffb0 offset: 0000
3fff7d88:  3fff679c 00000100 3fff4248 deadbeef
3fff7d98:  000000fe 00000000 00000000 00000000
3fff7da8:  00000000 00000000 00000000 00000000  
3fff7db8:  00000000 3fff4a58 deadbeef 003c3879
3fff7dc8:  00000008 00000000 3fff4264 402491fe
[... 228 lines cut ...]
3fff8c18:  2be008f8 f6a8c543 94f80204 d87abd2a
3fff8c28:  925d75d0 b2ab7f71 9c253522 d111146e
3fff8c38:  61a752df e27c4497 82ad2d63 f3c4920d  
3fff8c48:  60076b43 1a121fcf e5d40a88 fceb1271
3fff8c58:  a3512b51 b8de7bd4 4ae27cf2 a1a83470
3fff8c68:  6a46adfb 991ff23f d6583eba 416d6452
3fff8c78:  96a1036e 55dbb94f a1c835be bd5b74dc  
3fff8c88:  b5d04410 d6cb4f98 81e3772a 7cc0b192
3fff8c98:  00000d09 0201031a 04120040 02010501
3fff8ca8:  05030401 02030203 06010602 0e000003  
3fff8cb8:  00000000 00000000 0af511b0 0646057a
```

### EXPECTED BEHAVIOUR
It worked in the prior commit: c398ce73641af93c8582faca1facfb8399aff1ba

It continues to fail (has not been fixed) in the current: 3c4acc952868bcb1109bf9d756dd43b5d43f63dc

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2021-05-07:
Will take a look

s-hadinger, 2021-05-07:
I can't reproduce. I suspect it's a wrong size for the ThunkStack that causes the crash.

What are your compile options?

s-hadinger, 2021-05-08:
Can you try to raise all values of '_stackSize' to 5300/4 here: https://github.com/arendst/Tasmota/blob/ad4341146994798629118cf4c643a4b22a27eda7/lib/lib_ssl/tls_mini/src/StackThunk_light.cpp#L44

ksaye, 2021-05-08:
That was it!  It looks like the stack was just a little under the limit, as shown by the info below

```
15:32:01.024 MQT: Attempting connection...
15:32:01.668 MQT: TLS connected in 641 ms, max ThunkStack used 3700
15:32:01.669 MQT: MFLN not supported by TLS server
15:32:01.670 MQT: Connected
```

I will submit a small PR to address my use case.

closed 2021-05-08 by ksaye

ca-wo, 2021-05-09:
Sorry for bothering, but if using TLS with fingerprint, the stacksize of 3600 (line 49) in my case was to low either. So the value in the else case should also be raised.

As you can see, my Thucksize used is:

`08:17:43.006 MQT: Attempting connection...
08:17:43.332 MQT: TLS connected in 323 ms, max ThunkStack used 3684
08:17:43.333 MQT: Fingerprint learned: 55 .........
08:17:43.408 MQT: Connected`

I don't know, what the ideal value could be, in my case it is 3700
Maybe this helps any further

s-hadinger, 2021-05-09:
Thanks for the feedback, I have raised it to 3800 to have some headroom.

PR #12063 (2021-05-09): Increase min stack thunk to 3800 #12031
## Description:

Increase minimum stack thunk to 3800 bytes.

**Related issue (if applicable):** fixes #12031

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works on Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with core ESP32 V.1.0.6
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRI [... cut]