arendst/Tasmota #12051: Multipress functions not working with SetOption19 0 (ver 9.4.0)

owendemooy, 2021-05-08:
### PROBLEM DESCRIPTION
When I change my Sonoff Touch T1 from using SetOption19 1 to SetOption19 0, I lose the ability to use the multi-press functions on the device. When I double, triple or hold the button on the Sonoff a related MQTT message is not generated by the device. A single press still creates an MQTT message and toggles the relay as expected.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Sonoff Touch T1 US (single gang)
- [ x] Tasmota binary firmware version number used: 9.4.0(tasmota)
  - [ x] Pre-compiled
  - [ ] Self-compiled
- [ x] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
13:51:43.294 CMD: Backlog Template; Module; GPIO 255
13:51:43.369 MQT: tasmota/stat/kitchen/RESULT = {"NAME":"Generic","GPIO":[32,1,1,1,0,0,0,0,224,320,0,0,0,0],"FLAG":0,"BASE":28}
13:51:43.580 MQT: tasmota/stat/kitchen/RESULT = {"Module":{"28":"Sonoff T1 1CH"}}
13:51:43.836 MQT: tasmota/stat/kitchen/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
13:42:47.182 CMD: Backlog Rule1; Rule2; Rule3
13:42:47.252 MQT: tasmota/stat/kitchen/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
13:42:47.464 MQT: tasmota/stat/kitchen/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
13:42:47.716 MQT: tasmota/stat/kitchen/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
13:43:05.836 CMD: Status 0
13:43:05.843 MQT: tasmota/stat/kitchen/STATUS = {"Status":{"Module":28,"DeviceName":"kitchen_switch","FriendlyName":["kitchen_light"],"Topic":"kitchen","ButtonTopic":"","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
13:43:05.852 MQT: tasmota/stat/kitchen/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"1T06:57:13","StartupUTC":"2021-05-07T05:45:52","Sleep":50,"CfgHolder":4617,"BootCount":39,"BCResetTime":"2020-05-02T08:42:34","SaveCount":2176,"SaveAddress":"F6000"}}
13:43:05.861 MQT: tasmota/stat/kitchen/STATUS2 = {"StatusFWR":{"Version":"9.4.0(tasmota)","BuildDateTime":"2021-04-23T10:07:22","Boot":7,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"435/699"}}
13:43:05.867 MQT: tasmota/stat/kitchen/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["O.and.A (IoT)",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["400A880B","1405C8000100060000005A0A000000000000","00000280","00006000","00000000"]}}
13:43:05.879 MQT: tasmota/stat/kitchen/STATUS4 = {"StatusMEM":{"ProgramSize":599,"Free":404,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDAC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6"}}
13:43:05.886 MQT: tasmota/stat/kitchen/STATUS5 = {"StatusNET":{"Hostname":"sonoff_kitchen","IPAddress":"10.1.10.31","Gateway":"10.1.10.254","Subnetmask":"255.255.254.0","DNSServer":"10.1.10.254","Mac":"2C:F4:32:AA:7A:E9","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
13:43:05.891 MQT: tasmota/stat/kitchen/STATUS6 = {"StatusMQT":{"MqttHost":"#REMOVED FOR PRIVACY#","MqttPort":1883,"MqttClientMask":"sonoff_kitchen","MqttClient":"sonoff_kitchen","MqttUser":"DVES_USER","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
13:43:05.898 MQT: tasmota/stat/kitchen/STATUS7 = {"StatusTIM":{"UTC":"2021-05-08T12:43:05","Local":"2021-05-08T13:43:05","StartDST":"2021-03-28T02:00:00","EndDST":"2021-10-31T03:00:00","Timezone":"+01:00","Sunrise":"05:17","Sunset":"20:15"}}
13:43:05.905 MQT: tasmota/stat/kitchen/STATUS10 = {"StatusSNS":{"Time":"2021-05-08T13:43:05"}}
13:43:05.913 MQT: tasmota/stat/kitchen/STATUS11 = {"StatusSTS":{"Time":"2021-05-08T13:43:05","Uptime":"1T06:57:13","UptimeSec":111433,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER":"OFF","Wifi":{"AP":1,"SSId":"O.and.A (IoT)","BSSId":"82:2A:A8:11:4B:02","Channel":11,"RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
1. With SetOption19 = on
- Single tap switch
- Double tap switch

2. SetOption19 0
- Single tap switch
- Double tap switch

3. SetOption19 1
- Single tap switch
- Double tap switch

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
When I SetOption19 0 I expect Tasmota to send MQTT messages for action "Double", "Triple" or "HOLD"

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

wasp100, 2021-05-14:
Hi @owendemooy, it’s working fine with my setup. Make sure `Setoption73` is `on`.
Then set a rule to couple the relay to single press.

ascillato2, 2021-05-16:
Hi, any news on this?

owendemooy, 2021-05-17:
Okay so Im a bit confused SetOption73 is set to off... and doesnt not change when I SetOption19 to on or off... but when I SetOption19 to off I loose the multipress buttons

I believe SetOption73 just disconnects the buttons from the relay so you have to use rules... I dont want to use rules as my current setup doesnt require them.

From what I understood Tasmota introduced Multipress functionality without the need for rules a while back.... As I used to use rules but moved away from that when it was added.

As per this link I have ensured ButtonTopic, SetOption1, SetOption11 and SetOption13 are set to default values
https://tasmota.github.io/docs/Buttons-and-Switches/#multi-press-functions
ButtonTopic = 0
SetOption1 = 0
SetOption11 = 0
SetOption13 = 0

Interesting the above article states the MQTT messages for multi-press events should be published as such:
stat/tasmota/BUTTON<x> = {"ACTION":"SINGLE"}
stat/tasmota/BUTTON<x> = {"ACTION":"DOUBLE"}
stat/tasmota/BUTTON<x> = {"ACTION":"TRIPLE"}
stat/tasmota/BUTTON<x> = {"ACTION":"HOLD"}

Where as under my config it is posted as such
stat/kitchen/BUTTON1T = {"TRIG":"DOUBLE"}
stat/kitchen/BUTTON1T = {"TRIG":"HOLD"}

And in either config I have no rules as you can see from the original post or below
10:55:41.804 CMD: rule
10:55:41.812 MQT: tasmota/stat/kitchen/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

I have over time updated these devices from older versions of tasmota always following the upgrade instructions as I go.

owendemooy, 2021-05-17:
Okay so I fully reset the configuration of the device, and resetup everything... the only commands I entered were as follows:

SetOption30 = 1 - to publish relay and light in HA
LedState 0 - to disable blue status led

When I double, triple click the button nothing happens.

I then SetOption19 1 - to enable HA MQTT discovery

And when I now double, triple click I get the following example MQTT message published:
stat/kitchen/BUTTON1T = {"TRIG":"DOUBLE"}

I have not changed anything else.... why when HA MQTT discovery is enabled do I get different behavior.

owendemooy, 2021-05-23:
Hi, any news on this?

blakadder, 2021-05-27:
You're wrong. There is no expectation on having multipress MQTT messages until you turn on `SO73` neither it is documented that it will be so. When you turn on MQTT discovery the topics are generated due to MQTT discovery format demands.

In short: use `SO73 1` with `SO19 0`

closed 2021-05-27 by blakadder

Malakai-DB, 2021-10-06:
Sorry to hijack a closed issue, but I seem to have the same behavior with a Shelly 1 flashed with Tasmota 9.5.0 and I thought it would be better to write here than opening a new issue (as it seems this is not really an issue but probably a misconfiguration).
In my case, after enabling `SO73 1` and `SO19 0`, I still am able to control the relay locally (which from what I understood I shouldn't as the button press should be sent through MQTT instead) and nothing is sent over MQTT. If I double press the button, it simply toggles twice the light (I am using the Shelly 1 in a Shelly Button).
@owendemooy : did you manage to solve it? Was it a bad configuration?

Malakai-DB, 2022-01-28:
Hello,

Just wanted to report that the issue I was having was a misunderstanding of the SO73.
Like I said, I am using a Shelly 1 in a Shelly Button case which allows to trigger a push button to toggle the relay. Tried the SO73 on another hardware and it is working as expected.
But in my case, as the Shelly Button is an accessory to the Shelly Relay, the latter doesn't have any information regarding the button press, so the SO73 can't work on this hardware.
Instead, you can activate SO114 which allows to detach SWITCHES (and not BUTTONS) from relays and send MQTT messages instead. This time it's not the button press that is detected but the switch (which a Shelly 1 in a Shelly Button case can detect). You won't get multi-press like this (as it can only detect the toggle) but in my case I wanted to detach the switch from the relay. Be informed that you won't be able to control the relay from the button anymore in this way, but this fits my current need.

Just felt like mentioning it, it could help others trying to mimic this use case with a Shelly Relay and Shelly Button.