arendst/Tasmota #12058: OTA upgrade is possible without authentication

cociweb, 2021-05-08:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Tasmota is updateable without pasword

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): Sonoff S2X
- [X ] Tasmota binary firmware version number used: 9.4.0
  - [ ] Pre-compiled
  - [X ] Self-compiled
- [X ] Flashing tools used: platformio
- [X ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
Custom
```lua
  Configuration output here:
N/A

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
N/A

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
N/A

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

N/A
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
call the url from browser:
http://<IP>/cm?user=admin&password=joker&cmnd=Upgrade%201
it will upgrade your device, while the password is incorrect
### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Deny the upgrade, because the user/password was incorrect.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
N/A

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I've not checked that changing the OTAurl without auth is possible. If it is possible it's a very serious security issue.

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-05-08:
Trying http commands, I see the issue differently. When you give an invalid password in the URL, Tasmota will request user/password from the browser. If already having a valid logon in another browser tab, the browser will probably just silently supply the already verified details. When I execute the same command on another browser, without an active logon, I see the warning about needing user/password. This may be "surprising" to some, but may not really be a bug.

cociweb, 2021-05-09:
Hello sfromis,
Thank you for the heads-up!
Yes, It seems, I've worked from caches. As of now, I deleted them, and the update is restricted as it should.
Sorry for the inconvenience.
Br,
Cociweb

closed 2021-05-09 by cociweb