arendst/Tasmota #12088: wrong Json with PCF8574 sensor (missing curly brace)

megamarco833, 2021-05-12:
### PROBLEM DESCRIPTION
json format error on PCF8574 driver usage with sensor enabled

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): _____
- [x] Tasmota binary firmware version number used: _____
  - [] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: gitpod
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
15:16:21.558 MQT: stat/tasmota_156/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
15:16:21.781 MQT: stat/tasmota_156/RESULT = {"Module":{"0":"Generic"}}
15:16:22.036 MQT: stat/tasmota_156/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"608":"I2C SCL"},"GPIO2":{"33":"Button2"},"GPIO3":{"640":"I2C SDA"},"GPIO4":{"0":"None"},"GPIO5":{"34":"Button3"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
15:16:25.391 MQT: tele/tasmota_156/STATE = {"Time":"2021-05-12T15:16:25","Uptime":"0T01:45:20","UptimeSec":6320,"Heap":17,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","POWER8":"OFF","POWER9":"OFF","POWER10":"OFF","POWER11":"OFF","POWER12":"OFF","POWER13":"OFF","POWER14":"OFF","POWER15":"OFF","Wifi":{"AP":1,"SSId":"NETGEAR","BSSId":"E0:3F:49:A8:C5:70","Channel":11,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:03"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
compile with:
```
  #define USE_PCF8574                            // [I2cDriver2] Enable PCF8574 I/O Expander (I2C addresses 0x20 - 0x26 and 0x39 - 0x3F) (+1k9 code)
    #define USE_PCF8574_SENSOR                   // enable PCF8574 inputs and outputs in SENSOR message
    #define USE_PCF8574_DISPLAYINPUT             // enable PCF8574 inputs display in Web page
    #define USE_PCF8574_MQTTINPUT                // enable MQTT message & rule process on input change detection : stat/%topic%/PCF8574_INP = {"Time":"2021-03-07T16:19:23+01:00","PCF8574-1_INP":{"D1":1}}

```

### EXPECTED BEHAVIOUR
this is the output received:
13:19:07.459 MQT: tele/tasmota_156/SENSOR = {"Time":"2021-05-12T13:19:07","PCF8574-1":{"D0":1,"D1":1,"D2":1,"D3":1,"D4":1,"D5":1,"D6":1,"D7":1,"PCF8574-2":{"D0":1,"D1":1,"D2":1,"D3":1,"D4":0,"D5":1,"D6":1,"D7":1}}

it actually is not valid JSON. A curly brace after the first sensor data is missing.
should be:
13:19:07.459 MQT: tele/tasmota_156/SENSOR = {"Time":"2021-05-12T13:19:07","PCF8574-1":{"D0":1,"D1":1,"D2":1,"D3":1,"D4":1,"D5":1,"D6":1,"D7":1###right here###,"PCF8574-2":{"D0":1,"D1":1,"D2":1,"D3":1,"D4":0,"D5":1,"D6":1,"D7":1}}

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-05-12:
Looks like a simple omission at the end of line 222 in the [driver](https://github.com/arendst/Tasmota/blob/development/tasmota/xdrv_28_pcf8574.ino#L222)
`ResponseAppend_P(PSTR(",\"PCF8574%c%d\":{\"D0\":%i,\"D1\":%i,\"D2\":%i,\"D3\":%i,\"D4\":%i,\"D5\":%i,\"D6\":%i,\"D7\":%i"),`

arendst, 2021-05-12:
Thx for reporting

megamarco833, 2021-05-12:
thanks to you for fixing so quickly and to @sfromis  to help me again and again!

closed 2021-05-12 by ascillato2

tobschn, 2021-06-14:
I'm still having problems with this. In my case there is one currly brace too many at the end. 
Additionally to the above I compiled with USE_SR04.

The given output:

> `{"Time":"2021-06-14T11:43:59","SR04":{"Distance":23.596},"PCF8574-1":{"D0":1,"D1":1,"D2":1,"D3":1,"D4":1,"D5":1,"D6":1,"D7":1},"PCF8574-2":{"D0":1,"D1":0,"D2":1,"D3":1,"D4":1,"D5":0,"D6":1,"D7":0}}}`

how it should be:

>`{"Time":"2021-06-14T11:43:59","SR04":{"Distance":23.596},"PCF8574-1":{"D0":1,"D1":1,"D2":1,"D3":1,"D4":1,"D5":1,"D6":1,"D7":1},"PCF8574-2":{"D0":1,"D1":0,"D2":1,"D3":1,"D4":1,"D5":0,"D6":1,"D7":0}}`

arendst, 2021-06-14:
Thx for reporting.

Pls retry latest development release (pre-compiled available in an hour)

tobschn, 2021-06-14:
Issue is fixed. Works now correctly, thanks!