arendst/Tasmota #12107: Tasmota_sensors 9.4.0 - BME280 being identified as BMP280

mrneutron42, 2021-05-15:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Tasmota-sensors version 9.4.0 is identifying a BME280 as a BMP280.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic):  development board ESP12F_Relay_X1
- [x ] Tasmota binary firmware version number used: 9.4.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [x ] Flashing tools used: Tasmotizer 1.2
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:

```lua
  Configuration output here:
10:33:33.135 CMD: Backlog Template; Module; GPIO 255
10:33:33.139 SRC: WebConsole from 192.168.7.41
10:33:33.142 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
10:33:33.196 SRC: Backlog
10:33:33.200 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
10:33:33.206 MQT: sensors/BMP280/RESULT = {"NAME":"Generic","GPIO":[1,1,640,1,608,224,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
10:33:33.417 SRC: Backlog
10:33:33.420 CMD: Group 0, Index 1, Command "MODULE", Data ""
10:33:33.427 MQT: sensors/BMP280/RESULT = {"Module":{"18":"Generic"}}
10:33:33.668 SRC: Backlog
10:33:33.671 CMD: Group 0, Index 1, Command "GPIO", Data "255"
10:33:33.680 MQT: sensors/BMP280/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"640":"I2C SDA"},"GPIO3":{"0":"None"},"GPIO4":{"608":"I2C SCL"},"GPIO5":{"224":"Relay1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
10:33:39.168 MQT: sensors/BMP280/STATE = {"Time":"2021-05-15T10:33:39","Uptime":"0T00:03:10","UptimeSec":190,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"homeland","BSSId":"A4:13:4E:8B:8B:E1","Channel":11,"RSSI":100,"Signal":-49,"LinkCount":1,"Downtime":"0T00:00:03"}}
10:33:39.197 MQT: sensors/BMP280/SENSOR = {"Time":"2021-05-15T10:33:39","BMP280":{"Temperature":21.6,"Pressure":981.4},"PressureUnit":"hPa","TempUnit":"C"}

```
- [x ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
 NA

```
- [x ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
10:34:30.906 CMD: STATUS 0
10:34:30.909 SRC: WebConsole from 192.168.7.41
10:34:30.912 CMD: Group 0, Index 1, Command "STATUS", Data "0"
10:34:30.918 MQT: sensors/BMP280/STATUS = {"Status":{"Module":18,"DeviceName":"ESP-12F-Dev-Board","FriendlyName":["ESP-12F-Dev-Board"],"Topic":"BMP280","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
10:34:30.950 MQT: sensors/BMP280/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T00:04:01","StartupUTC":"2021-05-15T15:30:29","Sleep":50,"CfgHolder":4617,"BootCount":40,"BCResetTime":"2021-05-11T16:34:33","SaveCount":88,"SaveAddress":"F4000"}}
10:34:30.986 MQT: sensors/BMP280/STATUS2 = {"StatusFWR":{"Version":"9.4.0(sensors)","BuildDateTime":"2021-04-23T10:07:18","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"360/699"}}
10:34:31.002 MQT: sensors/BMP280/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["wifiAP",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A0A000000000000","00000080","00006000","00000000"]}}
10:34:31.033 MQT: sensors/BMP280/STATUS4 = {"StatusMEM":{"ProgramSize":653,"Free":348,"Heap":21,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8FDA8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","00000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74"}}
10:34:31.076 MQT: sensors/BMP280/STATUS5 = {"StatusNET":{"Hostname":"BMP280-1920","IPAddress":"192.168.7.132","Gateway":"192.168.7.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.7.1","Mac":"E0:98:06:0E:67:80","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
10:34:31.095 MQT: sensors/BMP280/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.7.22","MqttPort":1883,"MqttClientMask":"ESP-12F_1","MqttClient":"ESP-12F_1","MqttUser":"ESP-12F_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
10:34:31.121 MQT: sensors/BMP280/STATUS7 = {"StatusTIM":{"UTC":"2021-05-15T15:34:31","Local":"2021-05-15T10:34:31","StartDST":"2021-03-14T02:00:00","EndDST":"2021-11-07T02:00:00","Timezone":99,"Sunrise":"05:44","Sunset":"20:34"}}
10:34:31.142 MQT: sensors/BMP280/STATUS10 = {"StatusSNS":{"Time":"2021-05-15T10:34:31","BMP280":{"Temperature":21.5,"Pressure":981.4},"PressureUnit":"hPa","TempUnit":"C"}}
10:34:31.155 MQT: sensors/BMP280/STATUS11 = {"StatusSTS":{"Time":"2021-05-15T10:34:31","Uptime":"0T00:04:02","UptimeSec":242,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"wifiAP","BSSId":"xx:xx:xx:xx:xx:xx","Channel":11,"RSSI":100,"Signal":-48,"LinkCount":1,"Downtime":"0T00:00:03"}}}
10:34:33.011 WIF: Checking connection...

```
- [x ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:00:00.001 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at F6, Count 86
00:00:00.055 QPC: Count 1
00:00:00.056 CFG: CR 360/699, Busy 0
[... 37 lines cut ...]

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Based on: https://github.com/arendst/Tasmota/blob/development/tasmota/xsns_09_bmp.ino
I expect the Tasmota-sensors firmware to scan the i2c bus, read the chip ID register 0xD0 and determine if the chip is a 0x60 (BME280) or a 0x58 (BMP280).

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

I bought some BMP280 sensor boards first. (sensor parts have a K marker on them)  
They were recognized by Tasmota-sensors as BMP280.
I have been successfully reading and sending Temperature and Pressure data via MQTT.  
I named the MQTT topic sensors/BMP280.
Next, I got some BME280 boards. (sensor parts have a U marker on them)
I connected the BME280 board i2c pins to the same i2c SDA and SCK pins of the ESP-12F module.
But, Tasmota-sensors reads the BME280 and identifies it as a BMP280.

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2021-05-15:
My BME280 chips are detected perfectly. Have you verified that it is actually BME280? It has happened that a seller had a board supporting both sensors, populated with the older BMP280, and claimed it (knowingly or not) to be BME280.

I've got both, BME280 left and BMP280 right. The BME280 is near square, and has the hole center on one edge.
[image]

mrneutron42, 2021-05-15:
Oh.  The devices shipped to me as BME280 are rectangle rather than square.
It looks like I was shipped the wrong parts!
In case I need to prove these are BMP280 rather than BME280, is there a way to read the CHIP ID register via the Tasmota console?

sfromis, 2021-05-15:
AFAIK, not via the console. Using the [scripting language](https://tasmota.github.io/docs/Scripting-Language/), you should since rather recently be able to do basic I2C coding to read registers. Or using the "more standard" Berry scripting language on ESP32.

closed 2021-05-15 by mrneutron42