arendst/Tasmota #12124: MQTT Reconnect

CRCinAU, 2021-05-17:
### PROBLEM DESCRIPTION
When I restart mosquitto, Tasmota loaded on a Sonoff Pow R2, I note that I have to reboot the R2 to get stats delivered via MQTT again.

### REQUESTED INFORMATION
- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff Pow R2
- [X] Tasmota binary firmware version number used: 9.4.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Can't recall :(

Version:
Program Version | 9.4.0(tasmota)
-- | --
2021-04-23T10:07:22
2_7_4_9/2.2.2-dev(38a443e)

closed 2021-05-17 by close-issue-app[bot]

ascillato, 2021-05-17:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Tasmota/issues/new/choose) in order to have more information so as to properly help you? Also, the most important thing to be completed of the template, would be the console log when your issue happens. Remember to set weblog to 4 before that.

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

CRCinAU, 2021-05-18:
@ascillato  - I believe I've actually found the issue.....

When Tasmota reboots, it will send the DISCOVERY topic to the MQTT server - which HomeAssistant picks up and makes the device available.

If you reboot HA / MQTT, the DISCOVERY topic is gone and therefore HA marks the device unavailable. When Tasmota reconnects to the MQTT server, it does not send the DISCOVERY topic to the MQTT server.

As such, the device will always show as unavailable in Home Assistant until the DISCOVERY topic is published again. Currently, this will only happen on rebooting the Tasmota device.